platformio/platformio-core #274: install fails when using Develop.zip

gslender, 2015-08-22:
When installing from scratch using the 
`$ pip install https://github.com/platformio/platformio/archive/develop.zip`

You get this error when calling platform run
```
Error: The PlatformIO and `scons` aren't installed properly. Please use official installation manual:
http://docs.platformio.org/en/latest/installation.html
```

This was on OS X fresh install

gslender, 2015-08-22:
Solved the problem by executing `pip install --egg scons`

ivankravets, 2015-08-22:
Thanks for your reports!!! I've just significantly simplified installation process of PlatformIO. See http://docs.platformio.org/en/latest/installation.html#python-package-manager

Is this page clear now? Don't forget that you can directly edit https://github.com/platformio/platformio/blob/develop/docs/installation.rst and make pull-request. I'll merge your changes.

commit a1483c26 (2015-08-22): Simplified installation process of PlatformIO // Resolve #274
closed 2015-08-22 by ivankravets