platformio/platformio-core #275: Documentation about toolchains on platformio-storage / custom download location

pastcompute, 2015-08-22:
Hi & thanks for platformio!

I'd like to be able to build with my own custom toolchain or framework instead of using a pre-built one downloaded from platformio-storage. I have read http://docs.platformio.org/en/latest/platforms/creating_platform.html etc. but haven't found any documentation on how to integrate a custom toolchain with platformio.

Examples of use cases:

* use platformio to build using a MIPS openwrt toolchain existing on harddisk already
* regress to prior versions of the ESP8266 toolchain and API not available from platformio-storage anymore
* audit & certify toolchain and framework code for security purposes - at the moment there is no way of knowing which sources were used to build the pre-built toolchains

One way to provide some of this functionality might be to make the download URL (hardcoded in https://github.com/platformio/platformio-api/blob/develop/platformio_api/api.py ) configurable - I can try and hack a patch to make that location configurable if that helps.

It would be nice to be able to know how to symlink from the right place in ~/.platformio to a toolchain but I can live with running a local instance of platformio_api

One benefit: if the procedure used to build a toolchain or framework tarball was documented there'd probably be people submitting additional toolchain pull requests as well,

ivankravets, 2015-08-22:
Hi @pastcompute! Thank you that use PlatformIO. :+1: 

All your examples can be solved via custom development platform. Let's create it using http://docs.platformio.org/en/latest/platforms/creating_platform.html

# Platform

You should leave `PACKAGES` empty if your platform doesn't depend on PlatformIO pre-built tool/toolchain/framework.

My `test.py`:

```python
import os

from platformio.platforms.base import BasePlatform

class TestPlatform(BasePlatform):
    # This is a description of your platform.
    # Platformio uses it for the `platformio search / list` commands

    """
        My Test platform - test.py
    """

    PACKAGES = {
    }

    def get_build_script(self):
        """ Returns a path to build script """

        # You can return static path
        # return "/path/to/test-builder.py"

        # or detect dynamically if `test-builder.py` is located in the same
        # folder with `test.py`
        return os.path.join(
            os.path.dirname(os.path.realpath(__file__)),
            "test-builder.py"
        )

```

# Build script

Please replace `/path/to/toolchain/` with correct value and `toolchain-*`.

My `test-builder.py`:

```python
"""
    Build script for test.py
    test-builder.py
"""

from SCons.Script import Default, DefaultEnvironment

env = DefaultEnvironment()

# A full list with the available variables
# http://www.scons.org/doc/production/HTML/scons-user.html#app-variables
env.Replace(
    AR="toolchain-ar",
    AS="toolchain-as",
    CC="toolchain-gcc",
    CXX="toolchain-g++",
    OBJCOPY="toolchain-objcopy",
    RANLIB="toolchain-ranlib",
    SIZETOOL="toolchain-size"
)

env.PrependENVPath(
    "PATH",
    "/path/to/toolchain/bin"
)

#
# Target: Build executable and linkable firmware
#
target_bin = env.BuildProgram()

#
# Target: Define targets
#
Default(target_bin)

```

# Native platform

PlatformIO 2.3.0 has not beed released yet. It contains a few new platforms including `native`:
* [native platform](https://github.com/platformio/platformio/blob/develop/platformio/platforms/native.py)
* [native build script](https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/native.py)

This platform can be used as "base" platform and then extended via [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script).

However, you have a few solutions with stable PlatformIO 2.2.2:

* Create own platform and prepend own toolchain to $PATH
* Extend existing platform via  [extra_script](http://docs.platformio.org/en/latest/projectconf.html#extra-script) and change TOOLCHAIN to own via `env.PrependENVPath` - see above.

ivankravets, 2015-08-22:
> One way to provide some of this functionality might be to make the download URL

If you need to create own Packages Storage or add own private toolchain/tool/framework - then you need to override `PackageManager.get_manifest` and return own manifest. This should be done in one of your  own platform. For example, into `test.py` which is described above:
```python
import os

from platformio.pkgmanager import PackageManager
from platformio.platforms.base import BasePlatform
from platformio import util

def my_own_packages_manifest(cls):
    original_packages = util.get_api_result("/packages/manifest")
    # print original_packages

    #
    # Modify original packages or add own
    #

    return original_packages

PackageManager.get_manifest = my_own_packages_manifest

class TestPlatform(BasePlatform):
    # This is a description of your platform.
    # Platformio uses it for the `platformio search / list` commands

    """
        My Test platform - test.py
    """

    PACKAGES = {
    }

    def get_build_script(self):
        """ Returns a path to build script """

        # You can return static path
        # return "/path/to/test-builder.py"

        # or detect dynamically if `test-builder.py` is located in the same
        # folder with `test.py`
        return os.path.join(
            os.path.dirname(os.path.realpath(__file__)),
            "test-builder.py"
        )
```

Original package manifest http://api.platformio.org/packages/manifest

P.S: Paste this URL to JSON formatter https://jsonformatter.curiousconcept.com

pastcompute, 2015-08-23:
Hey, thanks for that
When I'm done I'll see if I can help with the wording on that manual page.

I'm still working through it, some minor observations I've made so far:

* it doesn't seem to like underscores or extra hyphens in the platform name
* you also need to be careful with case when mapping from whatever you call `test.py` to the class name `TestPlatform`

Otherwise it wont show up in `platformio search`

pastcompute, 2015-08-23:
Also:

the manual page and your example have `env.BuildProgram()` - I had to change it to `env.BuildFirmware()` otherwise I got an SCONS stack trace:

```
AttributeError: 'SConsEnvironment' object has no attribute 'BuildProgram':
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 132:
SConscript(env.subst("$BUILD_SCRIPT"))
File "/usr/lib/scons/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/usr/lib/scons/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/usr/lib/scons/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/home/andrew/.platformio/platforms/openwrtcarambola2-builder.py", line 39:
target_bin = env.BuildProgram()
```

Other than that, I'm now getting a file `.pioenvs/demo/firmware.elf` which looks like a good ELF binary that I can rename as required. So now its probably just a matter for fine tuning

ivankravets, 2015-08-23:
> When I'm done I'll see if I can help with the wording on that manual page

Please edit this page directly https://github.com/platformio/platformio/blob/develop/docs/platforms/creating_platform.rst

> the manual page and your example have env.BuildProgram() - I had to change it to env.BuildFirmware() ...

Since PlatformIO 2.3.0 this method will be renamed to `env.BuildProgram()`, because PlatformIO will have support for `desktop` platforms, where not meaning `firmware`.

You can switch to the latest `develop` version using http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2015-09-05:
@PlatformIO 2.3.0 has been released! Please reopen this issue if you still have problems/questions.

closed 2015-09-05 by ivankravets