platformio/platformio-core #276: Strange behavior of serialport monitor on Leonardo

ngollon, 2015-08-23:
I a uploading the sketch

    void setup() {
      Serial.begin(9600);
    }
    void loop() {
      Serial.println(micros(), DEC);
    }

to my Arduino leonardo. Eatching the output using the Arduino IDE's serial monitor shows everything working as expected: The microsecond output runs up continously.
Connecting to it using platformio serialports monitor leads to the output "jumping" up by several hundred milliseconds about one every two seconds without noticable pause in the output. These jumps are lead by an empty line.

ivankravets, 2015-08-23:
Try this code
```c
void setup() {
  Serial.begin(9600);
  while (!Serial) {
  }
}
void loop() {
  Serial.println(micros(), DEC);
}
```

ivankravets, 2015-08-23:
@valeros Could you reproduce this issue?

valeros, 2015-08-25:
Sorry, but I can't reproduce this behavior with my Leonardo, output runs up continuously (for both Windows and Linux).
@ngollon Could you provide more information about your system?

ngollon, 2015-08-25:
@ivankravets I tried your code and did not observe the problem. Indeed, I could lower the delay to 1ms, or delayMicroseconds(1000) without problems.
delayMicroseconds(500) still showed the effect.

When I increased the serial speed to 115200, delayMicroseconds(500) was ok, but delayMicroseconds(100) still was buggy.

An example output is:
    3885124
    3885472
    3885820
    3886168
    3886520
    3886864
    3887224
    3887568

    4303484
    4303828
    4304176
    4304528
    4304872
    4305228
    4305572

@valeros I am running Windows 10 64bit build 10525, Arduino IDE 1.0.5-r2 for the serial monitor, platformio 2.2.2. What information do you require?

ivankravets, 2015-08-31:
@ngollon I don't know how to fix this issue. @valeros reports that he can't reproduce this issue. 

What is more, PlatformIO uses [Miniterm](http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#platformio-serialports-monitor) terminal application, where I don't have influence on the source code of it.

P.S: Do you really need transfer results on each MCU clock?

ngollon, 2015-08-31:
@ivankravets No, of course not. I just noticed this issue when trying to debug my stepper motor code and thought I'd let you know. This is no show-stopper for me by any means, as I can just use the Arduino IDE's serial monitor.

ivankravets, 2015-08-31:
Thanks for the report and that using @PlatformIO! Don't forget to report any issues with it.

closed 2015-08-31 by ivankravets