platformio/platformio-core #278: Using std::string with stm32f407vg causes linker errors

apvanzanten, 2015-08-25:
Hello, I recently ran into some trouble when trying to use std::string with the stm32f407 discovery board. I'm still not 100% sure I'm not just doing something wrong, but anyway here is the simplest way to reproduce it (that I could come up with, anyway):

Initialize a project for the **disco_f407vg** board and add a main.cpp with the contents:

    #include <string>
    int main(void){
        std::string str;
        return 0;
    }

And then hit **platformio run**.

My results:

    [08/25/15 15:21:51] Processing disco_f407vg (platform: ststm32, targets: upload, framework: cmsis, board: disco_f407vg)
    --- 
    arm-none-eabi-g++ -o .pioenvs\disco_f407vg\firmware.elf -Os -Wl,--gc-sections,--relax -mthumb -mcpu=cortex-m4 -nostartfiles -nostdlib -Wl,-T C:\Users\arjen\.platformio\packages\ldscripts\stm32f405x6.ld .pioenvs\disco_f407vg\src\main.o -L.pioenvs\disco_f407vg -Wl,--start-group -lc -lgcc -lm -lstdc++ -lnosys .pioenvs\disco_f407vg\libFrameworkCMSISVariant.a -Wl,--end-group
    c:/users/arjen/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7e-m\libstdc++.a(system_error.o): In function `_GLOBAL__sub_I__ZNSt14error_categoryC2Ev':
    system_error.cc:(.text.startup._GLOBAL__sub_I__ZNSt14error_categoryC2Ev+0x24): undefined reference to `__dso_handle'
    c:/users/arjen/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: .pioenvs\disco_f407vg\firmware.elf: hidden symbol `__dso_handle' isn't defined
    c:/users/arjen/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/bin/ld.exe: final link failed: Bad value
    collect2.exe: error: ld returned 1 exit status
    scons: *** [.pioenvs\disco_f407vg\firmware.elf] Error 1
    === [ ERROR ] Took 1.17 seconds ===

I'm guessing it is a bad combination of build flags, but I can't figure out what would be wrong with them. Shouldn't -lcstd++ cover the string header? Any ideas?

Thanks for your time,
Arjen

valeros, 2015-08-25:
Looks like toolchain was builded without `disable-__cxa_atexit` option.
Could you try to declare the following global variable in your program:
```
#include <string>

int main(void){
    std::string str;
    return 0;
}

void* __dso_handle;
```

apvanzanten, 2015-08-25:
That does indeed work. Thanks!

So is this a problem in just the toolchain installation on my pc or in platformio in general? I'd rather fix the toolchain itself than work around it in code.

valeros, 2015-08-25:
The problem is in platformio toolchain package, we will try update it as soon as possible.
Currently, you can use the temporary solution in my previous comment.

apvanzanten, 2015-08-25:
Okay. That clears it up. Thanks for your help!

closed 2015-08-25 by apvanzanten