platformio/platformio-core #279: error: option --single-version-externally-managed not recognized

brakmic, 2015-08-28:
Hi,

I get an installation error under **Windows 10 x64**. :sob: 

```shell
D:\src\plarformio>pip install -U pip setuptools scons
Requirement already up-to-date: pip in c:\bin\python27\lib\site-packages
Requirement already up-to-date: setuptools in c:\bin\python27\lib\site-packages
Collecting scons
  Using cached scons-2.3.6.tar.gz
Installing collected packages: scons
  Running setup.py install for scons
    Complete output from command C:\bin\Python27\python.exe -c "import setuptools, tokenize;__file__='c:\\users\\harris\\appdata\\local\\
temp\\pip-build-dil9zr\\scons\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, '
exec'))" install --record c:\users\harris\appdata\local\temp\pip-35frnt-record\install-record.txt --single-version-externally-managed --c
ompile:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: option --single-version-externally-managed not recognized

    ----------------------------------------
Command "C:\bin\Python27\python.exe -c "import setuptools, tokenize;__file__='c:\\users\\harris\\appdata\\local\\temp\\pip-build-dil9zr\\
scons\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --recor
d c:\users\harris\appdata\local\temp\pip-35frnt-record\install-record.txt --single-version-externally-managed --compile" failed with erro
r code 1 in c:\users\harris\appdata\local\temp\pip-build-dil9zr\scons
```

According to <a href="http://stackoverflow.com/questions/14296531/what-does-error-option-single-version-externally-managed-not-recognized-ind" target="_blank">this SO thread</a> the problem disappears by using the flag "--egg". 

However, I know only very little about python and its environment and wouldn't say that this is a "proper" solution to this problem but at least it worked for me. Maybe there are some of you with similar problems. Just in the case of a more general problem it would be nice to put it somewhere in the docs.

Here's the command I used to install platformio.

```code
pip install -U pip setuptools --egg scons
```

And here's the proof :smile: 

```shell

D:\src\plarformio>pip install -U pip setuptools --egg scons
Requirement already up-to-date: pip in c:\bin\python27\lib\site-packages
Requirement already up-to-date: setuptools in c:\bin\python27\lib\site-packages
Collecting scons
  Using cached scons-2.3.6.tar.gz
Installing collected packages: scons
  Running setup.py install for scons
Successfully installed scons
```

Best regards,
Harris

ivankravets, 2015-08-28:
Thanks, sorry! I've just changed these lines in documentation. Firstly, need to upgrade `pip` and `setuptools` to the latest version and then install `scons` and `platformio`.

You won't need to install `scons` when PlatformIO 2.3.0 will be released.

brakmic, 2015-08-28:
Many thanks :+1:

ivankravets, 2015-08-28:
Which source did you use when install PlatformIO? Home site or documentation?

See updates: http://docs.platformio.org/en/latest/installation.html#python-package-manager

brakmic, 2015-08-28:
I used the tutorial from homesite.

ivankravets, 2015-08-28:
Thanks. The site has been updated too http://platformio.org/#!/get-started

sarfata, 2015-09-05:
I am still getting the same error. I can install scons with `--egg` but `pip install -U platformio` tries to re-download it and fails:

````
$ pip install --egg scons
Collecting scons
  Using cached scons-2.3.6.tar.gz
Installing collected packages: scons
  Running setup.py install for scons
Successfully installed scons
$ pip install platformio
Collecting platformio
  Using cached platformio-2.3.0.tar.gz
Requirement already satisfied (use --upgrade to upgrade): bottle in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): click>=3.2 in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): lockfile in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): pyserial in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.0 in ./.env/lib/python2.7/site-packages (from platformio)
Collecting SCons (from platformio)
  Using cached scons-2.3.6.tar.gz
Installing collected packages: SCons, platformio
  Running setup.py install for SCons
    Complete output from command /Users/thomas/Desktop/teensy/.env/bin/python -c "import setuptools, tokenize;__file__='/private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-yl0mcO/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-4WrOYc-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/thomas/Desktop/teensy/.env/include/site/python2.7/SCons:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: option --single-version-externally-managed not recognized

    ----------------------------------------
Command "/Users/thomas/Desktop/teensy/.env/bin/python -c "import setuptools, tokenize;__file__='/private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-yl0mcO/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-4WrOYc-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/thomas/Desktop/teensy/.env/include/site/python2.7/SCons" failed with error code 1 in /private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-yl0mcO/SCons
````

You said earlier that scons would not be required anymore in 2.3.0 but it is still in your `requirements.txt`.

ivankravets, 2015-09-06:
Don't install packages using `--egg`. You won't have ability to uninstall them properly later. 

1. Have you had install PlatformIO earlier?
2. You should upgrade `pip` to the latest version:
```
pip install -U pip setuptools

# then
pip install -U platformio
```

sarfata, 2015-09-06:
Yes I had platformio before and it was working. I started having problems when I tried to update yesterday.

Pip and setuptools are already up to date:
````
$ pip install -U pip setuptools
/Users/thomas/Desktop/teensy/.env/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
Requirement already up-to-date: pip in ./.env/lib/python2.7/site-packages
Collecting setuptools
  Downloading setuptools-18.3-py2.py3-none-any.whl (461kB)
    100% |████████████████████████████████| 462kB 923kB/s
Installing collected packages: setuptools
  Found existing installation: setuptools 18.2
    Uninstalling setuptools-18.2:
      Successfully uninstalled setuptools-18.2
Successfully installed setuptools-18.3
````

I still get the same problem when trying to install platformio:
````
$ pip install -U platformio
Collecting platformio
/Users/thomas/Desktop/teensy/.env/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:90: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. For more information, see https://urllib3.readthedocs.org/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Downloading platformio-2.3.1.tar.gz (76kB)
    100% |████████████████████████████████| 77kB 1.3MB/s
Requirement already up-to-date: bottle in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: click>=3.2 in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: lockfile in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: pyserial in ./.env/lib/python2.7/site-packages (from platformio)
Requirement already up-to-date: requests>=2.4.0 in ./.env/lib/python2.7/site-packages (from platformio)
Collecting SCons (from platformio)
  Using cached scons-2.3.6.tar.gz
Installing collected packages: SCons, platformio
  Running setup.py install for SCons
    Complete output from command /Users/thomas/Desktop/teensy/.env/bin/python -c "import setuptools, tokenize;__file__='/private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-cE50wR/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-UPVHUi-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/thomas/Desktop/teensy/.env/include/site/python2.7/SCons:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help

    error: option --single-version-externally-managed not recognized

    ----------------------------------------
Command "/Users/thomas/Desktop/teensy/.env/bin/python -c "import setuptools, tokenize;__file__='/private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-cE50wR/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-UPVHUi-record/install-record.txt --single-version-externally-managed --compile --install-headers /Users/thomas/Desktop/teensy/.env/include/site/python2.7/SCons" failed with error code 1 in /private/var/folders/xr/w6jn3gp92k78z7282m27v_1c0000gn/T/pip-build-cE50wR/SCons
````

ivankravets, 2015-09-06:
@sarfata please give me an output of this command:
```
pip list
```

cqql, 2015-09-08:
I experience the same problem with `pip 7.1.2`.

```sh
$ python -V
Python 2.7.9

$ pip list
bottle (0.12.8)
click (5.1)
lockfile (0.10.2)
pip (7.1.2)
pyserial (2.7)
requests (2.7.0)
setuptools (18.3.1)

$ pip install platformio
Collecting platformio
  Using cached platformio-2.3.1.tar.gz
Requirement already satisfied (use --upgrade to upgrade): bottle in /home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): click>=3.2 in /home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): lockfile in /home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): pyserial in /home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages (from platformio)
Requirement already satisfied (use --upgrade to upgrade): requests>=2.4.0 in /home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages (from platformio)
Collecting SCons (from platformio)
  Using cached scons-2.3.6.tar.gz
Installing collected packages: SCons, platformio
  Running setup.py install for SCons
    Complete output from command /home/cqql/.pyenv/versions/qopter-fw/bin/python2.7 -c "import setuptools, tokenize;__file__='/tmp/pip-build-WS8rhA/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-3IBDCV-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/cqql/.pyenv/versions/qopter-fw/include/site/python2.7/SCons:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help
    
    error: option --single-version-externally-managed not recognized
    
    ----------------------------------------
Command "/home/cqql/.pyenv/versions/qopter-fw/bin/python2.7 -c "import setuptools, tokenize;__file__='/tmp/pip-build-WS8rhA/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-3IBDCV-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/cqql/.pyenv/versions/qopter-fw/include/site/python2.7/SCons" failed with error code 1 in /tmp/pip-build-WS8rhA/SCons
```

ivankravets, 2015-09-08:
It seems that I've found where this issue is hidden. Give me please a whole GLOBAL LIST with installed packages(not from `virtualenv`). If one package depends on pip<7, pip will use it for the rest packages. New version of PlatformIO requires pip>=7 and new setuptools. I've just reproduced this issue in my virtual machine (Linux Ubuntu). Here `virtualenv` package is used old version of pip/setuptools. The 
```
[sudo] pip install -U setuptools
```
 resolved my issue.

Do you have installed `virtualenv` package?

cqql, 2015-09-08:
Yes, I am using [pyenv-virtualenv](https://github.com/yyuu/pyenv-virtualenv). I will uninstall all other packages and try it again without a virtualenv.

ivankravets, 2015-09-08:
Just upgrade GLOBAL packages to the latest versions:
```
[sudo] pip install -U pip setuptools virtualenv
```

cqql, 2015-09-08:
The problem persists outside any virtualenv. [log](https://gist.github.com/cqql/c4ce73b297a78b8c8401). And `virtualenv` is not even installed now.

ivankravets, 2015-09-08:
@cqql please try development build:
```shell
pip install https://github.com/platformio/platformio/archive/feature/issue-279.zip
```

cqql, 2015-09-08:
The error persists ([log](https://gist.github.com/cqql/31a10bb6cba569a79eb8)). Did you mean to try the installation twice? The quotes confused me. Trying it twice did not help.

I put some print statements in my `pip` installation and verified, that it is indeed running `7.1.2` and it passes `--single-version-externally-managed` to the `setup.py` every time `--egg` is not given. But somehow SCons' `setup.py` does not understand that option.

ivankravets, 2015-09-08:
@cqql Thanks that help me to resolve this issue. I can switch to previous logic (PlatformIO <2.3), where `scons` has installed via `pip install --egg scons`, but it isn't good solution. SCons team works on it. This isn't problem of PlatformIO. PlatformIO depends on SCons. The installation process of SCons was fixed in `pip>7` and `setuptools>=16`.

I revert code to previous logic (based on `--egg`) if we will not found common solution. Currently, only Linux users have these problems. 

Please try this build https://github.com/platformio/platformio/commit/3189a9de7654e2ae85a8f313608d3fc54fda9a8b:
```
pip install https://github.com/platformio/platformio/archive/feature/issue-279.zip
```

cqql, 2015-09-08:
I do not like scons. This is the second scons related issue I ran into (the other happened several months ago) and both times it took both of us a few hours to sort it out. 

I tried installing the new build, but to not avail. It is still the same error. I instrumented my pip again with some print statements in `req/req_install.py` right before the command is called to check the versions

```python
with indent_log():
    import setuptools
    import pip
    print(setuptools.__version__, pip.__version__)          
    call_subprocess( 
        install_args + install_options,
        cwd=self.source_dir,
        show_stdout=False,
    )
```
and the versions are
```sh
Installing collected packages: SCons, platformio
  Running setup.py install for SCons
('18.3.1', '7.1.2')
    Complete output from command /home/cqql/.pyenv/versions/2.7.10/bin/python2.7 -c "import setuptools, tokenize;__file__='/tmp/pip-build-uzEFJJ/SCons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-3hpEM3-record/install-record.txt --single-version-externally-managed --compile:
```
So apparently it is still not working.

ivankravets, 2015-09-08:
Thanks. Try 3-rd build https://github.com/platformio/platformio/commit/07e49cd9e4839d77d00a6a20705472a57dd55dd8:
```
pip install https://github.com/platformio/platformio/archive/feature/issue-279.zip
```

SCons has been removed from dependencies. Now install it via:
```
pip install --egg SCons
```

cqql, 2015-09-08:
That works. Thanks a lot. But of course it would sure be nicer, if `scons` could be declared as a dependency. Do you have an idea, why it broke, although the `scons` developers already worked on this issue?

ivankravets, 2015-09-08:
> Do you have an idea, why it broke, although the scons developers already worked on this issue?

They have worked on it a few years... I don't know what SCons uses in the kernel, but it is intended to be installed via OS package managers(not via `pip`). I asked them a few weeks ago to update `scons` package in PyPi, because PyPi had contained the packages from 2013 year. Now we have the latest 2.3.6 in the PyPi.

---------

You machine looks good to learn this issue. If you are ready we can try to find solution. First, let's use `virtualenv`:
```shell
pip install -U pip setuptools virtualenv
cd /tmp
virtualenv piotest
source piotest/bin/active
pip list
pip install scons
```

Give me please a log of these commands.

cqql, 2015-09-08:
The log is
```sh
$ pip install -U pip setuptools virtualenv
Requirement already up-to-date: pip in ./.pyenv/versions/2.7.10/lib/python2.7/site-packages
Requirement already up-to-date: setuptools in ./.pyenv/versions/2.7.10/lib/python2.7/site-packages
Requirement already up-to-date: virtualenv in ./.pyenv/versions/2.7.10/lib/python2.7/site-packages
$ cd /tmp
$ virtualenv piotest
New python executable in piotest/bin/python2.7
Also creating executable in piotest/bin/python
Installing setuptools, pip, wheel...done.
$ source piotest/bin/active
source: no such file or directory: piotest/bin/active
$ pip list
bottle (0.12.8)
click (5.1)
lockfile (0.10.2)
pip (7.1.2)
platformio (2.3.2.dev0)
pyserial (2.7)
requests (2.7.0)
setuptools (18.3.1)
virtualenv (13.1.2)
$ pip install scons
Collecting scons
  Using cached scons-2.3.6.tar.gz
Installing collected packages: scons
  Running setup.py install for scons
    Complete output from command /home/cqql/.pyenv/versions/2.7.10/bin/python2.7 -c "import setuptools, tokenize;__file__='/tmp/pip-build-Zl14Pi/scons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-UVJrfN-record/install-record.txt --single-version-externally-managed --compile:
    usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
       or: -c --help [cmd1 cmd2 ...]
       or: -c --help-commands
       or: -c cmd --help
    
    error: option --single-version-externally-managed not recognized
    
    ----------------------------------------
Command "/home/cqql/.pyenv/versions/2.7.10/bin/python2.7 -c "import setuptools, tokenize;__file__='/tmp/pip-build-Zl14Pi/scons/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-UVJrfN-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Zl14Pi/scons
```

ivankravets, 2015-09-08:
Sorry as for mistake in command "source piotest/bin/active", it should be `source piotest/bin/activate`. Please update your comment above, because you are outside virtualenv. Thanks.

cqql, 2015-09-08:
Sorry, I did not proofread the script and just executed it with `(while read l; do echo \$ $l; eval $l; done) < tmp.sh > log 2>&1`. I have updated my previous comment and it works now. Time to investigate some more ;)

cqql, 2015-09-08:
I absolutely do not understand this. Now it works.
```sh
$ pyenv virtualenv pio
$ pyenv shell pio                                                                           
$ pip list
pip (7.1.2)
setuptools (18.2)
$ pip install scons
Collecting scons
Installing collected packages: scons
Successfully installed scons-2.3.6
$ pip install platformio
Collecting platformio
  Using cached platformio-2.3.1.tar.gz
Collecting bottle (from platformio)
  Using cached bottle-0.12.8.tar.gz
Collecting click>=3.2 (from platformio)
  Using cached click-5.1-py2.py3-none-any.whl
Collecting lockfile (from platformio)
  Using cached lockfile-0.10.2-py2-none-any.whl
Collecting pyserial (from platformio)
  Using cached pyserial-2.7.tar.gz
Collecting requests>=2.4.0 (from platformio)
  Using cached requests-2.7.0-py2.py3-none-any.whl
Requirement already satisfied (use --upgrade to upgrade): SCons in ./.pyenv/versions/pio/lib/python2.7/site-packages (from platformio)
Installing collected packages: bottle, click, lockfile, pyserial, requests, platformio
  Running setup.py install for bottle
  Running setup.py install for pyserial
  Running setup.py install for platformio
Successfully installed bottle-0.12.8 click-5.1 lockfile-0.10.2 platformio-2.3.1 pyserial-2.7 requests-2.7.0
$ python -V
Python 2.7.10
$ pip list
bottle (0.12.8)
click (5.1)
lockfile (0.10.2)
pip (7.1.2)
platformio (2.3.1)
pyserial (2.7)
requests (2.7.0)
scons (2.3.6)
setuptools (18.2)
```

ivankravets, 2015-09-08:
This is `virtualenv`... I had the same behaviour :( Strange behaviour.  I can add to existing installation `virtualenv`, but I needn't it. Other solution, to add "Warning" block where propose to update `pip install -U pip setuptools virtualenv` when `error: option --single-version-externally-managed not recognized` occurs.

WDYT?

P.S: Here we have another problem - `scons --version` doesn't work. However, it seems that I can fix it for `scon` from PlatformIO's side. Give me a few minutes.

cqql, 2015-09-08:
It works for me
```sh
$ scons --version
SCons by Steven Knight et al.:
	script: v2.3.6.rel_2.3.5:3347:d31d5a4e74b6[MODIFIED], 2015/07/31 14:36:10, by bdbaddog on hpmicrodog
	engine: v2.3.6.rel_2.3.5:3347:d31d5a4e74b6[MODIFIED], 2015/07/31 14:36:10, by bdbaddog on hpmicrodog
	engine path: ['/home/cqql/.pyenv/versions/qopter-fw/lib/python2.7/site-packages/scons-2.3.6/SCons']
Copyright (c) 2001 - 2015 The SCons Foundation
```

Do you think, it was caused by an outdated `virtualenv`? I cannot believe that, because I tried it with a fresh install of `virtualenv` and I even cleaned all my python installations and started anew with `2.7.10`. The only thing that changed from 2 hours ago is, that my laptop was suspended for 2 hours.

ivankravets, 2015-09-08:
Try to downgrade `virtualenv`:
```
pip uninstall scons
pip install "virtualenv==1.3"
pip list
pip install scons
```

Have you reproduce issue again?

P.S: `scons` works for you because you use `pyenv`. However, it doesn't work when is installed to `/usr/local/*`.
P.S.S: I see that in previous reports you have used Python 2.7.9. An in the latest, you used Python 2.7.10. Could I ask you to try with Python 2.7.9?

sarfata, 2015-09-08:
I see we are making lots of progress. Just wanted to say that I am on Mac OS so this is not a Linux only problem. And I do have pip >7:

````
# pip --version
pip 7.1.2 from /Users/thomas/Desktop/teensy/.env/lib/python2.7/site-packages (python 2.7)
(.env)thomas@dapdap-mbp ~/work-perso/wearaboard 11:20
$ deactivate
thomas@dapdap-mbp ~/work-perso/wearaboard 11:21
$ pip --version
pip 7.1.2 from /Library/Python/2.7/site-packages (python 2.7)
````

Your test also worked for me:
````
$ sudo pip install -U pip setuptools virtualenv
Requirement already up-to-date: pip in /Library/Python/2.7/site-packages
Collecting setuptools
  Downloading setuptools-18.3.1-py2.py3-none-any.whl (462kB)
    100% |████████████████████████████████| 462kB 1.1MB/s
Collecting virtualenv
  Downloading virtualenv-13.1.2-py2.py3-none-any.whl (1.7MB)
    100% |████████████████████████████████| 1.7MB 314kB/s
Installing collected packages: setuptools, virtualenv
  Found existing installation: setuptools 18.2
    Uninstalling setuptools-18.2:
      Successfully uninstalled setuptools-18.2
  Found existing installation: virtualenv 1.11.4
    Uninstalling virtualenv-1.11.4:
      Successfully uninstalled virtualenv-1.11.4
Successfully installed setuptools-18.3.1 virtualenv-13.1.2
thomas@dapdap-mbp ~/Desktop/teensy 11:23 10110
$ virtualenv piotest
New python executable in piotest/bin/python
Installing setuptools, pip, wheel...done.
thomas@dapdap-mbp ~/Desktop/teensy 11:23 10111
$ source piotest/bin/activate
(piotest)thomas@dapdap-mbp ~/Desktop/teensy 11:23 10112
$ pip list
pip (7.1.2)
setuptools (18.2)
wheel (0.24.0)
(piotest)thomas@dapdap-mbp ~/Desktop/teensy 11:23 10113
$ pip install scons
Collecting scons
  Using cached scons-2.3.6.tar.gz
Building wheels for collected packages: scons
  Running setup.py bdist_wheel for scons
  Stored in directory: /Users/thomas/Library/Caches/pip/wheels/83/2c/cc/77439553dec9a4726e142c05c621cb49d1c2f774662f722f8a
Successfully built scons
Installing collected packages: scons
Successfully installed scons-2.3.6
````

sarfata, 2015-09-08:
Installing your 3rd build (07e49cd) also worked.

ivankravets, 2015-09-08:
@sarfata could you try to install directly without `virtualenv`?
```
pip install platformio
```

cqql, 2015-09-08:
It just works in every combination. I also tried to do it without virtualenvs, but as long as `pip` and `setuptools` are updated, everything works. However, why did it not work this afternoon?

Here are some [logs](https://gist.github.com/anonymous/232efdff2c446a55ab92) though they are not really interesting.

sarfata, 2015-09-09:
@ivankravets It just worked now (including installing Scons). I am not sure what changed before ... Sorry about that and thanks for your support!

ivankravets, 2015-09-09:
@cqql @sarfata does PlatformIO work entirely? Can you build any projects via `platformio run`?

sarfata, 2015-09-09:
@ivankravets yes it does.

cqql, 2015-09-10:
`platformio run` just installed a program on my arduino without problems.

ivankravets, 2015-09-10:
I reproduced this issue again and temporary solution is described in the original https://github.com/platformio/platformio/issues/279#issue-103778093

It looks stupid/magic for me, but I don't have other options. If someone will find more accurate solution, please report here.

PlatformIO 2.3.2 will be released today. 

This "known issue" has been added to main docs:

* http://docs.platformio.org/en/latest/installation.html#python-package-manager
* http://docs.platformio.org/en/latest/faq.html#faq-troubleshooting-sconssingverextmanaged

closed 2015-09-10 by ivankravets

robotdad, 2015-09-14:
I am running into this issue with PlatformIO 2.3.2 on Windows 10. However, pip won't install scons either. I can install scons via their distribution using setup.py or --egg. Even after doing so I still can't install platformio due to this error. Pip list does not show scons present. Scons --version reports 2.3.6.

robotdad, 2015-09-14:
I was able to install using the get-platformio.py script.

ivankravets, 2015-09-14:
@robotdad thank for the report! Have you used PlatformIO before 2.3 release?

robotdad, 2015-09-14:
Yes, I have 2.2 on another machine which is working fine but I'm getting this error on trying to upgrade. I can't recall if it was updated from previous versions or not.

ivankravets, 2015-09-14:
@robotdad give me please an output of this command (where `platformio upgrade` command doesn't work):
```
pip list
```

robotdad, 2015-09-14:
bottle (0.12.8)
click (5.1)
colorama (0.3.3)
lockfile (0.10.2)
pip (7.1.2)
platformio (2.2.2)
pyserial (2.7)
requests (2.7.0)
setuptools (18.3.1)
virtualenv (13.1.2)

Also, scons --version reports 2.3.6

hoosierEE, 2015-09-22:
Thanks @brakmic for the solution at the top of this page.  I dusted off a computer with an old version of platformio on it, and following the prompts caused it to become broken.  Your solution fixed it for me.

pastcompute, 2015-10-08:
I had this same problem, under Xubuntu 14.04 LTS. Except nothing I tried from above would get past the external version error.
In the end I had to do the following:
* removed everything under /usr/local/.../python... including platformio and all the other pip installed python packages
* uninstall the Ubuntu version of pip
* install pip as per https://pip.pypa.io/en/stable/installing/#python-os-support
After this, `pip install platformio` worked again

ivankravets, 2015-10-08:
@pastcompute, thank you, I added your comment to "solutions" list above.

danielricardogm, 2017-04-18:
Muchas gracias!!!

commit a815aa15 (2015-08-28): Fix installation issue with pip // Resolve #279
commit c72bf9ea (2015-09-10): Found solution for "pip/scons error: option –single-version-externally-managed not recognized" // Resolve #279