platformio/platformio-core #280: Eclipse doesn't know about __LONG_MAX__

gslender, 2015-08-30:
Error is `Symbol '__LONG_MAX__' could not be resolved` which tends to suggest the compiler symbols are unknown or missing. The build compiles fine, but Eclipse has lots of errors for anything that uses the Limits.h includes

ivankravets, 2015-08-30:
What is type of platform?

gslender, 2015-08-30:
Arduino Uno

ivankravets, 2015-08-31:
Do you use the latest `develop` build of PlatformIO? http://docs.platformio.org/en/latest/installation.html#development-version

Looks like this issue was fixed in https://github.com/platformio/platformio/issues/277

ivankravets, 2015-08-31:
See `ag` output:
```
➜  toolchain-atmelavr  ag __LONG_MAX__
lib/gcc/avr/4.8.1/include-fixed/limits.h
96:#define LONG_MAX __LONG_MAX__
```

Don't forget to re-init (`platformio init -b uno --ide eclipse`) your Eclipse-based project. It should add `../include-fixed` to `INCLUDES` list.

gslender, 2015-08-31:
No - this is missing symbols from the Eclipse project - these would be in the GCC compiler (so it compiles ok with Platformio), but the code completion and error checking fails in Eclipse because there is missing compiler symbols for all the base types etc.

ivankravets, 2015-08-31:
Hm... I don't see any macros in `AVR` related  packages/tool chains. See `ag` search results within all PlatformIO packages:
```
➜  packages  ag __LONG_MAX__
toolchain-atmelavr/lib/gcc/avr/4.8.1/include-fixed/limits.h
96:#define LONG_MAX __LONG_MAX__

toolchain-atmelavr/lib/gcc/avr/4.8.1/install-tools/include/limits.h
96:#define LONG_MAX __LONG_MAX__

toolchain-atmelavr/share/info/cpp.info
1978:`__LONG_MAX__'

toolchain-gccarmlinuxgnueabi/arm-willtm-linux-gnueabi/include/c++/4.8.2/limits
1126:      min() _GLIBCXX_USE_NOEXCEPT { return -__LONG_MAX__ - 1; }
1129:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__; }
1196:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__ * 2UL + 1; }

toolchain-gccarmlinuxgnueabi/arm-willtm-linux-gnueabi/sysroot/usr/include/bits/xopen_lim.h
127:#elif defined __LONG_MAX__
128:# if __LONG_MAX__ == 2147483647

toolchain-gccarmlinuxgnueabi/lib/gcc/arm-willtm-linux-gnueabi/4.8.2/include-fixed/limits.h
131:#define LONG_MAX __LONG_MAX__

toolchain-gccarmlinuxgnueabi/lib/gcc/arm-willtm-linux-gnueabi/4.8.2/install-tools/include/limits.h
131:#define LONG_MAX __LONG_MAX__

toolchain-gccarmlinuxgnueabi/share/info/cpp.info
1978:`__LONG_MAX__'

toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/4.8.3/limits
1126:      min() _GLIBCXX_USE_NOEXCEPT { return -__LONG_MAX__ - 1; }
1129:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__; }
1196:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__ * 2UL + 1; }

toolchain-gccarmnoneeabi/arm-none-eabi/include/limits.h
79:#   ifndef __LONG_MAX__
81:#     define __LONG_MAX__ 9223372036854775807L
83:#     define __LONG_MAX__ 2147483647L
89:#   define LONG_MAX __LONG_MAX__

toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/config.h
197:#ifndef __LONG_MAX__
199:#  define __LONG_MAX__ LONG_MAX
203:#   define __LONG_MAX__ 9223372036854775807L
205:#   define __LONG_MAX__ 2147483647L

toolchain-gccarmnoneeabi/arm-none-eabi/include/sys/reent.h
20:#if __LONG_MAX__ == 2147483647L

toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/4.8.3/include-fixed/limits.h
131:#define LONG_MAX __LONG_MAX__

toolchain-gccarmnoneeabi/lib/gcc/arm-none-eabi/4.8.3/install-tools/include/limits.h
131:#define LONG_MAX __LONG_MAX__

toolchain-gccarmnoneeabi/share/doc/gcc-arm-none-eabi/html/cpp/Common-Predefined-Macros.html
220:     <br><dt><code>__SCHAR_MAX__</code><dt><code>__WCHAR_MAX__</code><dt><code>__SHRT_MAX__</code><dt><code>__INT_MAX__</code><dt><code>__LONG_MAX__</code><dt><code>__LONG_LONG_MAX__</code><dt><code>__WINT_MAX__</code><dt><code>__SIZE_MAX__</code><dt><code>__PTRDIFF_MAX__</code><dt><code>__INTMAX_MAX__</code><dt><code>__UINTMAX_MAX__</code><dt><code>__SIG_ATOMIC_MAX__</code><dt><code>__INT8_MAX__</code><dt><code>__INT16_MAX__</code><dt><code>__INT32_MAX__</code><dt><code>__INT64_MAX__</code><dt><code>__UINT8_MAX__</code><dt><code>__UINT16_MAX__</code><dt><code>__UINT32_MAX__</code><dt><code>__UINT64_MAX__</code><dt><code>__INT_LEAST8_MAX__</code><dt><code>__INT_LEAST16_MAX__</code><dt><code>__INT_LEAST32_MAX__</code><dt><code>__INT_LEAST64_MAX__</code><dt><code>__UINT_LEAST8_MAX__</code><dt><code>__UINT_LEAST16_MAX__</code><dt><code>__UINT_LEAST32_MAX__</code><dt><code>__UINT_LEAST64_MAX__</code><dt><code>__INT_FAST8_MAX__</code><dt><code>__INT_FAST16_MAX__</code><dt><code>__INT_FAST32_MAX__</code><dt><code>__INT_FAST64_MAX__</code><dt><code>__UINT_FAST8_MAX__</code><dt><code>__UINT_FAST16_MAX__</code><dt><code>__UINT_FAST32_MAX__</code><dt><code>__UINT_FAST64_MAX__</code><dt><code>__INTPTR_MAX__</code><dt><code>__UINTPTR_MAX__</code><dt><code>__WCHAR_MIN__</code><dt><code>__WINT_MIN__</code><dt><code>__SIG_ATOMIC_MIN__</code><dd>Defined to the maximum value of the <code>signed char</code>, <code>wchar_t</code>,

toolchain-gccarmnoneeabi/share/doc/gcc-arm-none-eabi/info/cpp.info
1977:`__LONG_MAX__'

toolchain-gcclinux32/include/bits/xopen_lim.h
128:#elif defined __LONG_MAX__
129:# if __LONG_MAX__ == 2147483647

toolchain-gcclinux32/include/c++/4.8.1/limits
1126:      min() _GLIBCXX_USE_NOEXCEPT { return -__LONG_MAX__ - 1; }
1129:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__; }
1196:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__ * 2UL + 1; }
[... 59 lines cut ...]
96:#define LONG_MAX __LONG_MAX__

toolchain-xtensa/lib/gcc/xtensa-lx106-elf/4.8.2/install-tools/include/limits.h
96:#define LONG_MAX __LONG_MAX__

toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/limits
1126:      min() _GLIBCXX_USE_NOEXCEPT { return -__LONG_MAX__ - 1; }
1129:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__; }
1196:      max() _GLIBCXX_USE_NOEXCEPT { return __LONG_MAX__ * 2UL + 1; }

toolchain-xtensa/xtensa-lx106-elf/include/limits.h
79:#   ifndef __LONG_MAX__
81:#     define __LONG_MAX__ 9223372036854775807L
83:#     define __LONG_MAX__ 2147483647L
89:#   define LONG_MAX __LONG_MAX__

toolchain-xtensa/xtensa-lx106-elf/include/sys/config.h
187:#ifndef __LONG_MAX__
189:#  define __LONG_MAX__ LONG_MAX
193:#   define __LONG_MAX__ 9223372036854775807L
195:#   define __LONG_MAX__ 2147483647L

toolchain-xtensa/xtensa-lx106-elf/include/sys/reent.h
19:#if __LONG_MAX__ == 2147483647L
```

ivankravets, 2015-08-31:
It seems that `__LONG_MAX__` is compiled into toolchain binaries. Nevertheless, Eclipse IDE allows to add own symbols. Please add symbol `__LONG_MAX__` with value `2147483647L`. See instruction from http://www.ikravets.com/computer-life/programming/2014/06/20/building-and-debugging-atmel-avr-arduino-based-project-using-eclipse-ideplatformio

[image]

gslender, 2015-08-31:
Ok, so this seems still to be an Eclipse project setting issue - read this website http://www.eclipse.org/community/eclipse_newsletter/2013/october/article4.php

ivankravets, 2015-08-31:
Hmm... This is very obsolete article. Do you have any problems when set this symbol to IDE?

gslender, 2015-08-31:
Nah, haven't got this solved yet, but figured out that this command exposes the inbuilt avr-g++ symbols... so something about Eclipse's compiler parser etc does it...

```~/.platformio/packages/toolchain-atmelavr/bin/avr-g++ -E -P -v -dD main.cpp ```

My guess is that you can update the Eclipse project to have the correct Preprocessor and Include path setting using that web article I linked above. I think all the bits are there, just need to be put together correctly.

gslender, 2015-08-31:
Just trying to help you get the Eclipse integration 100% working... you've got the best Arduino IDE if you can smooth out these little bits. Obviously having the <limits.h> working with code analysis would be awesome... so close... nearly complete....

gslender, 2015-08-31:
ok, have figured out how to make this work... but not sure how to add to the default project settings...

Add CDT GCC Built-in Compiler Settings 

``` ${HOME}/.platformio/packages/toolchain-atmelavr/bin/avr-g++ ${FLAGS} -E -P -v -dD "${INPUTS}" ```

ivankravets, 2015-08-31:
Give me please PlatformIO-based project with "Eclipse IDE .cproject" file. I want to reproduce this issue. Thanks.

gslender, 2015-08-31:
Where do I send the zip file of the project?

Essentially it creates an entry in .cproject and also in .settings/language.settings.xml of the command above. This means Eclipse can execute the command for g++ that dumps the SYMBOLs and then use them in the inbuilt parser to code check everything. Neat hey!

gslender, 2015-08-31:
Have emailed zip file to your personal email account

ivankravets, 2015-08-31:
Send to me@ikravets.com. Thanks!

With best regards, Ivan Kravets
- "Creative comes from talent, and never from knowledge" (c)
_________________________
http://www.ikravets.com

ivankravets, 2015-09-04:
Please re-test with the latest development build.

Did it help you?

commit a032026f (2015-09-04): Significantly improve "Project Generator" // Resolve #280
closed 2015-09-04 by ivankravets