platformio/platformio-core #281: Add option to ignore update check when running

gslender, 2015-08-30:
When starting platformio command, it sits for 15 seconds or more checking for upgrades, but then fails with Failed to check for PlatformIO upgrades. Please check your Internet connection.

I'm behind a proxy and don't want PlatformIO to check, so can this be enabled/disabled by project ini and/or command line?

If that option exists, can you document it as I couldn't find it (not easily anyway).

Thanks

ivankravets, 2015-08-31:
Set these `check_*` settings to 1 year :) http://docs.platformio.org/en/latest/userguide/cmd_settings.html#check-platformio-interval

Like
```
platformio settings set check_platformio_interval 365
platformio settings set check_libraries_interval 365
platformio settings set check_platforms_interval 365
```

Did it help you?

gslender, 2015-09-01:
Yep, that's the right options... probably a bit coarse, but it does the job.

closed 2015-09-01 by gslender