platformio/platformio-core #282: support for RedBearLabs BLENano hardware revision 1.5

jrobeson, 2015-09-02:
Hardware revision 1.5 supports 32kb ram instead of the 16kb supported by 1.0. Does that mean a new board should be added called redBearLabBLENano15 ?

jrobeson, 2015-09-02:
It still doesn't show up on this list https://developer.mbed.org/platforms/?pvend=19 so it probably shouldn't be added until it does. I've asked the RedBearLabs folks about getting it added there.

valeros, 2015-09-03:
Hi @jrobeson
I think, you needn't add a new board. For now, the best way is to use `redBearLabBLENano` board with ld-script for 32 kbyte version:
`https://github.com/mbedmicro/mbed/blob/master/libraries/mbed/targets/cmsis/TARGET_NORDIC/TARGET_MCU_NRF51822/TOOLCHAIN_GCC_ARM/TARGET_MCU_NORDIC_32K/NRF51822.ld`

You can use `build_flags option` in your `platformio.ini` file:
```
[env:redBearLabBLENano]
platform = nordicnrf51
framework = mbed
board = redBearLabBLENano
build_flags = -Wl,-Tpath_to_script.ld
```
Is this acceptable for you?

jrobeson, 2015-09-03:
I don't think that's acceptable for a a user to have to specially configure any released board (unless other people have to do that). do any other boards have to do that? I see multiple versions of teensy for example.

ivankravets, 2015-09-05:
@jrobeson thanks! Please report us when this board will be added.

jrobeson, 2015-09-05:
I'd submit a PR myself if it's simply adding to the json file, as long as you suggest a name.

valeros, 2015-09-05:
It isn't so simple, so it would be better if you notify us when this board will be added.

rph8, 2015-09-06:
Why is adding that board so difficult?
Should the mbed framework not work with a lot of Cortex-M-based boards?

ivankravets, 2016-11-27:
Please check this issue with PlatformIO 3.1.

If it doesn't work, please open new issue in this repo https://github.com/platformio/platform-nordicnrf51/issues/new

closed 2016-11-27 by ivankravets