platformio/platformio-core #283: platformio run command blocks platformio init completion in subprocess

jrobeson, 2015-09-06:
Currently if you run `platformio init --ide=anything` on first run with a non-existent `~/.platformio` it will not run to completion and simply hangs waiting for user response.   This prompt is never shown to the user due to subprocess execution here:  https://github.com/platformio/platformio/blob/develop/platformio/ide/projectgenerator.py#L55  

Here is the prompt:
`The platform 'nordicnrf51' has not been installed yet. Would you like to install it now? [y/N]:`.  

Is there any way we can avoid calling another command here and simply do whatever run does internally instead?

ivankravets, 2015-09-06:
See "Troubleshooting" section from installation page http://docs.platformio.org/en/latest/installation.html#troubleshooting

You have to use `--force` flag before each comman when ru PlatformIO from subprocess. Like, `platformio -f init --ide=anything`

ivankravets, 2015-09-06:
Don't forget upgrade (`platformio upgrade`) PlatformIO to 2.3.0, which was released yesterday. It has improved Project Generator for the multiple IDE.

jrobeson, 2015-09-06:
this is in your code, so I don't see how you can close it. NOTE: I'm using what is in 2.3.0 by using the develop branch

ivankravets, 2015-09-06:
Thanks! I'll fix it. Looks like critical bug, maybe I'll release 2.3.1 today.

ivankravets, 2015-09-06:
Thanks a lot for this report!!! :+1: 

PlatformIO 2.3.1 is released!

commit 0311418b (2015-09-06): Fix critical issue when `platformio init --ide`__ command hangs PlatformIO // Resolve #283
closed 2015-09-06 by ivankravets