platformio/platformio-core #285: Add as platfrom STM8

namoaton, 2015-09-08:
Board  STM8L Discovery
compiler SDCC - Small Device C Compiler

xorbit, 2016-06-20:
I would love to have this added as well

bergernetch, 2016-12-07:
Would be great as there now is an ESP board with an STM8: ESP-14

jourjine, 2016-12-19:
+1, please  add support for stm8L family

javigf, 2017-01-04:
+2, please add support for stm8L family

tatulaiot, 2017-11-01:
Not yet in Platformio??

kissste, 2017-11-03:
nope

xorbit, 2017-12-28:
In the mean time, there's this: https://stm8-binutils-gdb.sourceforge.io/
It seems to work well, and maybe Platform.io can use it as a base for integration?

dontsovcmc, 2018-09-20:
As I see STM8 supported by [SDCC - Small Device C Compiler](http://sdcc.sourceforge.net/) https://www.fischl.de/stm8/stm8svl_discovery_linux/

harji2130, 2018-10-02:
+1 for STM8L

DrNorfolk, 2018-10-21:
I will be glad to see support of STM8!