platformio/platformio-core #287: Upload and programming a Leonardo board fails

gslender, 2015-09-10:
I'm not sure if this has been specifically tested, but on OS X, you cannot successfully program/upload to a Leonardo Arduino board. The board works fine with the Arduino IDE, but fails every time using Platformio!

The first issue is that it doesn't sequence the port detection correctly/simply (like Arduino IDE) but you can mostly get around that (by timing the reset of the board to have the port show up and then execute the upload command, but it is very hard and painful).

The second issue is that it fails checksum for some unknown reason.

```
avrdude: Device signature = 0x1e9587
avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as CB
avrdude: reading input file ".pioenvs/leonardo/firmware.hex"
avrdude: writing flash (32156 bytes):

Writing | ################################################## | 100% 2.23s

avrdude: 32156 bytes of flash written
avrdude: verifying flash memory against .pioenvs/leonardo/firmware.hex:
avrdude: load data flash data from input file .pioenvs/leonardo/firmware.hex:
avrdude: input file .pioenvs/leonardo/firmware.hex contains 32156 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.29s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x7000
0x55 != 0xc3
avrdude: verification error; content mismatch

avrdude: safemode: lfuse reads as FF
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as CB
avrdude: safemode: Fuses OK (H:CB, E:D8, L:FF)

avrdude done.  Thank you.

scons: *** [upload] Error 1
```

ivankravets, 2015-09-10:
Do you have problem when specify `--upload-port` option for `platformio run` command?

P.S: We have this board in our team (@valeros has it). However, he uses it under Windows and doesn't have any problems.

ivankravets, 2015-09-10:
Could you try this new build?
```
pip install https://github.com/platformio/platformio/archive/feature/issue-287.zip
```

ivankravets, 2015-09-10:
I merged it to main branch. Use update command above
```
pip install -U https://github.com/platformio/platformio/archive/develop.zip
```

gslender, 2015-09-10:
Yep. All sorted. Much better upload now. The 2nd issue was due to me hitting the leonardo flash limit. I forgot about the huge 4K boot loader and the program hit the 0x7000 memory limit.

closed 2015-09-10 by gslender

ivankravets, 2015-09-11:
Please upgrade to stable `2.3.2` where this issue is fixed:
```
platformio upgrade
```

Thanks.

commit b60c761c (2015-09-11): Fix firmware uploading to Arduino Leonardo board using Mac OS // Resolve #287