platformio/platformio-core #288: platformio unresponsive

Protoneer, 2015-09-20:
My environment:
* Win7
* Python 2.7
* Platformio 2.3.1/2
* Board ESP01
* Arduino Framework Library
* installed using "pip install platfomrio"

I ran "platformio init --board=esp01" and it started downloading all the dependencies. Did a "platfomrio run" it started building and cam back with a message saying that I was missing an H file. All good!!!

Then I looked into using lib-dir to point to my Arduino libraries and that is when it stopped responding.

Running platformlib returns the usual help text. As soon as I try paramters it just sits there and will only stop when I press control-c.

Have tried to reintalling and going back one version. Have restarted my PC and I am out of idea's... Is there some way to switch on verbose messages or is there a log somewhere that I can check for more details. 

I have also tried using the force parameter with no luck? Any ideas?

Would really love to make this work cause I can see there went lots of effort into this project.

ivankravets, 2015-09-21:
Thanks, I'm going to fix it today and to propose for testing.

Protoneer, 2015-09-23:
Thanks @ivankravets . I am willing to help with testing and providing logs if needed.

ivankravets, 2015-09-23:
Please try development version where this issue should be fixed http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-09-23 by ivankravets

Protoneer, 2015-09-23:
@ivankravets that seems to fix the issue.  Can you confirm where the board library is being downloaded from? From "platformio run" I can see that it is complaining about code in the board libaries that does not exists.(Recent additions) 

I know the link to the stable version for ESP8266/Arduino changed recently to : http://arduino.esp8266.com/stable/package_esp8266com_index.json

ivankravets, 2015-09-24:
@valeros please verify that we use the latest stable version of Espressif framework

Protoneer, 2015-09-28:
Hi @ivankravets , any luck with this? If you give me some directions on where to check I can have a look? (Thanks.)

ivankravets, 2015-09-28:
Which problems do yo have with this issue? I see that it's closed and you reported that all work fine.

Protoneer, 2015-09-28:
If its using the latest stable version of Espressif framework?

ivankravets, 2015-09-28:
Please open separate issue with it for @valeros

commit 46d1c2c8 (2015-09-23): Fix broken lock file for "appstate" storage // Resolve #288