platformio/platformio-core #291: Archlinux: libncurses.so.5: cannot open shared object file

james147, 2015-09-21:
Archlinux is now using ncurses6 which is causing the following error in platformio while trying to upload to the arduino platform:

```
avrdude: error while loading shared libraries: libncurses.so.5: cannot open shared object file: No such file or directory
```

The archlinux version of avrdude works with the quick workaround:

```bash
rm ~/.platformio/packages/tool-avrdude/avrdude
ln -s /usr/bin/avrdude ~/.platformio/packages/tool-avrdude/avrdude
```

However, the tools that platformio downloads should come with the dependencies that they require to run  or alternatively make use of the system tools when they are available.

ivankravets, 2015-09-22:
Could you try to install `lib32-ncurses` package?

james147, 2015-09-22:
No difference, it is version 6, when then downloaded version is looking for
version 5.

valeros, 2015-09-22:
Hi @james147 
Do you have the same issue when you upload sketch through Arduino IDE?

james147, 2015-09-22:
Not with the one in the [Arch User Repo](https://aur.archlinux.org/packages/arduino/), but it appears to replace the arduino provided avrdude with the working system one:

```bash
  # fix avrdude
  rm -f "${pkgdir}/usr/share/arduino/hardware/tools/avr/bin/avrdude"{,_bin}
  ln -s /usr/bin/avrdude "${pkgdir}/usr/share/arduino/hardware/tools/avr/bin/avrdude"
```
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=arduino

So I assume that the official arduino build of avrdude does have the same problem.

ivankravets, 2015-09-23:
@james147 I've just added to FAQ page info about the solution for this issue. If we recompile `avrdude` with the latest `libncurses.so.6`, it will not work with the rest Linux distributives (Debian, Ubuntu, etc.).

closed 2015-09-23 by ivankravets

cqql, 2015-10-18:
I had to replace the `avrdude` installed by platformio with the system one, so

```sh
AVRDUDE=~/.platformio/packages/tool-avrdude/avrdude
rm $AVRDUDE
ln -s /usr/bin/avrdude $AVRDUDE
```

albfan, 2015-11-15:
You can use AUR package or solutions commented there

https://aur.archlinux.org/packages/arduino/

milovanderlinden, 2016-03-04:
Thanks @cqql your solution works like a charm for me.

darkwingduck42, 2016-05-06:
@ivankravets Thats the reason why linux distributions have a package manager. So the obvious solution would be not to ship binaries at all!

darkwingduck42, 2016-05-06:
@ivankravets Thats the reason why linux distributions have a package manager. Either you ship the binary AND its dependencies or you let package manager do its work. Everything between is just a mess. So the obvious solution would be not to ship binaries at all!

Hamza5, 2016-10-03:
In addition to what @mdaffin posted in the first comment, I had to link the `avrdude` of the `toolchain-atmelavr` to the new `avrdude` also, so I had to add these commands :

    rm ~/.platformio/packages/toolchain-atmelavr/bin/avrdude
    ln -s ~/.platformio/packages/tool-avrdude/avrdude ~/.platformio/packages/toolchain-atmelavr/bin/avrdude

jwhendy, 2016-11-02:
Confirming that @Hamza5 's addition did it for me. I ended up symlinking both locations, but it only worked upon the second (in `toolchain-atmelavr` as well as `tool-avrdude`.

orzel, 2019-12-29:
I'm trying platformio on gentoo. Most 'packages' are broken in the same way:

piogdb: error while loading shared libraries: libncursesw.so.5: cannot open shared object file: No such file or directory

or

% ldd .platformio//packages/toolchain-gd32v/bin/riscv-nuclei-elf-gdb
        linux-vdso.so.1 (0x00007ffe46d73000)
        libncursesw.so.5 => not found
        libtinfo.so.5 => not found
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f67db6b4000)
        libexpat.so.1 => /usr/lib64/libexpat.so.1 (0x00007f67db676000)
        libmpfr.so.6 => /usr/lib64/libmpfr.so.6 (0x00007f67db5f6000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f67db4b5000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f67db2dc000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f67dbfb0000)
        libgmp.so.10 => /usr/lib64/libgmp.so.10 (0x00007f67db261000)

thoastbrot, 2020-02-20:
So that issue is part of the PIO FAQ... I can't really see a proper solution here. The AUR package is gone, the symlinks look like crude hacks...

waohwoah, 2020-07-11:
By downloading the **ncurses5-compat-libs** & **lib32-ncurses5-compat-libs** from AUR, this issue was solved for me.
Should work on other distros by downloading these same packages from their specific package managers.

commit ee7fe1fc (2015-09-23): Add to FAQ info about Archlinux issue with libncurses.so.5 // Resolve #291