platformio/platformio-core #294: Unable to successfully init->build project (CLion)

mseroczynski, 2015-09-28:
I'm trying to initialize project for CLion using ``yun`` board. After importing the project and using example copy pasted code (Blink) i'm unable to successfully build and upload it. I'm getting error log with "Board not recognized" message. (Meanwhile led on Arduino is blinking on uploading time, but only when i define "upload_port" to my /dev/cu.usbmodem1451)

ivankravets, 2015-09-28:
Give me please an output of
```
platformio serialports list
```

mseroczynski, 2015-09-28:
``Error: Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/__main__.py", line 86, in main
    cli(None, None, None)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/serialports.py", line 29, in serialports_list
    click.secho(item['port'], fg="cyan")
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/termui.py", line 417, in secho
    return echo(style(text, **styles), file=file, nl=nl, err=err, color=color)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/termui.py", line 390, in style
    return ''.join(bits)
TypeError: sequence item 1: expected string, NoneType found``

ivankravets, 2015-09-28:
:smile: and output of
```
pip list
```

mseroczynski, 2015-09-28:
bottle (0.12.8)
click (5.1)
fonttools (3.0)
lockfile (0.10.2)
pip (7.1.2)
platformio (2.3.2)
pylibpcap (0.6.4)
pyserial (2.7)
requests (2.7.0)
scapy (2.3.1)
scons (2.3.6)
setuptools (18.3.2)
vboxapi (1.0)
virtualenv (13.1.2)
wheel (0.26.0)

ivankravets, 2015-09-28:
Can you reproduce this issue using Terminal application? As I understand, it happens within CLion IDE?

mseroczynski, 2015-09-28:
Mmkay, so step by step on new project:
``platformio init --ide clion --board yun``
Then I import to CLion. Btw, I'm having this suspicious enty in CMakeLists.txt generated - ``add_definitions(-DUSB_PRODUCT=\&quot;Arduino Yun\&quot;)``

Now i add main.cpp file to /src with some simple example. Then I'm running PLATFORMIO_BUILD configuration with below IDE result:

``/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/ches/Library/Caches/clion11/cmake/generated/5ce0a0c2/5ce0a0c2/Debug --target PLATFORMIO_BUILD -- -j 8
[Mon Sep 28 20:44:13 2015] Processing yun (platform: atmelavr, targets: upload, framework: arduino, board: yun)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/yun/firmware.hex"])
Auto-detected UPLOAD_PORT/DISK: None
scons: *** [upload] No such file or directory
========================= [ ERROR ] Took 0.55 seconds =========================
make[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Error 1
make[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
make[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
make: *** [PLATFORMIO_BUILD] Error 2``

And in Terminal ``platformio run`` (in root of project):
[Mon Sep 28 20:45:28 2015] Processing yun (platform: atmelavr, targets: upload, framework: arduino, board: yun)
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/yun/firmware.hex"])
Auto-detected UPLOAD_PORT/DISK: None
scons: *** [upload] No such file or directory
========================= [ ERROR ] Took 0.54 seconds =========================

mseroczynski, 2015-09-30:
Btw, are you sure it's "help wanted" not a bug issue? Have CLion support for yun board ever been tested?

ivankravets, 2015-09-30:
@mseroczynski try the latest build `2.3.3.dev2` http://docs.platformio.org/en/latest/installation.html#development-version

Did it help you?

mseroczynski, 2015-09-30:
Ok, I've updated and now it correctly (?) generates ``add_definitions(-DUSB_PRODUCT="Arduino Yun")`` in ``CMakeLists.txt``, but for platformio.ini like below:

``[env:yun]
platform = atmelavr
framework = arduino
board = yun
targets = upload
upload_port = /dev/cu.usbmodem14141``

It still throws error when running PLATFORMIO_BUILD:
``scons: *** [upload] Explicit exit, status Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.
Error: Couldn't find a board on the selected port. Check that you have the correct port selected. If it is correct, try pressing the board's reset button after initiating the upload.``

ivankravets, 2015-09-30:
Do you still have problem with `platformio serialports list` command? 
It looks like PlatformIO can't detect automatically upload port.

Which is output of `ls -la /dev/cu.*` when board is connected?

mseroczynski, 2015-09-30:
``platformio serialports list``
``Error: Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/__main__.py", line 86, in main
    cli(None, None, None)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/platformio/commands/serialports.py", line 29, in serialports_list
    click.secho(item['port'], fg="cyan")
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/termui.py", line 417, in secho
    return echo(style(text, **styles), file=file, nl=nl, err=err, color=color)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/termui.py", line 390, in style
    return ''.join(bits)
TypeError: sequence item 1: expected string, NoneType found``

``ls -la /dev/cu.*``
``crw-rw-rw-  1 root  wheel   18,   1  6 wrz 11:31 /dev/cu.Bluetooth-Incoming-Port
crw-rw-rw-  1 root  wheel   18,   3  6 wrz 11:31 /dev/cu.Bluetooth-Modem
crw-rw-rw-  1 root  wheel   18, 0x00000189 30 wrz 20:02 /dev/cu.usbmodem14141``

mseroczynski, 2015-09-30:
You're right about detecting ``upload_port``. As i've mentioned before im adding it by myself. If i dont I get this log:

``(...)
avr-objcopy -O ihex -R .eeprom .pioenvs/yun/firmware.elf .pioenvs/yun/firmware.hex
BeforeUpload(["upload"], [".pioenvs/yun/firmware.hex"])
Auto-detected UPLOAD_PORT/DISK: None
scons: *** [upload] No such file or directory
========================= [ ERROR ] Took 2.68 seconds =========================
make[3]: *** [CMakeFiles/PLATFORMIO_UPLOAD] Error 1
make[2]: *** [CMakeFiles/PLATFORMIO_UPLOAD.dir/all] Error 2
make[1]: *** [CMakeFiles/PLATFORMIO_UPLOAD.dir/rule] Error 2
make: *** [PLATFORMIO_UPLOAD] Error 2``

ivankravets, 2015-10-01:
> /Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/click/termui.py", line 390, in style return ''.join(bits) TypeError: sequence item 1: expected string, NoneType found

This is problem of https://github.com/mitsuhiko/click , which PlatformIO uses. Which type of Terminal do you use? I've not seen similar errors before.

Could you try https://www.iterm2.com?

> ls -la /dev/cu.*

Thanks. Could you provide also raw information about serial ports which returns PlatformIO?
```shell
# launch Python interaction session
python

> from platformio.util import get_serialports
> get_serialports()
```

ivankravets, 2015-10-01:
I've updated my comment above. I found why you receive this error. Please answer to my comment above. Thanks.

mseroczynski, 2015-10-01:
This is awesome, I'll post my report after I get home from work in 3h ;)

ivankravets, 2015-10-03:
1. Please upgrade to the latest PlatformIO 2.3.3
2. `platformio serialports list` command should wok now
3. Give please result of `python -m serial.tools.list_ports` command

ivankravets, 2015-10-03:
The problem is that PlatformIO doesn't see your system serial ports. PlatformIO uses [PySerial](https://github.com/pyserial/pyserial) tool for it. Let's test the latest alpha version of this tool:
```shell
pip install -U virtualenv
cd /tmp
virtualenv piotest
source piotest/bin/activate
pip list
pip install https://github.com/pyserial/pyserial/archive/master.zip
python -m serial.tools.list_ports
```

mseroczynski, 2015-10-03:
After updating PySerial ``python -m serial.tools.list_ports -v``:

``Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 162, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/runpy.py", line 72, in _run_code
    exec code in run_globals
  File "/private/tmp/piotest/lib/python2.7/site-packages/serial/tools/list_ports.py", line 102, in <module>
    main()
  File "/private/tmp/piotest/lib/python2.7/site-packages/serial/tools/list_ports.py", line 84, in main
    iterator = sorted(comports())
  File "/private/tmp/piotest/lib/python2.7/site-packages/serial/tools/list_ports_osx.py", line 233, in comports
    serial_interfaces = scan_interfaces()
  File "/private/tmp/piotest/lib/python2.7/site-packages/serial/tools/list_ports_osx.py", line 209, in scan_interfaces
    device = get_string_property(service, "IOCalloutDevice")
  File "/private/tmp/piotest/lib/python2.7/site-packages/serial/tools/list_ports_osx.py", line 101, in get_string_property
    output = cf.CFStringGetCStringPtr(CFContainer, 0).decode('mac_roman')
AttributeError: 'NoneType' object has no attribute 'decode'``

And ``platformio serialports list`` returns nothing.

mseroczynski, 2015-10-03:
``bottle (0.12.8)
click (5.1)
fonttools (3.0)
lockfile (0.10.2)
pip (7.1.2)
platformio (2.3.4.dev0)
pylibpcap (0.6.4)
pyserial (2.7)
requests (2.7.0)
scapy (2.3.1)
scons (2.3.6)
setuptools (18.3.2)
vboxapi (1.0)
virtualenv (13.1.2)
wheel (0.26.0)
``

mseroczynski, 2015-10-03:
And after turning off CLion:
``MacBook-Pro-ches:~ ches$ python -m serial.tools.list_ports -v``

``None                
    desc: None
    hwid: USB VID:PID=2341:8041 SNR=None
None                
    desc: n/a
    hwid: n/a
2 ports found``

ivankravets, 2015-10-07:
Sorry, I can't help you here. Could I ask you to open this issue here @pyserial  (https://github.com/pyserial/pyserial/issues) with the report from https://github.com/platformio/platformio/issues/294#issuecomment-145293081 ?

mseroczynski, 2015-10-07:
Sorry for not updating issue, been busy lately. What I found today is "Serial re-enumeration on reset." section on https://www.arduino.cc/en/Guide/ArduinoYun. I thought it may be useful.

ivankravets, 2015-10-08:
Could you contact with this issue directly to @pyserial? I have idea how to make hook for this issue, but it isn't good solution. You can make back-link to this issue.

Thanks in advance!

ivankravets, 2015-10-14:
Please re-open this issue if you will have any problems with the fixed @pySerial. 

P.S: I've just opened new issue with adding support for @pySerial 3.0 https://github.com/platformio/platformio/issues/307

ivankravets, 2015-11-06:
@mseroczynski, have you resolved this issue? See https://github.com/pyserial/pyserial/issues/19

Try the latest version of @pyserial 
```
pip install https://github.com/pyserial/pyserial/archive/master.zip
```

mseroczynski, 2015-11-16:
Stiiill got it, verified with ``pip show pyserial``

ivankravets, 2015-11-16:
Great! Don't forget to use the latest PlatformIO 2.4.0, it has support for @pyserial 3.0

mseroczynski, 2015-11-16:
Confirmed on ``2.4.0.dev3``.

ivankravets, 2015-11-16:
Thanks! Have a nice using of @PlatformIO :smile:

mseroczynski, 2015-11-16:
Maybe it'll be any clue for you. Below is the log of working upload using Arduino IDE (sry for pasting the biggest log ever):
```
Sketch uses 4,744 bytes (16%) of program storage space. Maximum is 28,672 bytes.
Global variables use 151 bytes (5%) of dynamic memory, leaving 2,409 bytes for local variables. Maximum is 2,560 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem14141
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14141, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14141, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } => {}
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/tty.Bluetooth-Incoming-Port, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14141, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14141, } => {/dev/cu.usbmodem14141, /dev/tty.usbmodem14141, }
Found upload port: /dev/cu.usbmodem14141
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -patmega32u4 -cavr109 -P/dev/cu.usbmodem14141 -b57600 -D -Uflash:w:/var/folders/h6/hltlqscj729_mkg5phvqqp740000gn/T/build218845139059540782.tmp/Blink.cpp.hex:i 

avrdude: Version 6.0.1, compiled on Apr 14 2015 at 16:30:25
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/ches/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem14141
         Using Programmer              : avr109
         Overriding Baud Rate          : 57600
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : butterfly
         Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: .
Found programmer: Id = "CATERIN"; type = S
    Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
    Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587
avrdude: reading input file "/var/folders/h6/hltlqscj729_mkg5phvqqp740000gn/T/build218845139059540782.tmp/Blink.cpp.hex"
avrdude: writing flash (4744 bytes):

Writing | ################################################## | 100% 0.37s

avrdude: 4744 bytes of flash written
avrdude: verifying flash memory against /var/folders/h6/hltlqscj729_mkg5phvqqp740000gn/T/build218845139059540782.tmp/Blink.cpp.hex:
avrdude: load data flash data from input file /var/folders/h6/hltlqscj729_mkg5phvqqp740000gn/T/build218845139059540782.tmp/Blink.cpp.hex:
avrdude: input file /var/folders/h6/hltlqscj729_mkg5phvqqp740000gn/T/build218845139059540782.tmp/Blink.cpp.hex contains 4744 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.05s

avrdude: verifying ...
avrdude: 4744 bytes of flash verified

avrdude done.  Thank you.
```

ivankravets, 2015-11-16:
It seems that problem is linked with @PlatformIO and Yun board. I don't have this board and haven't tested uploading yet. Could you provide me access to PC where this board is connected? Need to make a few tests to determine problem. TeamViewer is preferable. Please contact me@ikravets.com with ID/Pass for the session (EET 13:00-23:00 http://www.timeanddate.com/time/zones/eet).
Thanks.

mseroczynski, 2015-11-16:
Of course! What about trying 17 Nov (for me still 'tomorrow' ;)) at 18:00 EET?

ivankravets, 2015-11-17:
Please re-test the latest development version

closed 2015-11-17 by ivankravets

mseroczynski, 2015-11-18:
Works perfectly now :). Good job!

commit 5454c354 (2015-11-17): Fix empty list with serial ports on Mac OS X // Resolve #294