platformio/platformio-core #295: Unable to install PlatformIO

tiferrei, 2015-09-28:
Hi, I am having a lot of trouble installing PlatformIO, my main goal is to use this to program (run, build, compile, etc.) arduino projects on Atom. As suggested here: https://viget.com/extend/arduino-development-in-atom-editor
I already tried all the options, throw pip, throw python curl, etc. But I always get this error:

    Exception:
    Traceback (most recent call last):
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/basecommand.py", line 211, in main
        status = self.run(options, args)
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/commands/install.py", line 311, in run
        root=options.root_path,
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_set.py", line 640, in install
        requirement.uninstall(auto_confirm=True)
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_install.py", line 716, in uninstall
        paths_to_remove.remove(auto_confirm)
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/req/req_uninstall.py", line 125, in remove
        renames(path, new_path)
      File "/Library/Python/2.7/site-packages/pip-7.1.2-py2.7.egg/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
        copytree(src, real_dst, symlinks=True)
      File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
        raise Error, errors
    Error:        [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', "[Errno 1] Operation not permitted: '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', "[Errno 1] Operation not permitted: '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', "[Errno 1] Operation not permitted: '/tmp/pip-m583fL-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib'")]

     [FAILURE]
    The installation process has been FAILED!
    Please report about this problem here

Thank you very much for your time! 

PS. Would appreciate detailed answers (as you can see) I'm a learner 😃

Thanks,
Tiago Ferreira

ivankravets, 2015-09-28:
Hi,

> [Errno 1] Operation not permitted

You should install using administrative privileges(`sudo`):
```shell
sudo pip install -U pip setuptools
sudo pip install -U platformio
```

tiferrei, 2015-09-28:
Sorry, already tried, it says to run with "sudo -H pip install -U pip setuptools", so I did it that way, exact same error :/

Thank you for your time!

ivankravets, 2015-09-28:
Have you tried installer script? http://docs.platformio.org/en/latest/installation.html#installer-script

tiferrei, 2015-09-28:
I did, same error :/

ivankravets, 2015-09-28:
Give me please an output of 
```
pip list
```

tiferrei, 2015-09-28:
altgraph (0.10.2)
bdist-mpkg (0.5.0)
bonjour-py (0.3)
bottle (0.12.8)
click (5.1)
lockfile (0.10.2)
macholib (1.5.1)
matplotlib (1.3.1)
modulegraph (0.10.4)
numpy (1.8.0rc1)
pip (7.1.2)
py2app (0.7.3)
pyobjc-core (2.5.1)
pyobjc-framework-Accounts (2.5.1)
pyobjc-framework-AddressBook (2.5.1)
pyobjc-framework-AppleScriptKit (2.5.1)
pyobjc-framework-AppleScriptObjC (2.5.1)
pyobjc-framework-Automator (2.5.1)
pyobjc-framework-CFNetwork (2.5.1)
pyobjc-framework-Cocoa (2.5.1)
pyobjc-framework-Collaboration (2.5.1)
pyobjc-framework-CoreData (2.5.1)
pyobjc-framework-CoreLocation (2.5.1)
pyobjc-framework-CoreText (2.5.1)
pyobjc-framework-DictionaryServices (2.5.1)
pyobjc-framework-EventKit (2.5.1)
pyobjc-framework-ExceptionHandling (2.5.1)
pyobjc-framework-FSEvents (2.5.1)
pyobjc-framework-InputMethodKit (2.5.1)
pyobjc-framework-InstallerPlugins (2.5.1)
pyobjc-framework-InstantMessage (2.5.1)
pyobjc-framework-LatentSemanticMapping (2.5.1)
pyobjc-framework-LaunchServices (2.5.1)
pyobjc-framework-Message (2.5.1)
pyobjc-framework-OpenDirectory (2.5.1)
pyobjc-framework-PreferencePanes (2.5.1)
pyobjc-framework-PubSub (2.5.1)
pyobjc-framework-QTKit (2.5.1)
pyobjc-framework-Quartz (2.5.1)
pyobjc-framework-ScreenSaver (2.5.1)
pyobjc-framework-ScriptingBridge (2.5.1)
pyobjc-framework-SearchKit (2.5.1)
pyobjc-framework-ServiceManagement (2.5.1)
pyobjc-framework-Social (2.5.1)
pyobjc-framework-SyncServices (2.5.1)
pyobjc-framework-SystemConfiguration (2.5.1)
pyobjc-framework-WebKit (2.5.1)
pyOpenSSL (0.13.1)
pyparsing (2.0.1)
pyserial (2.7)
python-dateutil (1.5)
pytz (2013.7)
requests (2.7.0)
scipy (0.13.0b1)
scons (2.3.6)
setuptools (1.1.6)
six (1.4.1)
vboxapi (1.0)
virtualenv (13.1.2)
wheel (0.26.0)
xattr (0.6.4)
zope.interface (4.1.1)

ivankravets, 2015-09-28:
Try
```
sudo pip --no-cache-dir install -U platformio
```

tiferrei, 2015-09-28:
It Worked!!! Thanks!! You made my day!!!! What was the problem by the way?

closed 2015-09-28 by tiferrei

ivankravets, 2015-09-28:
Thanks a lot for the report! I've just added this solution to FAQ page http://docs.platformio.org/en/latest/faq.html#faq

ivankravets, 2015-09-28:
> What was the problem by the way?

I can't answer to this question, I'm not maintainer for `pip` tool. I saw that `pip` has problem with `caches`, then the "in-mind" solution was to disable it :smile:

tiferrei, 2015-09-28:
Well anyway thank you very much!

parthpuneet, 2017-09-25:
sudo pip --no-cache-dir install -U predictionio facing same issues when trying to install predictionio

commit a714e608 (2015-09-28): Add to FAQ "Unable to install PlatformIO: [Errno 1] Operation not permitted" // Resolve #295