platformio/platformio-core #296: esp8266 compile errors about RAM size when adding 1 library

BSWG, 2015-09-28:
After adding just 1 library and trying the example code I am getting an error:

[09/28/15 23:42:08] Processing esp01 (platform: espressif, upload_port: COM3, targets: upload, framework: arduino, board: esp01)

xtensa-lx106-elf-g++ -o .pioenvs\esp01\src\main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -DPLATFORMIO=020302 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant -I.pioenvs\esp01\ESP8266WiFi -I.pioenvs\esp01\pubsubclient src\main.cpp
xtensa-lx106-elf-ar rcs .pioenvs\esp01\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\Esp.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\Esp.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\FileSystem.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\FileSystem.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\HardwareSerial.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\HardwareSerial.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\IPAddress.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\IPAddress.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\Print.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\Print.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\Stream.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\Stream.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\Tone.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\Tone.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\WMath.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\WMath.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\WString.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\WString.cpp.cpp
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\abi.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\abi.cpp.cpp
xtensa-lx106-elf-gcc -x assembler-with-cpp -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant -c -o .pioenvs\esp01\FrameworkArduino\cont.o .pioenvs\esp01\FrameworkArduino\cont.S
xtensa-lx106-elf-gcc -o .pioenvs\esp01\FrameworkArduino\cont_util.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\cont_util.c
xtensa-lx106-elf-g++ -o .pioenvs\esp01\FrameworkArduino\core_esp8266_main.cpp.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\core_esp8266_main.cpp.cpp
xtensa-lx106-elf-gcc -o .pioenvs\esp01\FrameworkArduino\core_esp8266_noniso.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenvs\esp01\FrameworkArduino -I.pioenvs\esp01\FrameworkArduinoVariant .pioenvs\esp01\FrameworkArduino\core_esp8266_noniso.c
xtensa-lx106-elf-gcc -o .pioenvs\esp01\FrameworkArduino\core_esp8266_si2c.o -c -std=c99 -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10604 -IC:\Users\Bart\.platformio\packages\framework-arduinoespressif\sdk\include -I.pioenv [... cut ...]
  pinMode(BUILTIN_LED, OUTPUT);     // Initialize the BUILTIN_LED pin as an output
  Serial.begin(115200);
  setup_wifi();
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
}

void loop() {

  if (!client.connected()) {
    reconnect();
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 2000) {
    lastMsg = now;
    ++value;
    snprintf (msg, 75, "hello world #%ld", value);
    Serial.print("Publish message: ");
    Serial.println(msg);
    client.publish("outTopic", msg);
  }
}

m4mbart, 2015-09-29:
When compiling ab example using the Arduino IDE and installed esp8266 module from the Board Manager the resulting message is:

`Sketch uses 301,840 bytes (69%) of program storage space. Maximum is 434,160 bytes.
Global variables use 48,456 bytes (59%) of dynamic memory, leaving 33,464 bytes for local variables. Maximum is 81,920 bytes.`

The library used is: https://github.com/morrissinger/ESP8266-Websocket
I tested this with the websocketclient example code.

ivankravets, 2015-09-29:
Thanks friends. We are working on it...

closed 2015-10-02 by valeros

m4mbart, 2015-10-02:
Cool! I have no development board available to check the actual uploading. But compiling works successful now :+1:

ivankravets, 2015-10-02:
@m4mbart thanks for the report! We are going to release PlatformIO 2.3.3 today. Don't forget to upgrade for the stable release.

daniellycosta, 2017-09-04:
I'm with the same problem

`"c:/users/dany/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: .pioenvs\d1_mini\firmware.elf section `.text' will not fit in
region `iram1_0_seg'`

commit 4dcef893 (2015-10-02): Update espressif platform // Resolve #296, #297
commit 948b75ae (2015-10-02): Merge pull request #299 from valeros/develop

Update espressif platform // Resolve #296, #297