platformio/platformio-core #298: Unable to use Souliss library

bcelary, 2015-09-30:
This problem is related to the souliss library (https://github.com/souliss/souliss) / souliss.net.

All the Arduino libraries I've tried so far with the Platformio worked fine. Unfortunately not the souliss library. I believe the problem might be related to the way the library uses precompiler but it works OK in the Arduino IDE.

I have searched for but haven't found any option to make platformio behave more like Arduino IDE when building a project. But since the goal of the tool is to work as official IDE, I thought it would be good if this library could be compiled. If there's a way to compile it somehow - sorry for inaccurate report.

ivankravets, 2015-10-01:
We can build this @souliss library using the latest PlatformIO. This library requires specific building logic. It has `*.cpp` files, but they should be compiled, because will be used by Preprocessor as `#include`. For example, see https://github.com/souliss/souliss/blob/friariello/webhook/webhook.h#L43:L47

The normal way is when compiler builds `C/C++` source code, which PlatformIO does by default. However, you receive `error: redefinition` which is supposed result. Firstly, the source code of `cpp` file will be included into file via `#include`, secondly, PlatformIO will build this `cpp`.

> it works OK in the Arduino IDE.

If something works in Arduino IDE it doesn't mean that it should work anywhere. Arduino Builder has specific logic: they build only source code from root of lib dir, `utility` dir, and `src` dir if exists. As result, Arduino IDE doesn't build any `*.CPP` file because doesn't see them.

PlatformIO Builder is more powerful. You can control which source code should be build using [src_filter](http://docs.platformio.org/en/latest/projectconf.html#src-filter) option. However, this option can be applied just for PROJECT, not for the libraries. I opened for it new issue #289 . It will allow developers to control which source code should be compiled.

bcelary, 2015-10-01:
Sounds good. Thanks for checking it. A way to "emulate" Arduino builder would be all I need in this case I guess. I will try it once the option is available.

ivankravets, 2015-10-01:
Temporary solution: you can rename `*.cpp` files to `.hpp`(for example), including the source code. Please it to your project's `lib` directory.

bcelary, 2015-10-12:
Finally found some time to check this.

The (mis?)feature of this lib is that it is compiled with the main file and the includes source everything from the library directly (along with cpp files). I guess this is needed so that the defines can be used to configure it. This means the library should not be compiled to an object file at all. What I did to get it compiled with PlatformIO is an addition of the following to the `platformio.ini` file:

    build_flags = -I/path/to/lib/souliss
    lib_ignore = souliss

I tried using `build_flags = -I%lib_dir%/souliss` but that does not seem to do the trick. I bet the PlatformIO logic gets confused by the fact that I am requesting to ignore the lib and include it at the same time :)

ivankravets, 2015-10-13:
This is normal solution when library is not intended to be compiled in common way.

> I tried using build_flags = -I%lib_dir%/souliss but that does not seem to do the trick.

Try
```ini
build_flags = -I${LIBSOURCE_DIRS[1]}/souliss
```

See https://github.com/platformio/platformio/blob/develop/platformio/builder/main.py#L94:L98

ivankravets, 2015-10-13:
Please reopen if you need additional help

closed 2015-10-13 by ivankravets

bcelary, 2015-10-13:
Yup. Now it works.

Thanks!

ivankravets, 2015-11-24:
Added to FAQ http://docs.platformio.org/en/latest/faq.html#can-not-compile-a-library-that-compiles-without-issue-with-arduino-ide

ivankravets, 2016-07-15:
Resolved in https://github.com/platformio/platformio/issues/289

bcelary, 2016-09-06:
I don't get it. Marked as resolved but the FAQ entry is gone and the behavior did not change - still that change in ini file is needed to get it to work. Why is that so? Am I missing something?

ivankravets, 2016-09-07:
@bcelary Please upgrade to PlatformIO 3.0 where this issue is completely resolved.