platformio/platformio-core #300: Unable to use custom library

deecewan, 2015-10-07:
I am trying to use a custom library (under lib/library_name).

The library includes a makefile, and is prebuilt, however all the source is still there.

When I try and build my project, I get errors due to the library needing -std=gnu99 (or c99).
I am looking for a way to either include the lib.a file (instead of PlatformIO building it itself), or to set a flag for that library to be built with -std=gnu99.  If I add that flag for the whole project, I get errors when FrameworkArduino tries to build.

The project is written in C, and I'm trying to get everything working under CLion.

ivankravets, 2015-10-07:
The normal solution for this issue depends on #289.

Temporary solution:
1. Place library source code outside `lib` directories. Like, `/tmp/library_name`. Build it using Makefile. You should receive `/tmp/library_name/foo_lib.a`
2. In [platformio.ini](http://docs.platformio.org/en/latest/projectconf.html) please specify [build_flags](http://docs.platformio.org/en/latest/projectconf.html#build-flags) to 
```ini
build_flags = -L/tmp/library_name -lfoo_lib.a
```

ivankravets, 2015-12-10:
please reopen this issue if temporary solution doesn't work for you

closed 2015-12-10 by ivankravets

janoist1, 2018-05-20:
@ivankravets  How to build `.a` file from an ESP-IDF project?