platformio/platformio-core #301: mbed: cstddef: No such file or directory

laurentS, 2015-10-07:
I am trying to build a very simple piece of code for mbed. My code starts with ``#include "mbed.h"`` which includes ``platform.h``. When running `platformio run` I get the following error:
```
.pioenvs/nucleo_l053r8/FrameworkMbedInc-250686161/platform.h:25:19: fatal error: cstddef: No such file or directory

#include <cstddef>
^
compilation terminated
```

I'm on ubuntu trusty (14.04), I read forum posts here and there that say I should upgrade to `arm-none-eabi-gcc` version 4.9.3, but this doesn't solve the problem.
I removed platformio entirely, reinstalled it with pip in a virtualenv, but no luck.

I eventually fixed the issue by adding
```
build_flags = -I/home/laurent/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/4.8.3/ -I/home/laurent/.platformio/packages/toolchain-gccarmnoneeabi/arm-none-eabi/include/c++/4.8.3/arm-none-eabi/
```
to my `platformio.ini`, but I don't feel like it's the right solution :)

Did I miss a step in the installation, or is this a bug? (maybe not with platformio...)

ivankravets, 2015-10-08:
@valeros is this a bug with toolchain?

valeros, 2015-10-09:
Hi @LaurentS !
Could you provide your project, please?

laurentS, 2015-10-09:
My code is a single file, copied from mbed Hello World at https://developer.mbed.org/handbook/mbed-SDK:
```C
#include "mbed.h"
DigitalOut myled(LED1);
 
int main() {
    while(1) {
        myled = 1;
        wait(0.2);
        myled = 0;
        wait(0.2);
    }
}
```
It is compiled in a virtual env installed with:
```bash
$ mkvirtualenv pio_venv
$ pip install platformio
$ platformio init -b nucleo_l152re
$ platformio run
```

I actually fixed the problem by renaming `src/main.c` to `src/main.cpp`. If this is expected behaviour, please accept my apologies for the waste of time.

versions just in case:
```bash
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
$ arm-none-eabi-g++ --version
arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
```

ivankravets, 2015-10-09:
mbed is `c++`-based framework. You should use `*.cpp` extension.

closed 2015-10-09 by ivankravets

polaroi8d, 2016-01-19:
Hy, i got the same problem, i used this tutorial https://developer.mbed.org/users/noritsuna/notebook/mbed-os-on-an-emulatorqemu/ 
I have 4.9.3 arm-none-eabi-gcc and when i try to '$yotta build' then crash the building.

ivankravets, 2016-01-19:
@polaroi8d please share here the code which allows us to reproduce this issue. Thanks.

polaroi8d, 2016-01-25:
@ivankravets sorry for not respoding.  Meanwhile i fixed the problem... I downloaded a new toolchain from  https://launchpad.net/gcc-arm-embedded/+download  and worked with this. ( After this, came up an other problem, the build directory hitched with the building process... fix: delet the building directory )

ivankravets, 2016-01-25:
Why you don't use built-in PlatformIO toolchain?

crespum, 2016-02-16:
I am having the same problem in Debian Testing with the simple code above and the plugin for [Atom](https://atom.io/packages/platomformio).

ivankravets, 2016-02-16:
@crespum Please take a look at official PlatformIO IDE for Atom package https://atom.io/packages/platformio-ide

crespum, 2016-02-16:
@ivankravets I found [your samples](https://github.com/platformio/platformio/tree/develop/examples/mbed/mbed-blink) and the only difference is that I was using `main.c` and you use `main.cpp`. Saving it as a C++ file solved this problem.

ivankravets, 2016-02-16:
@crespum I've added comment about it above https://github.com/platformio/platformio/issues/301#issue-110306963

3nd0y, 2019-12-24:
Hi @ivankravets 

Although my file use cpp extension, but it's still can't find cstddef, here my logs:

> Processing bluepill_f103c8 (platform: ststm32; framework: mbed; board: bluepill_f103c8)
 -------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/bluepill_f103c8.html
> PLATFORM: ST STM32 5.7.0 > BluePill F103C8
> HARDWARE: STM32F103C8T6 72MHz, 20KB RAM, 64KB Flash
> DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-mbed 5.51401.191023 (5.14.1)
> Collecting mbed sources...
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Looking for FATFileSystem library in registry
> Conflict: More than one library has been found by request {"requirements": null, "name": "FATFileSystem"}:
> FatFileSystem
> ===Automatically chose the first available library (use `--interactive` option to make a choice) =========
> #ID: 385
> Updated FAT File System driver. Features include: * Updated to R0.09 - Sep 06, 2011 [[http://elm-chan.org/fsw/ff/00index_e.html]] * Bug fixes from StГ©phane Bausseron ** [[http://mbed.org/forum/mbed/topic/2273/?page=1#comment-11521]] **...
> 
> Keywords: fat, file, filenames, fixes, long, system
> Compatible frameworks: mbed
> Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
> Authors: Adam Green
> 
> FatFileSystem
> =============
> #ID: 201
> Local copy of the FatFileSystem.
> 
> Keywords: fat
> Compatible frameworks: mbed
> Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
> Authors: Shinichiro Nakamura
> 
> FatFileSystem
> =============
> #ID: 206
> Updated get_fattime to use rtc and provide a date/time. This has been an annoying missing feature.
> 
> Keywords: fatfilesystem
> Compatible frameworks: mbed
> Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
> Authors: Andrew Lindsay
> 
> FatFileSystem
> =============
> #ID: 3585
> Some library needed for things? Licence unknown
> 
> Keywords: fatfilesystem
> Compatible frameworks: mbed
> Compatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500
> Authors: Tom Bloor
> 
> Found: https://platformio.org/lib/show/385/FatFileSystem
> LibraryManager: Installing id=385
> FatFileSystem @ 6ceefe1c53 is already installed
> Found 6 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <SDFileSystem-8db0d3b02cec>
> |   |-- <FatFileSystem>
> Building in release mode
> Compiling .pio\build\bluepill_f103c8\FrameworkMbed\drivers\source\AnalogIn.o
> ....
> ....
> Compiling .pio\build\bluepill_f103c8\lib7fc\FatFileSystem_ID385\FATDirHandle.o
[... 158 lines cut ...]
Here is the platformio.ini

> [env:bluepill_f103c8]
> platform = ststm32
> framework = mbed
> board = bluepill_f103c8
> ; build_flags = -D PIO_FRAMEWORK_MBED_RTOS_PRESENT
> lib_deps = FATFileSystem
> build_flags = -I C:\Users\fmukhtarif\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\c++\7.2.1

and the code:
```
#include "mbed.h"
#include "SDFileSystem.h"

SDFileSystem sd(PA_5, PA_6, PA_7, PA_8, "sd"); // the pinout on the mbed Cool Components workshop board
Serial pc(USBTX, USBRX); // tx, rx

int main() {
    pc.printf("Hello World!\n\r");
    while(1) {
        pc.putc(pc.getc() + 1); // echo input back to terminal
    }
} 
```