platformio/platformio-core #305: mbed support for teensy31 not properly implemented

heyer2, 2015-10-10:
Teensy support not properly implemented in the software (possible only the interface?)
Take a look at:
tformio/platformio/boards/teensy.json

Here it is indicated that the teensy31 supports the mbed framework.
However, actually using the software gives no such indication.

For example look at:
platformio/platformio/platforms/teensy.py

Here the mbed framework is not included!

ivankravets, 2015-10-10:
In board's `frameworks` field is defined default frameworks. `mbed` is optional framework for Teensy. See `meed-blink` example for Teensy 3.1 https://github.com/platformio/platformio/tree/develop/examples/mbed/mbed-blink

heyer2, 2015-10-10:
Yes, sorry about that. I only noticed the issue tracker after the email.
I looked at the blink example, but I can’t get it to work. I have a feeling it’s not just me doing something wrong.
Do you have skype? I can screen share you.

heyer2, 2015-10-10:
Here is the immediate error I get upon running the example:

>IOError: [Errno 2] No such file or directory: '/Users/heyer/.platformio/packages/framework-  mbed/variant/TEENSY3_1/TEENSY3_1.eix':

Seems the mbed package is not downloaded automatically when using the teensy31 board.

ivankravets, 2015-10-10:
1. Give me please an output of this command:
```
ls -la /Users/heyer/.platformio/packages/framework- mbed/variant/TEENSY3_1/
```

2. Try to reinstall `teensy` platform
```
platformio platforms uninstall teensy
platformio platforms install teensy
```

heyer2, 2015-10-11:
Result of first command:
>$ ls -la ~/.platformio/packages/framework-mbed/variant/TEENSY3_1/
ls: /Users/heyer/.platformio/packages/framework-mbed/variant/TEENSY3_1/: No such file or directory

I reinstalled the `teensy` platform using the commands you gave me, but it does not change the result of running the first command.

My version is 2.3.3

ivankravets, 2015-10-12:
Hm... Please  provide me this info:
```
python -c "import platform; print platform.uname()"

ls -la ~/.platformio/packages/framework-mbed/variant/
```

Thanks.

heyer2, 2015-10-12:
Python command:

    ('Darwin', 'Daniels-MacBook-Pro.local', '15.0.0', 'Darwin Kernel Version 15.0.0: Wed Aug 26 16:57:32 PDT 2015; root:xnu-3247.1.106~1/RELEASE_X86_64', 'x86_64', 'i386')

Upper directory listing (ls -la ~/.platformio/packages/framework-mbed, I removed the "variant" part of the query")

    ls: /Users/heyer/.platformio/packages/framework-mbed/: No such file or directory

I haven't really gone into your code, but I have a suspicion that the problem is that the teensy platform is not defined as using the 'mbed' platform. I can see the package is not installed when I install the teensy platform. Perhaps an issue in this file? 
https://github.com/platformio/platformio/blob/369d0e1a0040f7f4331cb7607dbaaaa8ab9d6f96/platformio/platforms/teensy.py

ivankravets, 2015-10-12:
Yes, this is bug!!! Thanks! :+1: 

Temporary solution, please install some platform which contains `framework-mbed`:
```
platformio platforms install nxplpc
```

heyer2, 2015-10-12:
It did indeed work.

commit 1164ef31 (2015-10-12): Fix missing of `framework-mbed` package for `teensy` platform // Resolve #305
closed 2015-10-12 by ivankravets