platformio/platformio-core #306: Use teensy CLI loader for upload of .hex files

heyer2, 2015-10-12:
At present platformIO uses the teensy GUI for the final upload of .hex files.
A more streamlined solution might be the command line loader available at:
https://www.pjrc.com/teensy/loader_cli.html

ivankravets, 2015-10-12:
We use CLI for Linux. Could you provide binaries for Mac OS X? I have discussion with @PaulStoffregen a few month before, he told me that there no support for Mac OS X yet.

PaulStoffregen, 2015-10-12:
Mac support was recently contributed.
https://github.com/PaulStoffregen/teensy_loader_cli/pull/12

You still have the compile the code to get a Mac executable.

heyer2, 2015-10-12:
I made a binary. I'm afraid I don't have my teensy nearby, but the program seems to work fine.
https://www.dropbox.com/s/l5850ic5umcb862/teensy_loader_cli?dl=0

ivankravets, 2015-10-12:
Thanks @heyer2! `tool-teensy` package has been updated. Try it:
```
platformio update
```

P.S: I retested it with my Teensy 3.1/Mac OS X and it works! :+1:

ivankravets, 2015-10-12:
@PaulStoffregen could we ask you to add link to http://platformio.org on these pages:

* https://www.pjrc.com/teensy/gcc.html
* https://www.pjrc.com/teensy/loader_cli.html

Thanks a lot!

closed 2015-10-12 by ivankravets

heyer2, 2015-10-13:
I can confirm it works on my end too.

PaulStoffregen, 2015-10-14:
@ivankravets 
I've added a "PlatformIO Integration" section.
https://www.pjrc.com/teensy/loader_cli.html
Can you recommend text to replace the "TODO" line?

ivankravets, 2015-10-14:
[PlatformIO](http://platformio.org) is a cross-platform code builder and the missing library manager (Ready for embedded development, IDE and Continuous Integration).

* [Install PlatformIO](http://platformio.org/#!/get-started)
* Print supported Teensy boards
```shell
platformio boards teensy
```
* [Initialise new project](http://docs.platformio.org/en/latest/userguide/cmd_init.html) with pre-configured settings for the specified board (for example, Teensy 3.1)
```shell
platformio init --board teensy31
````
* Put source code files to newly created `src` directory, build firmware and upload it
```shell
platformio run --target upload
```

Also, see examples of pre-configured Teensy projects:
* [HID USB Mouse](https://github.com/platformio/platformio/tree/develop/examples/teensy/teensy-hid-usb-mouse)
* [Chat Server with built-in library](https://github.com/platformio/platformio/tree/develop/examples/teensy/teensy-internal-libs)
* [MBED Blink](https://github.com/platformio/platformio/tree/develop/examples/mbed/mbed-blink)

ivankravets, 2015-10-14:
Dear @PaulStoffregen,

See my post above https://github.com/platformio/platformio/issues/306#issuecomment-148033650

You can modify it if you want.

Thanks in advance! :+1:  :beers:

commit f758d860 (2015-10-12): Use teensy CLI loader for upload of .hex files on OSX // Resolve #306