platformio/platformio-core #310: ESP-12E (nodeMCU) support?

grzegorzshub, 2015-10-20:
Hi
Any chance for nodeMCU support please?
Thank you

ivankravets, 2015-10-20:
Try this manual http://clemix.de/www/platformio

grzegorzshub, 2015-10-20:
Hello
Thank you for prompt answer.
I use Atom with platform on Mac. Does it mean that I can use -board esp01 for nodeMCU?
I’m a bit confused. That what I have tried but got some issues. The same code worked fine via Arduino IDE.
Thank you
Best
Grzegorz

ivankravets, 2015-10-20:
Hi @grzegorzshub,

Paste here source code which doesn't work with @PlatformIO.

Thanks.

grzegorzshub, 2015-10-20:
Hi Ivan
I’m trying to recreate the issue and get back to you with more details soon.
Thank you
Best
Grzegorz

grzegorzshub, 2015-10-20:
It looks like it is a Adafruit_NeoPixel library issue.
Code compiles with Atom+platformio without any warnings or errors, neopixel doesn’t work. Neopixel works fine when the same code is compiled with Arduino IDE.
The code:

#include <ESP8266WiFi.h>
#include <Adafruit_NeoPixel.h>

#define neoPin 14  //neopixel on nodeMCU pin D5
#define NUMPIXELS      1  // How many NeoPixels are attached to the Arduino?
Adafruit_NeoPixel pixels = Adafruit_NeoPixel(NUMPIXELS, neoPin, NEO_GRB + NEO_KHZ800);

void setup() {

pixels.begin();
pixels.setPixelColor(0, pixels.Color(255, 0, 0));
pixels.setBrightness(20);
pixels.show();
}

void loop() {
}

valeros, 2015-10-21:
Hi @grzegorzshub !
I've just added ESP-12E board.
Please update packages via `platformio update` and install development version of PlatformIO
http://docs.platformio.org/en/latest/installation.html#development-version

grzegorzshub, 2015-10-21:
Hi

Tested with PlatformIO version 2.4.0.dev0 and board = esp12e and still the same issue with neopixel.
Just noticed that arduino IDE works with nodeMCU ESP12E 80Mz, while platformio —boards 40Mz, does it matter?

ivankravets, 2015-10-21:
@grzegorzshub you can force MCU speed manually via http://docs.platformio.org/en/latest/projectconf.html#board-f-cpu

grzegorzshub, 2015-10-22:
neopixel library works fine with board_f_cpu = 80000000L
Thank you!

commit 32d3b632 (2015-11-02): Add support for Espressif ESP8266 ESP-12E board (NodeMCU) // Resolve #310
closed 2015-11-02 by ivankravets