platformio/platformio-core #311: Setting upload speed disables autoerase for flash memory

rnovacek, 2015-10-21:
I'm using PlatformIO to upload to standalone ATmega328p using Adruino Nano as programmer. 

My env config:
```
[env:ATMega328p_8MHz]
# Standalone ATMega328p running on 8MHz
platform = atmelavr
board = pro8MHzatmega328
board_mcu = atmega328p
board_f_cpu = 8000000L
framework = arduino
# Upload using programmer
upload_protocol = stk500v1
upload_speed =19200

build_flags = -Wall
# Adafruit-DHT, LowPower, RF24, RF24Network
lib_install = 19,38,433,435
```

With this config, `platformio run -t upload` fails with following error:
```
avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
0x08 != 0x6a
avrdude: verification error; content mismatch
```

The avrdude command is this:
```
"/home/radek/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328p -C "/home/radek/.platformio/packages/tool-avrdude/avrdude.conf" -c stk500v1 -b 19200 -D -P /dev/ttyUSB1 -U flash:w:.pioenvs/ATMega328p_8MHz/firmware.hex:i
```

I've figured out that if I remove '-D' command line option from avrdude command, it works fine.

When upload_speed is defined, the '-D' is automatically inserted:
https://github.com/platformio/platformio/blob/release/v2.3.4/platformio/builder/scripts/atmelavr.py#L35

ivankravets, 2015-10-21:
See related issue where `-D` option is required (Arduino IDE also uses it).

If you don't need this option, you can remove it using [extra-script](http://docs.platformio.org/en/latest/projectconf.html#extra-script).

```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(UPLOADERFLAGS=env['UPLOADERFLAGS'].remove("-D"))
```

P.S: I've not tested it. It just a hint.

ivankravets, 2015-10-21:
See https://github.com/platformio/platformio/blob/release/v2.3.4/platformio/builder/scripts/atmelavr.py#L90

rnovacek, 2015-10-21:
I can't get the extra-script to working, your example fails with `ValueError: list.remove(x): x not in list`. It seems that the BeforeUpload function is called after the extra-script, so that env var is not defined completely. The UPLOADERFLAGS env contains `['-v', '-p', '$BOARD_MCU', '-C', '"$PIOPACKAGES_DIR/tool-avrdude/avrdude.conf"', '-c', '$UPLOAD_PROTOCOL']` in the extra-script.

https://github.com/platformio/platformio/blob/develop/platformio/builder/scripts/atmelavr.py#L17

I'm not familiar with SCons, so I haven't yet figured out how to work it around.

rnovacek, 2015-10-21:
Are you sure that Arduino IDE needs `-D` option? It didn't used it when I tried - that's how I found that it's causing troubles for me. The difference might be that I used `Sketch -> Upload Using Programmer` option from the menu instead of normal `Upload` button (I need to program ATmega using Arduino as ISP).

ivankravets, 2015-10-21:
I've just tested it, sorry. Please see updated comment https://github.com/platformio/platformio/issues/311#issuecomment-149967444

rnovacek, 2015-10-21:
Now it works (I just had to add `"-P", "$UPLOAD_PORT"` to the list of options). Thank you very much.

Do you consider this use case as something that PlatformIO should supports? It would be great to do it without custom scripts.

ivankravets, 2015-10-21:
Which is default behaviour of Arduino IDE?

rnovacek, 2015-10-21:
It seems it depends if you program Arduino itself (`upload`) or program something using Arduino as ISP (`program`):

(from `platform.txt`):
```
tools.avrdude.upload.pattern="{cmd.path}" "-C{config.path}" {upload.verbose} -p{build.mcu} -c{upload.protocol} -P{serial.port} -b{upload.speed} -D "-Uflash:w:{build.path}/{build.project_name}.hex:i"

tools.avrdude.program.pattern="{cmd.path}" "-C{config.path}" {program.verbose} -p{build.mcu} -c{protocol} {program.extra_params} "-Uflash:w:{build.path}/{build.project_name}.hex:i"
```
I hope I understand it correctly, I'm not very familiar with this.

ivankravets, 2015-11-02:
Thanks. I will implement it via additional target named `program`. You should use `platformio run -t program` instead of `platformio run -t upload`

rnovacek, 2015-11-02:
That makes sense. Thank you.

ivankravets, 2015-11-02:
Please re-test it using the latest development version http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-11-02 by ivankravets

rnovacek, 2015-11-20:
Sorry for the delay. It works perfectly now. Thank you.

ivankravets, 2015-11-20:
Thanks for the report and that using @PlatformIO

commit 2c53bd49 (2015-11-02): Upload firmware using external programmer via "platformio run –target program" // Resolve #311