platformio/platformio-core #313: upload_port is not honored without a framework specified

countrodrigo, 2015-10-26:
The `upload_port` environment option doesn't seem to be honored when not using a framework. Here is my platform.ini:

```
[env:arduino_pro5v]
platform = atmelavr

#framework = arduino
#board = uno

board_mcu = atmega328
board_f_cpu = 16000000L

upload_protocol = arduino
upload_port = /dev/cu.usbmodem1451
upload_speed = 115200

targets = upload
```

If I use the `framework` and `board` options that are commented instead of the direct settings, it works just fine.

The avrdude command it tries to run is:
`/Users/countrodrigo/.platformio/packages/tool-avrdude/avrdude" -v -p atmega328 -C "/Users/countrodrigo/.platformio/packages/tool-avrdude/avrdude.conf" -c arduino -b 115200 -D -U flash:w:.pioenvs/arduino_pro5v/firmware.hex:i`
This is missing the -P option. If I run that command manually and pass the `-P /dev/cu.usbmodem1451`, then it can program the device just fine.

ivankravets, 2015-10-26:
Thanks for the report, I see this bug. However, this issue isn't linked with missed `framework` option, but with missed `board` option. Try to specify similar `board` to you hardware which is listed in `platformio boards`.

ivankravets, 2015-11-02:
Please re-test it in the latest develop version http://docs.platformio.org/en/latest/installation.html#development-version

closed 2015-11-02 by ivankravets

ivankravets, 2017-08-03:
People don't like to read documentation http://docs.platformio.org/en/latest/platforms/atmelavr.html#upload-using-programmer

alexbonhomme, 2017-08-03:
People don't often like this kind of comments too.

The documentation isn't that obvious (as you think it is). There is some examples with the `-P` flags, others without and others with `-P` and `-b` one. There is not clear explanation saying that the `upload_port` doesn't work as expected.

I would do a PR about it. Thanks for your help.

ivankravets, 2017-08-03:
@alexbonhomme fill free to make PR to https://github.com/platformio/platformio-docs/blob/develop/platforms/atmelavr_extra.rst

commit caa09613 (2015-11-02): Fix handling of upload port when board option is not specified in platformio.ini // Resolve #313