platformio/platformio-core #314: Builder exits with error - teensy & arduino

unsalted, 2015-10-28:
Mac OSX 10.10.5 (14F1021)
Python 2.7.10
PlatformIO, version 2.3.4

Can't think of a reason for this.

If I setup the board myself with a main.ino with setup(), loop() and my own config file I don't seem to have the problem.

``platformio ci --board teensylc
``

```
.pioenvs/teensylc/libFrameworkArduino.a(main.o): In function main':
/private/var/folders/r4/pmcrnkrj5rzg_4x2mf6hj3400000gn/T/tmp57Olfd/.pioenvs/teensylc/FrameworkArduino/main.cpp:21: undefined reference to 'setup'
/private/var/folders/r4/pmcrnkrj5rzg_4x2mf6hj3400000gn/T/tmp57Olfd/.pioenvs/teensylc/FrameworkArduino/main.cpp:23: undefined reference to 'loop'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/teensylc/firmware.elf] Error 1
```

ivankravets, 2015-10-28:
You should specify the path to source file or define global environment variable. See documentation http://docs.platformio.org/en/latest/userguide/cmd_ci.html

ivankravets, 2015-11-02:
If you still need help, please reopen this issue.

closed 2015-11-02 by ivankravets