platformio/platformio-core #317: builder in %home_dir%/platforms ignored

rph8, 2015-10-30:
In the docs [1], it says
```
Copy test.py and test-builder.py files to platforms directory.
```
This does not work however.
When copying test-builder.py to the directory, it is ignored.
It only worked after I copied that file to the installation directory /platformio/builder/scripts and named the file test.py like the others.

Used version 2.3.4.

[1]: http://docs.platformio.org/en/latest/platforms/creating_platform.html

ivankravets, 2015-11-02:
Did you place these files in `~/.platformio/platforms` directory?

ivankravets, 2015-11-05:
Please reopen issue if you still have this problem.

rph8, 2015-11-05:
Yes, I copied it to the platforms directory as described in the docs.

rph8, 2015-11-05:
So into ~/.platformio/platforms to be precise.

ivankravets, 2015-11-05:
I've just tried to reproduce this issue and all works good!

See where is located files from `test` platform:
```
.platformio  tree -L 2
.
├── appstate.json
├── lib
│   ├── DallasTemperature_ID54
│   ├── OctoWS2811_ID69
│   ├── OneWire_ID1
│   ├── RadioHead_ID124
│   └── souliss
├── packages
│   ├── framework-arduinoavr
│   ├── framework-arduinoespressif
│   ├── framework-arduinomsp430
│   ├── framework-arduinosam
│   ├── framework-arduinoteensy
│   ├── framework-cmsis
│   ├── framework-energiamsp430
│   ├── framework-energiativa
│   ├── framework-libopencm3
│   ├── framework-mbed
│   ├── framework-spl
│   ├── ldscripts
│   ├── sdk-esp8266
│   ├── tool-avrdude
│   ├── tool-bossac
│   ├── tool-esptool
│   ├── tool-lm4flash
│   ├── tool-micronucleus
│   ├── tool-mspdebug
│   ├── tool-stlink
│   ├── tool-teensy
│   ├── toolchain-atmelavr
│   ├── toolchain-gccarmlinuxgnueabi
│   ├── toolchain-gccarmnoneeabi
│   ├── toolchain-gcclinux32
│   ├── toolchain-gcclinux64
│   ├── toolchain-gccmingw32
│   ├── toolchain-timsp430
│   └── toolchain-xtensa
└── platforms
    ├── test-builder.py
    └── test.py

37 directories, 5 files
➜  .platformio
```

The content of files was copied from http://docs.platformio.org/en/latest/platforms/creating_platform.html

Result
```
platformio platforms list
atmelavr    with packages: toolchain-atmelavr, tool-avrdude, framework-arduinoavr, tool-micronucleus
atmelsam    with packages: framework-arduinosam, ldscripts, toolchain-gccarmnoneeabi, tool-bossac
espressif   with packages: ldscripts, sdk-esp8266, tool-esptool, framework-arduinoespressif, toolchain-xtensa
freescalekinetis with packages: framework-mbed, toolchain-gccarmnoneeabi
linux_arm   with packages: toolchain-gccarmlinuxgnueabi
linux_i686  with packages: toolchain-gcclinux32
linux_x86_64 with packages: toolchain-gcclinux64
native      with packages:
nordicnrf51 with packages: framework-mbed, toolchain-gccarmnoneeabi
nxplpc      with packages: framework-mbed, toolchain-gccarmnoneeabi
siliconlabsefm32 with packages: framework-mbed, toolchain-gccarmnoneeabi
ststm32     with packages: framework-libopencm3, toolchain-gccarmnoneeabi, tool-stlink, framework-spl, framework-cmsis, framework-mbed, ldscripts
teensy      with packages: framework-arduinoteensy, tool-teensy, toolchain-gccarmnoneeabi, framework-mbed, toolchain-atmelavr, ldscripts
test        with packages:
timsp430    with packages: toolchain-timsp430, tool-mspdebug, framework-energiamsp430, framework-arduinomsp430
titiva      with packages: ldscripts, framework-libopencm3, toolchain-gccarmnoneeabi, tool-lm4flash, framework-energiativa
windows_x86 with packages: toolchain-gccmingw32
```

closed 2015-11-05 by ivankravets