platformio/platformio-core #319: Add support for RFduino

JohnDBodkin, 2015-11-06:
Please consider adding a board type for RFduino that does not currently exist. http://www.rfduino.com/

ivankravets, 2015-11-09:
Please try develop branch of PlatformIO 
```shell
pip install https://github.com/platformio/platformio/archive/feature/issue-319-rfduino-support.zip

platformio platforms install nordicnrf51 --with-package "framework-arduinonordicnrf51"
```

Then use this data in `platformio.ini`
```ini
platform = nordicnrf51
framework = arduino
board = rfduino
```

Does it work for you?

JohnDBodkin, 2015-11-10:
I used your commands. Then I used "platformio init -d .\pioproj --ide eclipse --board=rfduino" and it created a platformio.ini file with the data you entered above. However when I imported the project into Eclipse I get:

Program "C:\Users\<USER_NAME>arm-none-eabi-g++" not found in PATH	
		Project Properties, C++ Preprocessor Include.../Providers, [CDT Cross GCC Built-in Compiler Settings] options	C/C++ Scanner Discovery Problem

Which maps to:
${USERPROFILE}arm-none-eabi-g++ ${FLAGS} -E -P -v -dD "${INPUTS}"

ivankravets, 2015-11-10:
Does it work without IDE? Could you compile source code and upload it?
```shell
# go to project directory where is located `platformio.ini`, then
platformio run --target upload
```

JohnDBodkin, 2015-11-11:
No it does not work without the IDE. 

C:\Users\UserName\pioproj>platformio run --target upload
[11/10/15 19:38:49] Processing rfduino (platform: nordicnrf51, targets: upload,
framework: arduino, board: rfduino)
-------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs\rfduino\src\tmp_ino_to.o -c -fno-rtti -fno-excepti
ons -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0 -no
stdlib -MMD -fno-builtin -DF_CPU=16000000L -D__RFduino__ -DARDUINO=231 -DPLATFOR
MIO=020400 -IC:\Users\UserName\.platformio\packages\framework-arduinonordicnrf51\
system\CMSIS\CMSIS\Include -IC:\Users\UserName\.platformio\packages\framework-ard
uinonordicnrf51\system\RFduino -IC:\Users\UserName\.platformio\packages\framework
-arduinonordicnrf51\system\RFduino\include -I.pioenvs\rfduino\FrameworkArduino -
I.pioenvs\rfduino\FrameworkArduinoVariant -I.pioenvs\rfduino\RFduinoBLE src\tmp_
ino_to.cpp
'arm-none-eabi-g++' is not recognized as an internal or external command,
operable program or batch file.
scons: *** [.pioenvs\rfduino\src\tmp_ino_to.o] Error 1
========================= [ ERROR ] Took 1.72 seconds =========================

C:\Users\UserName\pioproj>

ivankravets, 2015-11-11:
1. Did you apply any changes to this log? 
2. Is it real user name `UserName` in your system? 

I see problem like `C:\Users\UserName.platformio`. The `\` is missed. Should be like `C:\Users\UserName\.platformio...`

JohnDBodkin, 2015-11-11:
I changed the real user name to UserName with a search and replace in
Notepad++. Do I need to add the complete provided by the Arduino IDE or
from RFduino to the path?

ivankravets, 2015-11-11:
If you don't want to print this log here, please send it (without modifications) directly to me@ikravets.com. Thanks.

ivankravets, 2015-11-12:
Re-assigned to @ivankravets

TODO:
- [ ] Missed packages `tool-rfdloader` and `ldscripts` in platform
- [ ] Upload port should be required for this board

ivankravets, 2015-11-12:
@JohnDBodkin temporary solution: please define `upload_port` manually in http://docs.platformio.org/en/latest/projectconf.html#upload-port

closed 2015-12-26 by ivankravets

ivankravets, 2015-12-26:
@JohnDBodkin please install development version and try it http://docs.platformio.org/en/latest/installation.html#d-development-version

I'm waiting feedback from you. Thanks!

Happy Holidays and Merry Christmas!

JohnDBodkin, 2016-01-10:
I just took it for a test drive. I was on travel for the past month. It
works from the command line

ivankravets, 2016-01-13:
@JohnDBodkin thanks a lot for the report!

teknomage, 2022-02-16:
Hello, am attempting to program an RFduino using PlatformIO, using ST-link as the programmer. I understand that support for RFduino was already added as a board, but I am unable to call it up within PIO on VScode. This is what I've modified inside platformio.ini:
`[env:nrf51_dk]
platform = nordicnrf51
framework = arduino
board = rfduino
upload_protocol = stlink
debug_tool = stlink`

And when attempting to Build, this is the error I get:
`Processing nrf51_dk (platform: nordicnrf51; framework: arduino; board: rfduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Error: Unknown board ID 'rfduino'
The terminal process "C:\Users\SoundTek\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
`

Infact, I also checked %username%\.platformio\platforms\nordicnrf51\boards  and I do not see RFduino listed in there. I thought this board was already added in as a feature earlier??

commit 1b7b2ca0 (2015-12-26): Add support for RFDuino // Resolve #319