platformio/platformio-core #320: STM32F746 Discovery board support

mojzu, 2015-11-07:
Made an attempt at adding this myself but ended up making a lot of changes to default package directories which I'm unsure about. I can compile firmware successfully, but upload fails with scons error code 309 (guessing the tool-stlink binaries don't support the processor yet, but managed to get around this using ST's flashing tool and Em::block's STlinkGDB for debugging).

Changes:
- To support the Cortex-M7 I updated the toolchain in `.platformio/packages/toolchain-gccarmnoneeabi`
- Added new/changed files to `.platformio/packages/framework-cmsis/cores/stm32` from `stm32cubef7/Drivers/CMSIS/Include`
- Added directory `.platformio/packages/framework-cmsis/variants/stm32f7` with contents from `stm32cubef7/Drivers/CMSIS/Device/ST/STM32F7xx`
- Added linker script to `.platformio/packages/ldscripts/stm32f746ng.ld` from `stm32cubef7/Projects/STM32746G-Discovery/Templates/SW4STM32/STM32746G_Discovery/STM32F746NGHx_FLASH.ld`

And in the repository I added a board definition to `platformio/boards/ststm32.json`:
```
"disco_f746ng": {
    "build": {
        "core": "stm32",
        "extra_flags": "-DSTM32F746xx",
        "f_cpu": "216000000L",
        "ldscript": "stm32f746ng.ld",
        "cpu": "cortex-m7",
        "mcu": "stm32f746ngh6",
        "variant": "stm32f7"
    },
    "frameworks": ["cmsis"],
    "name": "ST STM32F7DISCOVERY",
    "platform": "ststm32",
    "upload": {
        "maximum_ram_size": 327680,
        "maximum_size": 1048576
    },
    "url": "http://www.st.com/web/catalog/mmc/FM141/SC1169/SS1858/LN1902/PF260869",
    "vendor": "ST"
},
```

ivankravets, 2015-11-10:
> To support the Cortex-M7 I updated the toolchain in .platformio/packages/toolchain-gccarmnoneeabi

Which toolchain have you used?

mojzu, 2015-11-11:
Version 4.9 update 3 from launchpad.

ivankravets, 2016-11-27:
It looks that issue is fixed in PlatformIO 3.0. If it doesn't work, pelase open new issue here https://github.com/platformio/platform-ststm32/issues/new

closed 2016-11-27 by ivankravets