platformio/platformio-core #322: Issue uploading with MSP430, libmsp430

dazhbog, 2015-11-09:
Hi, Ive been getting this annoying error with libmsp430.

I have energia installed, i can compile and upload code fine. Using platformio OR energia's command from the command line results in the following error.

```
dazhbog$ platformio run
[Tue Nov 10 05:57:25 2015] Processing lpmsp430f5529 (platform: timsp430, targets: upload, framework: energia, board: lpmsp430f5529)
----------------------------------------------------------------------------------------------------------------------------------------
/Users/dazhbog/.platformio/packages/tool-mspdebug/mspdebug tilib --force-reset "prog .pioenvs/lpmsp430f5529/firmware.hex"
MSPDebug version 0.20 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2012 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

tilib: can't find libmsp430.so: dlopen(libmsp430.so, 1): image not found
scons: *** [upload] Error 255
===================================================== [ ERROR ] Took 1.26 seconds =====================================================
```

MacOS 10.10.1

Thanks!

ivankravets, 2015-11-10:
I can't reproduce this issue on my Mac OS 10.11.1.

How about this solution? http://forum.43oh.com/topic/4930-msp430f5529-and-mspdebugger/

```shell
$ export LD_LIBRARY_PATH=~/.platformio/packages/tool-mspdebug
$ platformio run
```

ivankravets, 2015-11-15:
Please re-open this issue if you still have problem with uploading.

closed 2015-11-15 by ivankravets