platformio/platformio-core #323: Add support for TI SimpleLink platform (CC3200 Launchpad board)

berezovskyi, 2015-11-10:
Hi,

I would love to see CC3200 Launchpad supported by PlatformIO!

I looked through the `platforms` and `boards`directories in the source tree and new work seems to consist only of crafting proper JSON files? If it is so, I could try doing it myself initially.

Did you have a chance to work with the SimpleLink platform before?

ivankravets, 2015-11-10:
Hi,

> I looked through the platforms and boards directories in the source tree and new work seems to consist only of crafting proper JSON files? If it is so, I could try doing it myself initially.

No, need to create separate platform `tisimplelink`. The required core files are located in http://energia.nu application, called Energia. Also, need to create own build script and etc. If you are interested in it and want help us, I would be thankful.

> Did you have a chance to work with the SimpleLink platform before?

Yes, we have tried to work with TI CC2538DK and TI CC2650DK for @SmartAnthill project. Nevertheless, we didn't use PlatformIO because it was only research.

berezovskyi, 2016-03-08:
Hi Ivan, sorry for the late answer, the focus in my thesis got shifted to other tasks, but I will see if have the time to look into it afterward. Thank you for the pointers as to where to look. Best of luck to PlatformIO!

ivankravets, 2016-03-08:
Hey @berezovskyi,

Good luck with your thesis :+1: 

P.S: Don't forget about PlatformIO :blush:

ivankravets, 2016-11-27:
Moved to https://github.com/platformio/platform-titiva/issues/1

closed 2016-11-27 by ivankravets