platformio/platformio-core #324: Error installing with pip. "SyntaxError: Missing parentheses in call to 'print'"

jeffeb3, 2015-11-11:
I have python 2.7.9, and I get this error when trying to install:

    sudo pip install -U platformio
    Collecting platformio
      Using cached platformio-2.3.4.tar.gz
        Complete output from command python setup.py egg_info:
        Traceback (most recent call last):
          File "<string>", line 20, in <module>
          File "/tmp/pip-build-i9yd9he_/platformio/setup.py", line 8, in <module>
            from platformio import (__author__, __description__, __email__, __license__,
          File "/tmp/pip-build-i9yd9he_/platformio/platformio/util.py", line 50
            print line
                     ^
        SyntaxError: Missing parentheses in call to 'print'
    
        ----------------------------------------
    Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-i9yd9he_/platformio

pip version 7.1.2
setuptools 18.5
Linux (debian)

ivankravets, 2015-11-11:
Are you sure that use Python 2? See http://stackoverflow.com/questions/25445439/what-does-syntaxerror-missing-parentheses-in-call-to-print-mean-in-python

Give me please an output of
```
python --version
pip --version
which pip2
```

jeffeb3, 2015-11-11:
You are right pip is version 3 somehow. I will fix.

    python --version
    Python 2.7.9
    pip --version
    pip 7.1.2 from /usr/local/lib/python3.4/dist-packages (python 3.4)

jeffeb3, 2015-11-11:
That fixed it thank you.

closed 2015-11-11 by jeffeb3

ivankravets, 2015-11-11:
Sorry, PlatformIO doesn't support Python 3. It depends on SCons tool which doesn't work with Python 3 :cry:

serdaromer, 2020-11-16:
sudo python2 setup.py install komutu ile bu sorunu çözdüm