platformio/platformio-core #325: Monitor for emulated serial port?

hoosierEE, 2015-11-12:
Does anyone know how to monitor an *emulated* serial port?  I have a Teensy 3.1 and when I use RawHID + Serial, the Serial is in fact "emulated".  The Arduino IDE detects this condition and routes it to the serial monitor correctly.  Would it be possible to have this in platformio also?

Maybe it's as simple as specifying the port, but I'm doubtful...

ivankravets, 2015-11-12:
What do you see with
```shell
platformio serialports list
```

How about [platformio serialports monitor --port](http://docs.platformio.org/en/latest/userguide/cmd_serialports.html#cmdoption-platformio-serialports-monitor-p) option?

hoosierEE, 2015-11-12:
I think the problem is that the emulated serial is sent over the HID channel.  I'm guessing the Arduino IDE treats this as a special case.

Here's what I see from `platformio serialports list`:

    /dev/cu.Bluetooth-Incoming-Port
    -------------------------------
    Hardware ID: n/a
    Description: n/a

ivankravets, 2015-11-13:
Try these
```shell
# uninstall stable PySerial 2.0
pip uninstall pyserial
# install latest PySerial 3.0
pip install https://github.com/pyserial/pyserial/archive/master.zip

platformio serialports list

# revert original stable PySerial
pip uninstall pyserial
pip install pyserial
```

hoosierEE, 2015-11-13:
No difference with the latest pyserial.

ivankravets, 2015-11-13:
Which is your OS?

hoosierEE, 2015-11-13:
OSX 10.11.1

I should probably mention that when I ran `pip uninstall pyserial` it gave me this:

```
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/site-packages/pip/commands/uninstall.py", line 76, in run
    requirement_set.uninstall(auto_confirm=options.yes)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_set.py", line 305, in uninstall
    req.uninstall(auto_confirm=auto_confirm)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/usr/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 303, in move
    os.unlink(src)
OSError: [Errno 13] Permission denied: '/Library/Python/2.7/site-packages/pyserial-2.7.dist-info/DESCRIPTION.rst'
```

So naturally my follow-up was this:

    $ sudo !!

Which gave me this warning:

```
The directory '/Users/lx/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
```

After that, additional `pip` commands worked without asking for sudo.

hoosierEE, 2015-11-13:
OH!  This is probably important:

https://github.com/PaulStoffregen/Arduino-1.6.6-Teensyduino/blob/db190dcd6d9ccf56c28dfe5ee13d6989de03849a/app/src/processing/app/TeensyMonitor.java#L139-L153

There's special code in Teensyduino to allow Arduino's serial monitor to use the Teensy's emulated serial port.

ivankravets, 2015-11-13:
I have the same OS and @PaulStoffregen's Teensy 3.1. Could you provide source code where I can reproduce this issue? Thanks.

hoosierEE, 2015-11-13:
Sure: 
https://github.com/hoosierEE/DoubleTouch

hoosierEE, 2015-11-13:
Specifically, when I add [this line](https://github.com/hoosierEE/DoubleTouch/blob/master/src/DualRTouchPanels.h#L100) and the requisite [flag](https://github.com/hoosierEE/DoubleTouch/blob/master/platformio.ini#L13), platformio no longer detects the serial port, but the Arduino IDE (with Teensyduino addon) does detect it.

Without those two lines, `platformio serialports monitor` works normally.

ivankravets, 2015-11-16:
@hoosierEE sorry for the late answer. I've just looked in your code and can't understand what do you want receive. Firstly, see my PR as for `build_flags` https://github.com/hoosierEE/DoubleTouch/pull/1

Secondly, your code works good and my OS detects Teensy HID device. See screenshot. 
[image]

----------
PlatformIO doesn't have ability to communicate with HID devices. It can monitor only serial ports. HID != Serial. 

As for "Why does it work in Teensyduino?" It works because @PaulStoffregen created gateway between Teensyduino IDE and Teensyduino HID device. From Teensyduino @PaulStoffregen communicates with gateway via sockets. See what I found https://github.com/PaulStoffregen/Arduino-1.6.6-Teensyduino/blob/db190dcd6d9ccf56c28dfe5ee13d6989de03849a/app/src/processing/app/TeensyMonitor.java#L282

The `teensy_gateway` tool is located within Teensyduino IDE in tools folder. If you need it I can add it to `tool-teensy` package.

--------

Sorry, I'm not sure that we should implement the same `FakeSerial` within @PlatformIO. However, you can communicate with `teensy_gateway` and emulate virtual serial port. Then use native `platformio serialports monitor`.

hoosierEE, 2015-11-16:
I made a [video](https://youtu.be/AIbsaxIfAT4) of the issue I'm experiencing.  You can see in the video it starts off properly (ASCII numbers going to Arduino, raw byte array going to my logger), but if I kill the process and restart, it gets the wrong stream.

The problem eventually goes away after repeated kill-and-restart attempts.

The behavior seems to be that the "fake" serial ASCII bytes get sent to RawHID *instead of* whoever is supposed to listen to that socket.

ivankravets, 2015-11-16:
Do you have the same problems with firmware using Teensyduino?

hoosierEE, 2015-11-17:
That's a good idea, I'll upload it that way when I'm back at work tomorrow.  If it's the same then I'd just go ahead and mark this issue as 'closed'.  My issue has evolved into something like "why does the operating system *sometimes* route the output of `teensy_gateway` to a program which is only supposed to be listening to RawHID?"

However, I also just realized I didn't respond to some of your points from above, so let me fix that:

While I agree with your sentiment that implementing the `teensy_gateway` for just the Teensy is probably too vendor-specific, and therefore probably not a good fit for PlatformIO, I think sockets are an attractive alternative to serial as a communications channel.  It's a better fit for modern computers, which rarely have true hardware serial ports anymore.  Anyway, I wouldn't expect PlatformIO to provide a socket-based interface, but it's a nice thought.

hoosierEE, 2015-11-17:
I made a new [branch](https://github.com/hoosierEE/DoubleTouch/tree/ArduinoIDE) of my code and compiled with the Arduino IDE 1.6.5 and Teensyduino 1.25

Results: no behavior at all.  The Teensy shows up as a USB device but does not output any data.  No activity in Arduino's serial monitor, and nothing in my RawHID logger.  Strange.

ivankravets, 2015-11-17:
You should open new subject using https://forum.pjrc.com/forum.php

I hope you understand that I don't have any relations to Teensy or etc. I can help you with building your source code. Please contact with @PaulStoffregen.

Please re-open this issue or create new if you will have any problems with @PaltformIO builder. Thanks!

closed 2015-11-17 by ivankravets

hoosierEE, 2015-11-17:
Thanks for looking into this, Ivan.  I totally understand and appreciate that you took some time to help me troubleshoot even though my problem is not fully a part of PlatformIO's domain.

Your assistance helped isolate the problematic parts of the system, so thanks!

ivankravets, 2015-11-17:
You are welcome!

Thanks a lot that using @PlatformIO :+1: