platformio/platformio-core #327: Add sources in CLion

madhead, 2015-11-15:
I want to create a wrapper class to control Servo:

```cpp
# Gate.h
#ifndef PRACTICE_1_GATE_H
#define PRACTICE_1_GATE_H

#include <Arduino.h>
#include <Servo.h>

class Gate {
private:
    Servo servo;
public:
    Gate(uint8_t);

    void open();

    void close();
};

#endif //PRACTICE_1_GATE_H
```

```cpp
```

valeros, 2015-11-16:
Hi @madhead!
Have you tried to add the path to the library?
For example:
`include_directories("~/user/.platformio/packages/framework-arduinoavr/libraries/Servo/src")`

madhead, 2015-11-16:
@valeros `Servo.h` resolves ok when in `src/main.cpp`.

ivankravets, 2015-11-16:
@madhead where it doesn't resolve?

madhead, 2015-11-16:
It doesn't resolve in `src/gate/Gate.h`, `src/gate/Gate.cpp`. Including these files explicitly (`add_executable(practice_1 src/main.cpp src/gate/Gate.cpp)` instead of just `add_executable(practice_1 src/main.cpp src/gate)`) solves the problem.

It's not generally a bug, it's just a "help wanted" question. I'd like to hear from more experienced people how do they use multiple source files in platformio / CLion.

Also, what how should I treat `lib` folder? Putting sources there breaks even vanilla `platformio run` with exception like "Cannot find implementation for `Gate::Gate` / `Gate::open` / `Gate::close`". I include file as `#include "../lib/Gate.h"` and looks like `Gate.cpp` is no seen at all.

valeros, 2015-11-16:
@madhead 
Strange, I have only one file in add_executable option: `add_executable(clion src/blink.cpp)`, and autocomplete works fine in `gate.cpp`:
[image]

madhead, 2015-11-16:
@valeros can you please share whole your project as zip somewhere?

valeros, 2015-11-16:
Sure, bit only for Windows:https://dl.dropboxusercontent.com/u/10025728/clion.zip

madhead, 2015-11-19:
Got it. Thanks for the example. The trick was adding `include_directories("${HOME}/.platformio/packages/framework-arduinoavr/libraries/Servo/src")` manually.

One more question: how to correctly use `lib` folder that is generated by `platformio init`.

ivankravets, 2015-11-19:
> One more question: how to correctly use lib folder that is generated by platformio init.

This information is located in `lib/readme.txt`, for example,  https://github.com/platformio/platformio/blob/develop/examples/wiring-blink/lib/readme.txt

madhead, 2015-11-19:
Not sure why that didn't work for me for the first time, but now it looks ok.

Thank you guys for the help!

ivankravets, 2015-11-20:
Thank you too that using @PlatformIO! Don't forget to follow us on [Twitter](https://twitter.com/PlatformIO_Org).

closed 2015-11-20 by ivankravets