platformio/platformio-core #330: appstate.json is reset after multiple simultaneous compiles + cancellation

mplewis, 2015-11-19:
Hello! We (@punchthrough) use Platform.io for a cloud compiler service. Unfortunately, we have been consistently running into an issue with Platform.io.

Our stack looks like this:

[image]

When a user submits an Arduino sketch to be compiled, our Flask app does the following:

1. Validates the request
2. Checks the target board in the request (Bean or Bean+)
3. Creates a Platform.io project directory and puts the sketch into the `src` directory
4. Creates `platformio.ini` that specifies the target board from the request data
5. Calls `platformio run` in the project directory
6. Reads the output hex file
7. Returns the hex to the client

Logs from a good compile look like this:
```
App 102 stdout: [Wed Nov 18 23:35:50 2015] Processing bean (platform: atmelavr, board: lightblue-bean, framework: arduino)
App 102 stdout: --------------------------------------------------------------------------------
App 102 stdout: ............................................................................
App 102 stdout: ========================= [SUCCESS] Took 3.86 seconds =========================
App 102 stdout:
```

`appstate.json` looks like this during a good compile:
```
{
    "installed_packages": {
        "framework-arduinoavr": {
            "time": 1447888924,
            "version": 22
        },
        "tool-avrdude": {
            "time": 1447888919,
            "version": 3
        },
        "tool-micronucleus": {
            "time": 1447888924,
            "version": 1
        },
        "toolchain-atmelavr": {
            "time": 1447888919,
            "version": 2
        }
    },
    "installed_platforms": [
        "atmelavr"
    ],
    "last_check": {
        "libraries_update": 1447888881,
        "platformio_upgrade": 1447888881,
        "platforms_update": 1447888881
    },
    "last_version": "2.2.2",
    "settings": {
        "enable_prompts": false
    }
}
```

We've been able to reproduce an issue that seemingly causes `~/.platformio/appstate.json` to be reset to default settings. Here are our steps to reproduce:

1. Start a compile from the web by sending the .ino data to the Flask server
2. Cancel the request by killing `curl` before the request completes
3. If compiles are still working, repeat steps 1 and 2

After step 2, logs from all compiles look like this:

```
App 246 stdout:
App 246 stdout: [Wed Nov 18 23:35:54 2015] Processing bean (platform: atmelavr, board: lightblue-bean, framework: arduino)
App 246 stdout: --------------------------------------------------------------------------------
App 246 stdout: The platform 'atmelavr' has not been installed yet. Would you like to install it now? [y/N]:
```

And `appstate.json` looks like this:

```
{
    "last_check": {
        "libraries_update": 1447889752,
        "platformio_upgrade": 1447889752,
        "platforms_update": 1447889752
    },
    "last_version": "2.2.2",
    "telemetry": {
        "backup": []
    }
}
```

We aren't sure what is causing this issue. We know that we can consistently cause it to occur by starting a compile, then canceling it before the request completes.

I suspect there may be some state in Platform.io that is not cleaned up if the app is forcefully quit, but I am looking forward to seeing what you think. Thanks for your help.

ivankravets, 2015-11-19:
Dear Matthew,

Thanks a lot that using PlatformIO :+1:  
Initially, PlatformIO was developed to be used as cloud compiler for @smartanthill. I will try to explain all tricks how to use it correctly. 

1. Please don't use "Platform.io", I don't know who they are. From their site "Open Real-Time Advertising Platform". Please use `PlatformIO` instead. Thanks! 
2. From your diagram I see that you initialize new PlatformIO-based project for each user's call. This isn't good for performance. The first trick is built-in incremental building logic. PlatformIO scans for the files from your project and calculate checksum for them. If the structure of the project is modified, it will automatically clean-up appropriate `.pioenvs/X` directory. Use pre-compiled frameworks, libs between sessions.
3. `Creates a Platform.io project directory and puts the sketch`... If you have only 1 source file which modifies, then use the one project per all calls. In this case, PlatformIO will recompile only this 1 source and libraries (if they have been changed or added/removed within project). The frameworks, libs and etc - will not be re-compiled. It should significantly increase build time.
4. As for the issue with `appstate.json`. You use 2.2.2 version. This bug was fixed in PlatformIO 2.3.0. See https://github.com/platformio/platformio/commit/0d8d5fdb32872b593ce18d5d2daa1063d39099e4. Please upgrade to latest stable PlatformIO 2.3.5
5. You should not afraid to use the latest version of PlatformIO. The HIGH LEVEL API (from the user's side, like `platformio.ini`, project structure and requirements) has not been changed since the 0.0.1 version. This is our main aim to be compatible with the PlatformIO-based projects from the different versions.

If you need additional help for the infrastructure where PlatformIO is used, I can provide consulting  totally for free. Contact me@ikravets.com.

mplewis, 2015-11-19:
Sorry, that was my mistake. I am using platformio.org, not platform.io :)

I'll take a look at the suggestions you made. It's great to know about the philosophy of PlatformIO and how it should be used.

I will close this ticket for now. If I have further questions, would you prefer I emailed you or reopened this ticket?

Thanks for the help!

ivankravets, 2015-11-19:
> I will close this ticket for now. If I have further questions, would you prefer I emailed you or reopened this ticket?

Just re-open this issue. This information is useful for the rest PlatformIO users. Thanks!

mplewis, 2015-11-19:
I am glad you noticed that we froze PlatformIO to version 2.2.2. I did that because we had problems in the past upgrading from 2.0.11.1. Something changed in the way error output was presented, and it broke our parser. [Here's an example](https://gist.github.com/mplewis/1ead3dde21e5c1f9bf9a) of our old and new parser, along with old and new error output.

I realize parsing stdout is a pretty fragile way of doing things. Is there a way to get output that is more machine-readable?

ivankravets, 2015-11-19:
How do you run PlatformIO? I see that verbose level is decreased.http://docs.platformio.org/en/latest/userguide/cmd_run.html#cmdoption-platformio-run-v

The run command should look like
```
platformio -f run 
```

where `-f/--force` will "agree" all question for install missed libs, platforms and etc.

mplewis, 2015-11-19:
We manually run `platformio settings set enable_prompts no` on install to disable prompts. Then, to call PlatformIO, we run `platformio run -v` with the working directory set to the project folder.

mplewis, 2015-11-19:
I never realized that using `-v` could *decrease* verbosity. I assumed it always increased it! I'll try `-vvv` and see how that does.

ivankravets, 2015-11-19:
1. `platformio settings set enable_prompts no` is good. However, `platformio -f CMD` is more safer.
2. `platformio -f run -vvv` is equal to `platformio -f run` :smile:

mplewis, 2015-11-19:
Awesome! I will switch to simply using `platformio -f CMD` then.

ivankravets, 2015-11-19:
Good! I'll wait for any feedbacks from you :+1: :beer:

ivankravets, 2015-11-19:
The 1 request with build process should take < 1-2sec :smile:

ivankravets, 2015-11-19:
The 1 request with build process should take < 1-2sec :smile: 

Try this 
```
platformio init -b lightblue-bean
cp test.ino src/

# the first build will take ~3-5sec
platformio -f run

# Step 1: modify src/test.ino

# Step 2: the second build will take < 1-2sec
platformio -f run

# repeat 1,2 N times :)
```

mplewis, 2015-11-19:
I noticed you mentioned we should use one project directory and change the sketch source files in that directory. I tried this and it does make our builds go faster! However, does this mean we can't compile two different sketches concurrently since they share the same directory?

We are making modifications to the Cloud Compiler to build sketches for two different hardware targets. A sketch only ever needs to be built for one target, so we would like to save time by only building the sketch for the target requested by the client.

My current approach is to create two project directories, one with `platformio.ini` for target A and one for target B. Then, move the source file into the directory that corresponds to the target, build, and retrieve the hex from there. Does that solution sound reasonable?

ivankravets, 2015-11-19:
> I noticed you mentioned we should use one project directory and change the sketch source files in that directory. I tried this and it does make our builds go faster! However, does this mean we can't compile two different sketches concurrently since they share the same directory?

You should write simple logic based on https://pypi.python.org/pypi/lockfile:
1. User makes request to build `test.ino`
2. `CACHE_DIR="/dev/shm/piocache"` Check if `CACHE_DIR` exists, if not then create dir `CACHE_DIR`. I recommend to use shared RAM. It will take ~1-2Mb
3. `MAX_CONCURRENT_WORKERS = N`, then list available projects via `os.listdir(CACHE_DIR)` and find non-locked project. If project has not been found and num project < MAX_CONCURRENT_WORKERS, `platformio init`, LockFile().acquire().

You system will automatically initialize new projects until < MAX_CONCURRENT_WORKERS, or wait while one of MAX_CONCURRENT_WORKERS will be available.

Is these steps clear for you?

> My current approach is to create two project directories, one with platformio.ini for target A and one for target B. Then, move the source file into the directory that corresponds to the target, build, and retrieve the hex from there. Does that solution sound reasonable?

It is so close to `CACHE_KEY`. If target `lightblue-bean`, then `CACHE_KEY=LB`, or if target `my-board`, then `CACHE_KEY=MB`. Then add this key to project's dir as prefix, like https://docs.python.org/2/library/tempfile.html#tempfile.mkdtemp , where `tempfile.mkdtemp(prefix=CACHE_KEY, dir=CACHE_DIR)`

mplewis, 2015-11-20:
Thanks for the tip @ivankravets. I spent today implementing the resource pool pattern and it is working great!

closed 2015-11-20 by mplewis