platformio/platformio-core #331: Won't compile a library that compiles without issue with Arduino IDE

al-uminium, 2015-11-20:
Reproduce Steps:
======
1. `platformio init` a new project
2. Download https://github.com/mysensors/Arduino/archive/development.zip
3. Copy `libraries/*` into new project `lib` folder 
4. Copy example from zip file's `libraries\MySensors\examples\GatewaySerial\GatewaySerial.ino` into new project `src` directory
5. `platformio run`

Expected Result
======
Project compiles
Works in Arduino IDE ver 10606

Real Result
======
Project fails to compile

valeros, 2015-11-23:
Hi @al-uminium!

Looks like MySensor library needs unusual building logic, the same as for Souliss library in #298 
Could you try to add in your `platformio.ini` next options:
```
build_flags = -I/path/to/lib/MySensors
lib_ignore = MySensors
```

al-uminium, 2015-11-23:
YES. Good. YAY. Issue can be close.
Did I not see documentation?

ivankravets, 2015-11-24:
Dear @al-uminium, I will add it to docs.

ivankravets, 2015-11-24:
Added to FAQ http://docs.platformio.org/en/latest/faq.html#can-not-compile-a-library-that-compiles-without-issue-with-arduino-ide

closed 2015-11-24 by ivankravets

ivankravets, 2016-07-15:
Resolved in https://github.com/platformio/platformio/issues/289

commit d63f6cff (2015-11-24): Update FAQ & History // Resolve #331