platformio/platformio-core #332: How to manually upgrade a platform myself?

celsworth, 2015-11-21:
Hello,

I have been digging for a while but I can't find how platformio does this so I can replicate it. Essentially I want to update the `framework-arduinoespressif` platform to the latest git HEAD (or the 2.0.0rc1 they released a few days ago).. there's a lot of new good stuff going in that I'd like to try with platformio building (Arduino IDE is just far too limiting to even consider).

Is there a simple way to do this? I initially wondered if I could just git pull into the correct place, but I can see there is at least one directory (sdk) that isn't in their git repository, but is in the platformio directory - so how is this directory structure created please?

thanks :)

celsworth, 2015-11-21:
Actually I think I found it, the only missing directory was sdk, and that's just in the tools directory, moved it up a level and trying to build a few things now.

ivankravets, 2015-11-21:
By default all PlatformIO packages are located in http://docs.platformio.org/en/latest/projectconf.html#home-dir

Have you resolved this issue?

celsworth, 2015-11-21:
Yes I've got it working, please feel free to close, sorry for the spam. The only bit I was missing was `tools/sdk` had to be moved to `sdk` in `framework-arduinoespressif` (and I had to make a versions.txt to squash an error) .. now I can play with the latest features in platformio environment :)

ivankravets, 2015-11-21:
You can ping @valeros when new release of `framework-arduinoespressif` will be available. He will update it.

Have a nice coding with @PlatformIO :+1:

ivankravets, 2015-11-21:
I've forgotten! Please give us a star [here](https://github.com/platformio/platformio/stargazers) :blush: It helps us to understand our audience. Thanks!

closed 2015-11-21 by ivankravets

celsworth, 2015-11-21:
I'll ping when there's a new public release, its still just RC. Star given! Nice work with platformio and thanks :+1: