platformio/platformio-core #334: simulAVR and gdb-avr missed in toolchain-atmelavr-linux

OmarBelkhiria, 2015-11-22:
When I tried to configure external tools before debugging my .cpp code (Eclipse), I didn't find 'simulavr' and 'gdb-avr' files at this folder: ${HOME}/.platformio/packages/toolchain-atmelavr/bin/
I use Ubuntu 14.04 64-bit.

valeros, 2015-11-24:
Hi @OmarBelkhiria 
`avr-gdb` is restored, please update packages via `platformio update`
http://docs.platformio.org/en/latest/installation.html#development-version

ivankravets, 2015-11-30:
@OmarBelkhiria please re-open this issue if you still have ant problems

closed 2015-11-30 by ivankravets

OmarBelkhiria, 2015-12-01:
Thank you @valeros but  'simulavr' is still not included. How to re-open the issue?

valeros, 2015-12-01:
@OmarBelkhiria Please, update your packages via `platformio update`

OmarBelkhiria, 2015-12-02:
@valeros it's all right now; thanks.