platformio/platformio-core #335: moteinomega board fails to build

karrots, 2015-11-23:
Build for the board `moteinomega` fails with the following error. Builds for the `moteino` board work fine.

```
In file included from src/tmp_ino_to.cpp:1:0:
.pioenvs/moteinomega/FrameworkArduino/Arduino.h:249:26: fatal error: pins_arduino.h: No such file or directory
#include "pins_arduino.h"
```

closed 2015-11-24 by valeros

valeros, 2015-11-24:
Hi @karrots 
Please update packages via `platformio update` and install development version of PlatformIO
http://docs.platformio.org/en/latest/installation.html#development-version

karrots, 2015-11-24:
Dev version returns the message below when build is attempted.

```
.pioenvs/moteinomega/libFrameworkArduino.a(main.o): In function `main':
/home/username/arduino/weathermote/.pioenvs/moteinomega/FrameworkArduino/main.cpp:37: undefined reference to `setup'
/home/username/arduino/weathermote/.pioenvs/moteinomega/FrameworkArduino/main.cpp:47: undefined reference to `loop'
```

ivankravets, 2015-11-24:
@karrots have you implemented `setup()` and `loop()` prototypes?

karrots, 2015-11-24:
Never had to before with Arduino. Worked fine before I updated to dev
version just now.

ivankravets, 2015-11-24:
Please post the source code here.

P.S: Don't forget to give us the :star: https://github.com/platformio/platformio/stargazers

karrots, 2015-11-24:
https://github.com/LowPowerLab/RFM69/blob/master/Examples/WeatherNode/WeatherNode.ino

Builds for `motenio` board worked fine as mentioned in the opening message. Now both `motenio` and `moteniomega` return this message. Source builds find in the Arduino IDE.

ivankravets, 2015-11-24:
Have you placed this `WeatherNode.ino` to `src` directory?

valeros, 2015-11-24:
@karrots Can you share your entire project as zip file somewhere?

karrots, 2015-11-24:
ok your correct. I had changed the src_dir in the platform.ini to make it compatible with the IDE. Didn't realize I messed it up. Works fine now.

commit 238ce2e8 (2015-11-24): fix moteinomega configuration // Resolve #335