platformio/platformio-core #336: click 6.0 breaks `platforms install`

mplewis, 2015-11-24:
[click 6.0 was released today](https://github.com/mitsuhiko/click/blob/master/CHANGES). Since PlatformIO selects `click>=3.2` in requires.txt, a fresh install of PlatformIO today will see this conflict.

You can reproduce this issue by installing PlatformIO with this requirements.txt:

```
platformio==2.3.5
click==6.0
```

Failure occurs when you run `platformio platforms install atmelavr`:

```bash
$ platformio platforms install atmelavr

********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- give us a star on GitHub > https://github.com/platformio/platformio
- vote for the new features/issues on Bountysource > https://www.bountysource.com/teams/platformio/issues
********************************************************************************

Installing toolchain-atmelavr package:
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 695, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 675, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1039, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1039, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 868, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 513, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/dist-packages/platformio/platforms/base.py", line 302, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 102, in install
    dlpath = self.download(info['url'], pkg_dir, info['sha1'])
  File "/usr/local/lib/python2.7/dist-packages/platformio/pkgmanager.py", line 45, in download
    fd.start()
  File "/usr/local/lib/python2.7/dist-packages/platformio/downloader.py", line 71, in start
    with click.progressbar(length=chunks, label="Downloading") as pb:
  File "/usr/local/lib/python2.7/dist-packages/click/_termui_impl.py", line 94, in __enter__
    self.render_progress()
  File "/usr/local/lib/python2.7/dist-packages/click/_termui_impl.py", line 222, in render_progress
    buf.append(' ' * (clear_width - line_len))
UnboundLocalError: local variable 'clear_width' referenced before assignment
```

A temporary fix is to pin click to 5.1 in your requirements.txt:

```
platformio==2.3.5
click==5.1
```

ivankravets, 2015-11-24:
Hm... I can't reproduce this issue :(

```
(22)➜  /tmp  platformio platforms uninstall atmelavr teensy
Uninstalling tool-avrdude package: 	[OK]
Uninstalling framework-arduinoavr package: 	[OK]
Uninstalling tool-micronucleus package: 	[OK]
The platform 'atmelavr' has been successfully uninstalled!
Uninstalling framework-arduinoteensy package: 	[OK]
Uninstalling tool-teensy package: 	[OK]
Uninstalling toolchain-atmelavr package: 	[OK]
The platform 'teensy' has been successfully uninstalled!
(22)➜  /tmp  platformio platforms install atmelavr teensy
Installing toolchain-atmelavr package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing tool-avrdude package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing framework-arduinoavr package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing tool-micronucleus package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
The platform 'atmelavr' has been successfully installed!
Installing framework-arduinoteensy package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing tool-teensy package:
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Installing toolchain-gccarmnoneeabi package:
Already installed
Installing framework-mbed package:
Already installed
Installing toolchain-atmelavr package:
Already installed
Installing ldscripts package:
Already installed
The platform 'teensy' has been successfully installed!
(22)➜  /tmp  pip list
bottle (0.12.9)
click (6.0)
lockfile (0.11.0)
pbr (0.11.0)
pip (7.1.2)
platformio (2.3.5)
pyserial (2.7)
requests (2.8.1)
scons (2.4.1)
setuptools (18.2)
wheel (0.24.0)
(22)➜  /tmp
```

P.S: Also, tested in Windows 7, can't reproduce it.

ivankravets, 2015-11-24:
I've just re-run a tests on CI witch @Click 6.0:

* https://circleci.com/gh/platformio/platformio/338
* https://travis-ci.org/platformio/platformio/jobs/92990445

ivankravets, 2015-11-25:
Fixed in https://github.com/mitsuhiko/click/commit/b47b11755f45788878b7206bcbbceb4a324b9906

closed 2015-11-25 by ivankravets

ivankravets, 2015-11-27:
Click 6.1 is out https://github.com/mitsuhiko/click/blob/master/CHANGES#L11where this is issue is fixed.