platformio/platformio-core #338: How to specify a custom core

foufee, 2015-11-26:


ivankravets, 2015-11-26:
1. See [Custom Board](http://docs.platformio.org/en/latest/platforms/creating_board.html) doc. You can use @mplewis's code https://github.com/platformio/platformio/blob/develop/platformio/boards/punchthrough.json
2. Put your core to `~/.platformio/packages/framework-arduinoavr/cores` and specify it for board https://github.com/platformio/platformio/blob/develop/platformio/boards/punchthrough.json#L11

ivankravets, 2015-11-26:
Don't forget that your "core" will be removed when new  `framework-arduinoavr` will be available. In this case I recommend to use symlink and keep own "core" outside @PlatformIO's packages.

mplewis, 2015-11-27:
When we build our server container with PlatformIO included, it always starts by installing PlatformIO, then installing AtmelAVR, then copying custom cores. That way we are always sure that our core hasn't been overwritten by an update of arduinoavr framework.

ivankravets, 2015-12-10:
@foufee please reopen this issue if you still need help

closed 2015-12-10 by ivankravets