platformio/platformio-core #339: missing library for esp8266 ?

krissfr, 2015-11-26:
Hi,

i use function strtoul() that doesn't compile fine for the esp8266 board with platformio.

On my IDE Arduino with the  staging version of https://github.com/esp8266/Arduino , it compil fine.

Is there a way to tell to platformio to use the staging version of https://github.com/esp8266/Arduino for compiling on esp8266 ?

thanks

valeros, 2015-11-27:
Hi @krissfr! 
You can try next steps for esp-01 board (and others with 512KB flash):
1. Temporarily install special version of PlatformIO:
`pip install https://github.com/platformio/platformio/archive/feature/issue-339-espressif-rc-framework.zip`
2. Download staging version of framework from here:
`https://github.com/esp8266/Arduino/releases/download/2.0.0-rc2/esp8266-2.0.0-rc2.zip`
3. Copy content from the archive to `framework-arduinoespressif` in PlatformIO home directory:
`~/.platformio/packages/framework-arduinoespressif`
4. Create file `version.txt` and write there this text: `2.0.0`

`framework-arduinoespressif` package should have this structure:
├── bootloaders
├── cores
├── libraries
├── tools
├── variants
├── version.txt
├── programmers.txt
├── platform.txt
└── boards.txt

6. Copy content from here `https://gist.github.com/valeros/8ad9019dce8eae7e3dd9` to `esp8266.flash.512k.ld` file in  `~/.platformio/packages/ldscripts`

7. Try `platformio run`

*Don't forget that your modifications will be removed when new versions will be available.

krissfr, 2015-11-27:
Hi Valeros,

Thank you for your reply :)

I have a nodemcu 0.9 . Does it matter ?

Actually in the arduino IDE, i select the nodemcu 0.9 board (4M of flash) and i compile fine with the staging version of  https://github.com/esp8266/Arduino .

thanks

valeros, 2015-11-27:
Yes, board type and flash size are important.
I've updated my comment above for your board.

skorokithakis, 2015-11-27:
@valeros, sorry for hijacking this, I just got a NodeMCU and have been trying all day to deploy a program with PlatformIO, but it has been crashing a lot. In the end, I tried the Arduino IDE, which worked like a charm. Would this bug be applicable to me for the crashes, or should I file another one?

valeros, 2015-11-28:
Hi @skorokithakis!
You definitely should try, it might help.

krissfr, 2015-11-29:
Hi Valeros,

How to your workaround with the Continuous Integration system ?
I use the shippable yml ( http://docs.platformio.org/en/latest/ci/shippable.html ) . Do i need to write specific lines in this yml ?

thanks

ivankravets, 2015-11-29:
Hi @krissfr,

Does @valeros's solution work for you? If yes, we will make 2.3.6 release and you won't need to write specific lines in Shippable YML.

ivankravets, 2015-11-29:
@skorokithakis do you mean that your code works fine with Arduino IDE? Do yo use the same framework version?

skorokithakis, 2015-11-29:
As a data point, I fixed a bunch of problems and crashes by copying the Arduino SDK's stable directory to the `framework-arduinoespressif` directory. It seems that the version bundled with PlatformIO is not as stable as the Arduino one.

ivankravets, 2015-11-29:
@skorokithakis PlatformIO proposes stable frameworks. This issue is intended for users who want to try unstable/RC versions. Please perform these commands:
```
platformio platforms uninstall espressif
platformio platforms install espressif
```

You should have stable packages for `espressif` platform.

skorokithakis, 2015-11-29:
To recap, because I'm confusing everyone posting disparately: I got a NodeMCU, tried deploying various scripts with PlatformIO but they would randomly crash. After two days of debugging, finally copying the stable version of the platform from the Arduino IDE to PlatformIO has been working perfectly ever since. Using the same version as the Arduino IDE in PlatformIO has solved many problems for me, so I propose that PlatformIO use the same version as the Arduino IDE.

skorokithakis, 2015-11-29:
@ivankravets, yes, sorry, I hijacked this comment and confused people. Please disregard my comments here.

krissfr, 2015-11-29:
@ivankravets @valeros Solution's seem ok ( i have other problem, but  no relation with specific esp board libraries). Can you please tell me when the 2.3.6 release will be available ?

Thank you @valeros and @ivankravets

ivankravets, 2015-11-30:
> Can you please tell me when the 2.3.6 release will be available ?

I hope that tomorrow.

krissfr, 2015-11-30:
@ivankravets thank you :)

ivankravets, 2015-11-30:
Please don't forget to grunt us with a star here https://github.com/platformio/platformio/stargazers for our efforts.

Thanks!

krissfr, 2015-11-30:
done :)

ivankravets, 2015-12-01:
Resolved in https://github.com/platformio/platformio/issues/345

closed 2015-12-01 by ivankravets