platformio/platformio-core #343: Some combination of the ESP8622 Arduino HEAD and PlatformIO's linkers break the NodeMCU

skorokithakis, 2015-11-29:
I installed the latest Arduino HEAD to PlatformIO, but everything after a specific commit (that looks like it would break the linker) crashes the NodeMCU on startup.

Related bug report (that ends up working fine in the Arduino IDE):

https://github.com/esp8266/Arduino/issues/1101

valeros, 2015-11-30:
Hi @skorokithakis 
I've just tested NodeMCU board with PlatformIO and Arduino HEAD  (WifiScan example) and I didn't have any crashes.
Maybe you can provide your project as zip somewhere?

skorokithakis, 2015-11-30:
It was literally just the sample sketch here: https://github.com/esp8266/Arduino/issues/1101#issuecomment-160314168. If you want, you can close this issue and I will reopen it if I manage to reproduce it. I was able to reproduce it reliably and bisect it to that specific commit before replacing everything with stable, though.

skorokithakis, 2015-11-30:
I just tried using the Arduino version 2.0.0 and can confirm that my NodeMCU won't boot when compiling with it.

skorokithakis, 2015-11-30:
I can confirm that flashing with the Arduino SDK works fine. I'll wait until tomorrow to test it with 2.0.0 before continuing.

valeros, 2015-11-30:
@skorokithakis thanks for your help.
It will be great if you test PlatformIO with your board before release.

skorokithakis, 2015-11-30:
Can you give me the instructions for that? Do I just install the latest master tarball? Do I need to update platforms afterwards?

skorokithakis, 2015-11-30:
I installed the latest HEAD from develop, I get this crash (it can't find the platform files and is not offering to upgrade):

~~~
IOError: [Errno 2] No such file or directory: '/home/stavros/.platformio/packages/framework-arduinoespressif/version.txt':
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/platformio/builder/main.py", line 162:
SConscript(env.subst("$BUILD_SCRIPT"))
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/platformio/builder/scripts/espressif.py", line 189:
target_elf = env.BuildProgram()
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 49:
env.BuildFramework()
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/platformio/builder/tools/platformio.py", line 186:
"%s.py" % framework))
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/scons-2.4.1/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/home/stavros/.virtualenvs/generic/lib/python2.7/site-packages/platformio/builder/scripts/frameworks/arduino.py", line 87:
"version.txt")).read().replace(".", "").strip())
~~~

valeros, 2015-11-30:
Yes, tomorrow you will need to install PlatformIO from develop branch and update packages in you home directory. I will give you download links with new packages as soon as I've tested packages myself with my boards.

skorokithakis, 2015-11-30:
Sorry, the above error was because I didn't remove the `appstate.json`. It's upgrading now.

skorokithakis, 2015-11-30:
Looks like version 2.0.0 isn't available yet. Let me know what you want me to test and I'll get to it tomorrow.

ivankravets, 2015-12-01:
Resolved in https://github.com/platformio/platformio/issues/345

closed 2015-12-01 by ivankravets