platformio/platformio-core #345: Update framework-arduinoespressif to 2.0.0

celsworth, 2015-11-30:
esp8266/Arduino have released 2.0.0: https://github.com/esp8266/Arduino/releases/tag/2.0.0

Could we get the platformio package updated please :)

valeros, 2015-11-30:
Hi @celsworth thanks for your prompt!
Sure, the package will be updated tomorrow with new PlatformIO 2.4.0 release.

celsworth, 2015-11-30:
Thanks :)

One thing I wanted to ask, in the 2.0 changelog they have:

    Fix placement of code into RAM, enable gc-sections

But from what I see, this project uses different ldscripts - will you integrate those changes as well? It should fix some out of space issues with irom0.

skorokithakis, 2015-11-30:
+1, hopefully this will work fine for me. I can test #343 when this gets updated.

valeros, 2015-11-30:
We will use original ldscripts from Arduino framework with some minor modifications.

closed 2015-12-01 by ivankravets

valeros, 2015-12-01:
Hi @skorokithakis @celsworth @krissfr!
I would like to ask you guys to test latest PlatformIO with new packages before the stable release.
Please install latest develop branch:
`http://docs.platformio.org/en/latest/installation.html#d-development-version`
Update packages in your home directory:
https://www.dropbox.com/s/5z67p6kb3lakcu4/ldscripts.zip?dl=0
https://www.dropbox.com/s/qy07ihoq16y7cdc/framework-arduinoespressif.zip?dl=0
Looking forward to your results!

skorokithakis, 2015-12-01:
Mine crashes because `version_to_int` can't handle a version that is `2.4.b1` somewhere.

celsworth, 2015-12-01:
Exactly the same here. I tried changing to 2.4.0b1 in appstate.json, but it must be elsewhere too as its still not happy:

```
ValueError: invalid literal for int() with base 10: '':
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 162:
SConscript(env.subst("$BUILD_SCRIPT"))
File "/usr/local/lib/scons-2.3.6/SCons/Script/SConscript.py", line 614:
return method(*args, **kw)
File "/usr/local/lib/scons-2.3.6/SCons/Script/SConscript.py", line 551:
return _SConscript(self.fs, *files, **subst_kw)
File "/usr/local/lib/scons-2.3.6/SCons/Script/SConscript.py", line 260:
exec _file_ in call_stack[-1].globals
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/scripts/espressif.py", line 192:
target_elf = env.BuildProgram()
File "/usr/local/lib/scons-2.3.6/SCons/Environment.py", line 224:
return self.method(*nargs, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 78:
*pioversion_to_intstr())]
File "/usr/local/lib/python2.7/dist-packages/platformio/util.py", line 137:
return [int(i) for i in vermatch.group(1).split(".")[:3]]
```

skorokithakis, 2015-12-01:
I grepped, it's getting it from the internet again and setting it in. I gutted half the upgrader script but still no luck.

valeros, 2015-12-01:
Sorry, fixed.
Please test it again.

celsworth, 2015-12-01:
Thats better! Builds now, thanks. Will have a play around and report back

skorokithakis, 2015-12-01:
Still the same problem here. I installed develop again but no luck. What else do I need to do?

celsworth, 2015-12-01:
I don't think I did anything special, just pip uninstall then pip reinstall. You might try blowing away your `.platformio` directory and starting again?

@valeros this looks real good. Everything I tried works so far, including OTA updating which didn't work on platformio last night, so good job :)

skorokithakis, 2015-12-01:
Alright, I deleted the entire dir and it's running now. Let's see...

skorokithakis, 2015-12-01:
So, the previous version didn't work with even a simple sketch that did nothing (it would dump core). This version works perfectly, OTA seems to work fine, although I'm having some hiccups where the ESP won't come back up after an OTA upgrade reset. I don't think it's a platformio problem, as I was kind of seeing the same with the Arduino IDE last night, but with the Arduino it worked reliably after some point, whereas I can't get it to work reliably now.

skorokithakis, 2015-12-01:
Yeah, I can't get FOTA to work. It looks like the uploaded firmware is corrupt, as it won't boot.

celsworth, 2015-12-01:
Getting a little offtopic for here now, but does the BasicOTA example work from the esp8266/Arduino repository?

skorokithakis, 2015-12-01:
Yep, I tried that last night and it started working reliably after one or two tries. I'm seeing issues now where even non-OTA resets won't boot properly.

skorokithakis, 2015-12-01:
Hmm, scratch that, a reboot just succeeded. Let me investigate further and report back.

skorokithakis, 2015-12-01:
Yep:

~~~
Connected to Wifi.

Connecting to MQTT...

fatal: MQTT server connection failed. Rebooting.

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
~~~

And then it just sits there.

skorokithakis, 2015-12-01:
Never mind, the same thing happens with the Arduino IDE, so it's not a platformio issue.

skorokithakis, 2015-12-01:
OTA still doesn't work for me, as far as I can tell the resulting image is corrupt, or I'm flashing it wrong.

valeros, 2015-12-01:
@skorokithakis Does OTA with Arduino IDE work for you?

skorokithakis, 2015-12-01:
Yes, that one works fine.

valeros, 2015-12-01:
@skorokithakis  What's your board version?

skorokithakis, 2015-12-01:
Sorry, I'm an idiot :( `platformio run` output `firmware.elf` so I thought that was the one I should flash. `firmware.bin` works with OTA, so everything works fine for me too.

commit 9f4b73d8 (2015-12-01): Update Arduino core for Espressif platform to 2.0.0 // Resolve #345