platformio/platformio-core #346: serialports monitor ignores parameters (Python < 3)

MrSnowflake, 2015-12-01:
Hi,

The options for the serial monitor are ignored when using Python < 3. (I use 2.7). The following command should open a serial monitor with baudrate 115200:

    platformio serialports monitor -b 115200

But it opens as 9600, which is the default value.

The bug appears on line 159 in file /commands/serialports.py. miniterm.main() should have the option values set as its arguments, but it has no arguments.

MrSnowflake, 2015-12-01:
It seems to be something with the parsing of the options in serial.tools.miniterm. But I can't figure out what exactly.

ivankravets, 2015-12-01:
Hi @MrSnowflake,

This issue is linked with the latest Click 6.0. I will fix it as soon as possible.

Temporary solution:
```
pip uninstall click
pip install "click<6"
```

ivankravets, 2015-12-01:
Also, don't forget about :star: ( https://github.com/platformio/platformio/stargazers ) for @valeros :blush:

ivankravets, 2015-12-01:
I've just opened related issue https://github.com/mitsuhiko/click/issues/473

If this isn't Click's bug, I'll drop Click version to "click<6".

ivankravets, 2015-12-01:
Fixed in 2.4.0

closed 2015-12-01 by ivankravets