platformio/platformio-core #347: ESP8266_DirectConnection.ino example does not compile

lbuchy, 2015-12-01:
Having problems building the example code with this library.  Specifically I am using the ESP8266_DirectConnection.ino example file:

```c
#define BLYNK_USE_DIRECT_CONNECT
//#define BLYNK_DEBUG
#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <ESP8266WiFi.h>
#include <BlynkSimpleUserDefined.h>

// Uncomment this to set WiFi to Access Point mode
#define WIFI_MODE_AP

// Set these to your credentials.
const char ssid[]     = "Blynk-AP";    // WiFi name
const char password[] = "12345678";   // WiFi password

// Set port of the server
const int port = 8442;

// Auth token doesn't matter with direct-connect
char auth[] = "";

// Description of the dashboard
char profile[] = R"json(
  {
    "dashBoards": [
      {
        "id": 1,
        "name": "Direct connect",
        "boardType": "ESP8266",
        "widgets": [
          {
            "id": 2,
            "type": "DIGIT4_DISPLAY",
            "pinType": "VIRTUAL",
            "pin": 9,
            "x": 5,
            "y": 1,
            "frequency": 1000
          },
          {
            "id": 3,
            "type": "BUTTON",
            "pinType": "VIRTUAL",
            "pin": 1,
            "x": 2,
            "y": 1
          }
        ]
      }
    ]
  }
)json";

// Virtual handlers for our widgets...

BLYNK_READ(V9) {
  Blynk.virtualWrite(9, millis() / 1000);
}

BLYNK_WRITE(V1) {
  BLYNK_LOG("Button event");
}

// Next goes the hard stuff: connection management, etc...

WiFiServer server(port);
WiFiClient client;

// This function is used by Blynk to receive data
size_t BlynkStreamRead(void* buf, size_t len)
{
  return client.readBytes((byte*)buf, len);
}

// This function is used by Blynk to send data
size_t BlynkStreamWrite(const void* buf, size_t len)
{
  return client.write((byte*)buf, len);
}

void setup()
{
  delay(1000);
  // Setup your connection here.
  Serial.begin(9600);
  Serial.println();

  Blynk.begin(auth);
  Blynk.setProfile(profile);

#ifdef WIFI_MODE_AP
  BLYNK_LOG("Configuring WiFi access point: %s", ssid);
  WiFi.softAP(ssid, password);
  Serial.print("AP IP address: ");
  Serial.println(WiFi.softAPIP());
#else
  BLYNK_LOG("Connecting to WiFi %s", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();
  Serial.print("STA IP address: ");
  Serial.println(WiFi.localIP());
#endif

  server.begin();
  BLYNK_LOG("Listening on port: %d", port);
}

void loop()
{
  client = server.available();
  // Make sure that Blynk.run() is called
  // only when the connection is established
  if (client) {
    Serial.print("Client connected from ");
    Serial.print(client.remoteIP());
    Serial.print(":");
    Serial.println(client.remotePort());

    // Wait for connection
    unsigned long start = millis();
    while (!client.connected() && (millis() - start < 1000)) {
      delay(5);
    }

    // Go!!!
    Blynk.startSession();
    while (client.connected()) {
      // Okay, handle Blynk protocol
      bool hasIncomingData = (client.available() > 0);
      // Tell Blynk if it has incoming data
      // (this allows to skip unneeded BlynkStreamRead calls)
      if (!Blynk.run(hasIncomingData)) {
        Serial.print("Error happened or disconnected.");
        break;
      }
    }
    client.stop();
    Serial.println("Client disconnected.");
  }
}
```

Compilation gives the following error:

```
<above omitted>
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/Blynk_ID415/linux/BlynkDebug.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10605 -I/home/lbuchy/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/esp12e/FrameworkArduino -I.pioenvs/esp12e/FrameworkArduinoVariant -I.pioenvs/esp12e/ESP8266WiFi -I.pioenvs/esp12e/Blynk_ID415 -I.pioenvs/esp12e/Blynk_ID415/utility .pioenvs/esp12e/Blynk_ID415/linux/BlynkDebug.cpp
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/Blynk_ID415/linux/main.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -DARDUINO=10605 -I/home/lbuchy/.platformio/packages/framework-arduinoespressif/sdk/include -I.pioenvs/esp12e/FrameworkArduino -I.pioenvs/esp12e/FrameworkArduinoVariant -I.pioenvs/esp12e/ESP8266WiFi -I.pioenvs/esp12e/Blynk_ID415 -I.pioenvs/esp12e/Blynk_ID415/utility .pioenvs/esp12e/Blynk_ID415/linux/main.cpp
.pioenvs/esp12e/Blynk_ID415/linux/main.cpp:15:28: fatal error: BlynkApiLinux.h: No such file or directory
#include <blynkapilinux.h>
^
compilation terminated.
scons: *** [.pioenvs/esp12e/Blynk_ID415/linux/main.o] Error 1
```

It appears that the include directories specified are insufficient. I think this might be that this particular example requires that we compile the linux portion of the library also to create the server.  I managed to get around the issue by specifying the -I.pioenvs/esp12e/Blynk_ID415/linux build_flag but it continues to fail when trying to include <sys/socket.h>.  I guess this would work if there was a posix wrapper that could be added to the project.

I'll maybe take another look after having some coffee.

valeros, 2015-12-02:
Hi @lbuchy 
Temporarily remove `linux` folder in `~/.platformio/lib/Blynk_ID415/`

lbuchy, 2015-12-02:
@valeros Yeah that fixed the compilation.  I haven't had success with connecting the Blynk app with the example but it is besides the point for this issue.

What was the issue?  Let me guess: The build system 'grepped' for all .c files in the library to fill in the list of source files to compile?

ivankravets, 2015-12-02:
@lbuchy I've just excluded `linux` folder from this library, see https://github.com/blynkkk/blynk-library/pull/49

I'll report you when new library will be available in PlatformIO Library Registry.

ivankravets, 2015-12-04:
The Blynk library has been updated. Try `platformio update` command.

Please reopen this issue if you need any help.

closed 2015-12-04 by ivankravets