platformio/platformio-core #349: Switch to Click 6.0

ivankravets, 2015-12-02:
Related issue https://github.com/mitsuhiko/click/issues/473

W-Floyd, 2017-04-12:
Trying to get platformio working on Solus, but, as expected, click is too new. Any news?

ivankravets, 2017-04-12:
@W-Floyd you can use `virtualenv` temporary if you need to mix multiple click versions

W-Floyd, 2017-04-12:
Hmm, I've not used it before, but from what I can tell, it's not adequate for packaging purposes, as a permanent solution. I'll have to check with the devs, but it probably means platformio isn't suitable for packaging on Solus.

olymk2, 2018-07-15:
That would be nice or use argparser its built in and less likely to get breaking changes, kinda want the integration into my current ide thats the appeal of platformio it already has support if you have platformio installed just that my distro has bang up to date packages.

But good to know you will consider I guess a change will be needed at some point if the versioon of click you need will no longer be maintained.

commit 62b80c39 (2019-05-08): Added support for the latest Python "Click" package (CLI Builder) // Resolve #349
closed 2019-05-08 by ivankravets