platformio/platformio-core #351: esptool reset mode

andriyadi, 2015-12-03:
Would be great if we can change the reset mode to be used by esptool, for uploading firmware to esp8266. As we know, esptool supports 2 reset modes now via `-cd` option: ck and nodemcu.

I check in this file `platformio/platformio/builder/scripts/espressif.py`, the `-cd` option is hardcoded to `ck`. Can I change that via platform.ini, or I should the source code of that espressif.py file?

Anyway, great progress so far! Definitely gonna keep using platformio.
Thanks.

ivankravets, 2015-12-03:
Take a look on http://docs.platformio.org/en/latest/projectconf.html#extra-script

See related https://github.com/platformio/platformio/issues/247#issuecomment-118169728

ivankravets, 2015-12-03:
Do you need a help with http://docs.platformio.org/en/latest/projectconf.html#extra-script ?

andriyadi, 2015-12-03:
It would be great if you could provide an example for adding additional options, as my Python skill is very low. But that `extra_script` should do it for now.

If it's not too much for asking, the Arduino for esp8266 framework comes with `boards.txt` file. Is it possible to respect the settings in it? For example:
```
generic.upload.tool=esptool
generic.upload.speed=115200
generic.upload.resetmethod=ck
generic.upload.maximum_size=434160
generic.upload.maximum_data_size=81920
generic.upload.wait_for_upload_port=true
generic.serial.disableDTR=true
generic.serial.disableRTS=true
```

Thanks!

ivankravets, 2015-12-03:
* Create `platformio_extra_script.py` in the project near `platformio.ini` with the content:
```python
from SCons.Script import DefaultEnvironment

env = DefaultEnvironment()

env.Replace(
    MYUPLOADERFLAGS=[
        "-vv",
        "-cd", "nodemcu",
        "-cb", "$UPLOAD_SPEED",
        "-cp", "$UPLOAD_PORT",
        "-ca", "0x00000",
        "-cf", "$SOURCE"
    ],
    UPLOADCMD='$UPLOADER $MYUPLOADERFLAGS',
)
```
* Specify `extra_script`, for example
```ini

[env:esp01]
platform = espressif
framework = arduino
board = esp01
extra_script = platformio_extra_script.py
```

Did it help you?

andriyadi, 2015-12-04:
Wow... Super helpful! Thanks man!

I'll try tonight. Will confirm the result.

ivankravets, 2015-12-05:
Please re-open this issue if you still need help.

closed 2015-12-05 by ivankravets

andriyadi, 2015-12-07:
Sorry for late update. I can confirm that your guidance works. Thanks!

liangzelang, 2016-05-11:
@ivankravets  could you explain the meaning of the  `platformio_extra_script.py`    because  i  don't learn Python.

valeros, 2016-05-11:
Hi @andriyadi 
Why do you need extra script option?
In two words it's just a python code snippet that can interact with build environment.