platformio/platformio-core #355: Getting started CLion errors

orome, 2015-12-05:
After following the instructions for setting up a first project in "Getting Started" and then configuring CLion (1.2) integration I get

```
Error:Configuration Debug
Cannot read project file /Users/Orome/Library/Caches/CLion12/cmake/generated/41b438d2/41b438d2/Debug/learn.cbp
Error:Configuration Release
Cannot read project file /Users/Orome/Library/Caches/CLion12/cmake/generated/41b438d2/41b438d2/Release/learn.cbp
Error:Configuration RelWithDebInfo
Cannot read project file /Users/Orome/Library/Caches/CLion12/cmake/generated/41b438d2/41b438d2/RelWithDebInfo/learn.cbp
Error:Configuration MinSizeRel
Cannot read project file /Users/Orome/Library/Caches/CLion12/cmake/generated/41b438d2/41b438d2/MinSizeRel/learn.cbp
```

Neither deleting these files nor invalidating caches in CLion has any effect on these errors: I get them every time I launch CLion.

ivankravets, 2015-12-05:
How about to delete `/Users/Orome/Library/Caches/CLion12/` ?

orome, 2015-12-05:
Tried that already (see above). They get remade and generate the same errors.

ivankravets, 2015-12-05:
Does some project (not PlatformIO-based) work for you?

orome, 2015-12-05:
All work (I can build), but when I load or open a PlatformIO-based project I get these errors.

ivankravets, 2015-12-05:
Please share your project including `CMakeLists.txt` where I can reproduce this issue. If don't want to share it globally, mail me@ikravets.com

orome, 2015-12-06:
https://github.com/orome/learn-arduino

ivankravets, 2015-12-06:
I've just compiled your project without any problems. See screenshot below.

Try to replace these lines https://github.com/orome/learn-arduino/blob/master/CMakeLists.txt#L4:L5 with 
```
#set(ENV{PATH} "/Users/Orome/.cabal/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/Users/Orome/bin:/opt/X11/bin:/Library/TeX/texbin ")
set(PLATFORMIO_CMD "platformio")
```

[image]

orome, 2015-12-06:
The project *builds* fine. That's not the issue. The issue is that I can't even open CLion on this project without seeing the errors above.

I continue to get the errors after invalidating and deleting caches, and after making the suggested changed.

ivankravets, 2015-12-06:
Sorry, I don't understand your problem. Could you write some video? Thanks.

orome, 2015-12-06:
What happens is if I either (a) open this project from within CLion, or (b) open CLion when it was the last project opened — i.e., no matter how I open the project — *the act of opening the project*, produces these errors.

I don't have to run any other commands or take any other actions. Just opening the project is enough.

ivankravets, 2015-12-06:
Ok, see my video http://cl.ly/1M3B3525091E

orome, 2015-12-06:
My errors appear at about 0:10.

ivankravets, 2015-12-06:
Do you have normal Mac App uninstaller? Try to uninstall CLion IDE and remove these files/folders:
* `/Users/Orome/Library/Saved Application State/com.jetbrains.CLion.savedState`
* `/Users/Orome/Library/Preferences/com.jetbrains.CLion.plist`
* `/Users/Orome/Library/Application Support/CLion***/`
* `/Users/Orome/Library/Caches/CLion***/`

If it wont help you, please report here https://www.jetbrains.com/support/clion/

orome, 2015-12-06:
Actually, even if that does help, I should probably [report it](https://youtrack.jetbrains.com/issue/CPP-5301).

orome, 2015-12-06:
And FWIW, under the Caches tab I see:
```
CMAKE_OBJCOPY-NOTFOUND
CMAKE_OBJDUMP-NOTFOUND
```

orome, 2015-12-07:
@ivankravets — I can confirm that I get these errors after following [the suggestion above](https://github.com/platformio/platformio/issues/355#issuecomment-162357535) and starting from a completely fresh install if CLion 1.2.1 and a completely new project (just `src/maim.cpp` and execution of `platformio init --ide clion --board=uno`).

ivankravets, 2015-12-07:
Could you provide me access to https://www.teamviewer.com ? Please mail me@ikravets.com ID/Pass. Thanks.

orome, 2015-12-07:
Try this: change the name of the folder to something like `Micro & Embedded` and then open it with CLion.

ivankravets, 2015-12-09:
@orome yes, I've just reproduced this issue! Looks that CLion IDE doesn't like "&" in the path. Please report here https://www.jetbrains.com/support/clion/

closed 2015-12-09 by ivankravets

orome, 2015-12-09:
@ivankravets — [Already done.](https://youtrack.jetbrains.com/issue/CPP-5301)