platformio/platformio-core #356: Arduino ZERO as board type?

orome, 2015-12-05:
Is there a way to specify the Arduino ZERO as a `board`?

ivankravets, 2015-12-05:
We will add support for Arduino Zero as soon as possible.

orome, 2015-12-05:
Cool. Thanks!

ivankravets, 2015-12-09:
@orome do you have this board?

orome, 2015-12-09:
@ivankravets — Yes (but I'm a total beginner, so I'm not sure I'll be the best tester — thought I'm happy to try and help — if that's where you're headed).

valeros, 2015-12-18:
Hi @orome 
I've implemented some basic support for this board in special branch and we would be grateful if you could test it with PlatformIO.

Please install special version of PlatformIO:
`pip uninstall platformio`
`pip install https://github.com/valeros/platformio/archive/feature/issue-356-arduino-zero.zip`

Replace `ldscripts` and `framework-arduinosam` packages in PlatformIO home directory with these:
https://dl.dropboxusercontent.com/u/10025728/ldscripts.zip
https://dl.dropboxusercontent.com/u/10025728/framework-arduinosam.zip
Connect your board through the `native USB` port.
Create simple blink project with the board `zeroUSB` and upload firmware.

Looking forward to your results!

orome, 2015-12-18:
@valeros : Cool. I should have time later today for a test (if I can locate the appropriate USB adapter).

orome, 2015-12-18:
@valeros : Where will I find  `ldscripts` and `framework-arduinosam` to replace? My `~/.platformio/packages` contains only: 
```
framework-arduinoavr	
tool-avrdude		
tool-micronucleus	
toolchain-atmelavr
```
Do you mean move rather than replace?

valeros, 2015-12-18:
@orome  `platformio platforms install atmelsam`

orome, 2015-12-18:
@valeros : There they are! So replace after that?

orome, 2015-12-18:
@valeros : Actually, I can't get through `platformio platforms install atmelsam`:
```
nstalling ldscripts package:
Already installed
Installing toolchain-gccarmnoneeabi package:
Downloading  [#############-----------------------]   38%  00:09:11
Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/platformio/__main__.py", line 97, in main
    cli(None, None, None)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/site-packages/platformio/commands/platforms.py", line 40, in platforms_install
    if p.install(with_package, without_package, skip_default_package):
  File "/usr/local/lib/python2.7/site-packages/platformio/platforms/base.py", line 311, in install
    pm.install(name)
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 102, in install
    dlpath = self.download(info['url'], pkg_dir, info['sha1'])
  File "/usr/local/lib/python2.7/site-packages/platformio/pkgmanager.py", line 45, in download
    fd.start()
  File "/usr/local/lib/python2.7/site-packages/platformio/downloader.py", line 73, in start
    f.write(next(itercontent))
  File "/usr/local/lib/python2.7/site-packages/requests/models.py", line 660, in generate
    raise ChunkedEncodingError(e)
ChunkedEncodingError: ("Connection broken: error(54, 'Connection reset by peer')", error(54, 'Connection reset by peer'))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `platformio upgrade` command
* Report this problem to the developers
  https://github.com/platformio/platformio/issues

============================================================
```

ivankravets, 2015-12-18:
Please `platformio upgrade` to the latest 2.6.1

ivankravets, 2015-12-18:
As for `ChunkedEncodingError: ("Connection broken: error(54, 'Connection reset by peer')", error(54, 'Connection reset by peer'))` - I'll fix it.

orome, 2015-12-18:
@ivankravets : Actually, can you spell this out better. I thought I was supposed to use `pip install https://github.com/valeros/platformio/archive/feature/issue-356-arduino-zero.zip`. Now I'm supposed to use the latest?

ivankravets, 2015-12-18:
Sorry, it was my mistake. I've not seen that this is branch in @valeros repo. Don't make any upgrade operations. You are in right branch.

Please try again `platformio platforms install atmelsam`

orome, 2015-12-18:
@ivankravets : Can you spell out the full instructions (all steps to get from a previous installation to where I need to be)? It's totally unclear what code I'm supposed to be testing!

ivankravets, 2015-12-18:
1. ``pip uninstall platformio``
2. ``pip install -U pip install https://github.com/valeros/platformio/archive/feature/issue-356-arduino-zero.zip``
3. ``platformio platforms install atmelsam``
4. Run blink example

orome, 2015-12-18:
@ivankravets : `pip install -U pip install ...`?

ivankravets, 2015-12-18:
Edited. Sorry, it is hard day for me today. Need to go out and relax.

orome, 2015-12-18:
```
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/Roy/Library/Caches/CLion12/cmake/generated/581a5b46/581a5b46/Debug --target PLATFORMIO_BUILD -- -j 4
Scanning dependencies of target PLATFORMIO_BUILD
[Fri Dec 18 16:41:38 2015] Processing zeroUSB (platform: atmelsam, board: zeroUSB, framework: arduino)
--------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/zeroUSB/src/main.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -DPLATFORMIO=020601 -I.pioenvs/zeroUSB/FrameworkArduino -I.pioenvs/zeroUSB/FrameworkCMSISInc -I.pioenvs/zeroUSB/FrameworkLibSam -I.pioenvs/zeroUSB/FrameworkLibSam/include -I.pioenvs/zeroUSB/FrameworkDeviceInc -I.pioenvs/zeroUSB/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zeroUSB/FrameworkArduinoVariant src/main.cpp
src/main.cpp:1:21: fatal error: Arduino.h: No such file or directory
#include "Arduino.h"
^
compilation terminated.
scons: *** [.pioenvs/zeroUSB/src/main.o] Error 1
========================= [ ERROR ] Took 0.95 seconds =========================
make[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Error 1
make[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
make[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
make: *** [PLATFORMIO_BUILD] Error 2
```

ivankravets, 2015-12-18:
I've just updated/formatted comments by @valeros  https://github.com/platformio/platformio/issues/356#issuecomment-165796874

Have you done all steps?

orome, 2015-12-18:
@ivankravets : Only the last version of the steps, not the **new** version!

orome, 2015-12-18:
@ivankravets , @valeros : Looks like it works.

Full output of upload:
```
--------------------------------------------------------------------------------
BeforeUpload(["upload"], [".pioenvs/zeroUSB/firmware.bin"])
Auto-detected UPLOAD_PORT/DISK: /dev/cu.usbmodem1411
"/Users/Roy/.platformio/packages/tool-bossac/bossac" --info --port cu.usbmodem1411 --erase --write --verify --reset --debug .pioenvs/zeroUSB/firmware.bin
Set binary mode
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
version()=v1.1 [Arduino:XYZ] Jun 10 2015 11:08:10
chipId=0x10010005
Connected at 921600 baud
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
Atmel SMART device 0x10010005 found
write(addr=0x20004000,size=0x34)
writeWord(addr=0x20004030,value=0x10)
writeWord(addr=0x20004020,value=0x20008000)
Device       : ATSAMD21G18A
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
Chip ID      : 10010005
version()=v1.1 [Arduino:XYZ] Jun 10 2015 11:08:10
Version      : v1.1 [Arduino:XYZ] Jun 10 2015 11:08:10
Address      : 8192
Pages        : 4096
Page Size    : 64 bytes
Total Size   : 256KB
Planes       : 1
Lock Regions : 16
Locked       : readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
readWord(addr=0x41004020)=0xffff
none
readWord(addr=0x41004018)=0
Security     : false
Boot Flash   : true
readWord(addr=0x40000834)=0x7000a
BOD          : true
readWord(addr=0x40000834)=0x7000a
BOR          : true
Arduino      : FAST_CHIP_ERASE
Arduino      : FAST_MULTI_PAGE_WRITE
Arduino      : CAN_CHECKSUM_MEMORY_BUFFER
Erase flash
chipErase(addr=0x2000)
done in 0.817 seconds

Write 11136 bytes to flash (174 pages)
write(addr=0x20005000,size=0x1000)
writeBuffer(scr_addr=0x20005000, dst_addr=0x2000, size=0x1000)
[===========                   ] 36% (64/174 pages)write(addr=0x20005000,size=0x1000)
writeBuffer(scr_addr=0x20005000, dst_addr=0x3000, size=0x1000)
[======================        ] 73% (128/174 pages)write(addr=0x20005000,size=0xb80)
writeBuffer(scr_addr=0x20005000, dst_addr=0x4000, size=0xb80)
[==============================] 100% (174/174 pages)
done in 0.096 seconds

Verify 11136 bytes of flash with checksum.
checksumBuffer(start_addr=0x2000, size=0x1000) = 278c
checksumBuffer(start_addr=0x3000, size=0x1000) = 3346
checksumBuffer(start_addr=0x4000, size=0xb80) = f931
Verify successful
done in 0.044 seconds
CPU reset.
readWord(addr=0)=0x200023a0
readWord(addr=0xe000ed00)=0x410cc601
readWord(addr=0x41002018)=0x10010205
writeWord(addr=0xe000ed0c,value=0x5fa0004)
========================= [SUCCESS] Took 5.89 seconds =========================
Built target PLATFORMIO_UPLOAD
```

ivankravets, 2015-12-18:
Great! Will ask @valeros to merge his feature into main repository and update all related packages. Please wait for the further instructions from @valeros 

Thanks for the help and testing.

orome, 2015-12-18:
Cool, thanks!

BTW, how do I maintain a project that uploads to either an Uno or a Zero (depending on which is plugged in)? How do I configure that and build and upload with it. Can I simply rerun `platform init` on an existing project with both boards specified?

ivankravets, 2015-12-18:
Sure! See http://docs.platformio.org/en/latest/userguide/cmd_init.html and "Initialize new PlatformIO based project or **update existing with new data.**"

Here is docs how to control environments http://docs.platformio.org/en/latest/userguide/cmd_run.html

orome, 2015-12-18:
@ivankravets : Can I select which board to upload to from within CLion?

orome, 2015-12-18:
@ivankravets : And what about the third item: how to check what board is targeted in my source (is there a variable that's set?) 's I can, for example, exclude code that's inappropriate for the board?

ivankravets, 2015-12-18:
There a few options

1. Define own macros via `build_src_flags = -DMYMACROS`, then in code use `#ifdef`.
2. Use http://docs.platformio.org/en/latest/projectconf.html#src-filter

ivankravets, 2015-12-18:
> how to check what board is targeted in my source (is there a variable that's set?)

See in console output. @PlatformIO specifies via `-D` type of the current board.

orome, 2015-12-18:
@ivankravets : And can I select which board to upload to from within CLion?

ivankravets, 2015-12-18:
Sure! Please create own target 

```cmake
add_custom_target(
    PLATFORMIO_UPLOAD_ZERO ALL
    COMMAND ${PLATFORMIO_CMD} -f -c clion run --target upload -e zeroUSB
    WORKING_DIRECTORY ${CMAKE_CURRENT_SOURCE_DIR}
)
```

You should like @PlatformIO after it :blush:

orome, 2015-12-22:
@ivankravets , @valeros : Does 2.6.3 have any of these updates in it. Can I use it with the Zero?

ivankravets, 2015-12-22:
@valeros please create new `feature` form the latest `develop` and push your changes. I'll merge it in `2.7.0`.

valeros, 2015-12-22:
We need to test upload with `openOCD` package for this board.
@orome Could you specify your OS, please? I will give you another firmware uploader to test.

orome, 2015-12-22:
@valeros : OS X 10.11.2.

valeros, 2015-12-22:
Thanks @orome, please follow next steps:
Download new uploader https://dl.dropboxusercontent.com/u/10025728/tool-openocd.tar.gz
Place folder `tool-openocd` in PlatformIO home directory `~/.platformio/packages`
Connect your board through the programming USB port.
Create simple blink project with the board `zero` and upload firmware.

orome, 2015-12-22:
@valeros : I currently have `2.6.3`. Should I keep that and just do the above (only)?

valeros, 2015-12-22:
@orome Unfortunately, it will work only with my special version above.

orome, 2015-12-22:
@valeros : So I should repeat [these instructions](https://github.com/platformio/platformio/issues/356#issuecomment-165796874) first, right?

valeros, 2015-12-22:
@orome Yes, 1&2 steps.

orome, 2015-12-22:
@valeros : K, thx. Probably later today.

orome, 2015-12-22:
@valeros : How should I initialize the project? Not `zeroUSB`, right?

valeros, 2015-12-22:
@orome `zero`

orome, 2015-12-22:
@valeros : No luck. It asks to "allow incoming connections", which I allow, and then hangs:
````
/Applications/CLion.app/Contents/bin/cmake/bin/cmake --build /Users/Roy/Library/Caches/CLion12/cmake/generated/2ed14bee/2ed14bee/Debug0 --target PLATFORMIO_UPLOAD -- -j 4
Scanning dependencies of target PLATFORMIO_UPLOAD
[Tue Dec 22 13:38:15 2015] Processing zero (platform: atmelsam, board: zero, framework: arduino)
--------------------------------------------------------------------------------
arm-none-eabi-g++ -o .pioenvs/zero/src/main.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -DPLATFORMIO=020601 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant src/main.cpp
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduinoVariant/variant.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant .pioenvs/zero/FrameworkArduinoVariant/variant.cpp
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduino/IPAddress.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant .pioenvs/zero/FrameworkArduino/IPAddress.cpp
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduino/Print.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant .pioenvs/zero/FrameworkArduino/Print.cpp
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduino/Reset.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant .pioenvs/zero/FrameworkArduino/Reset.cpp
arm-none-eabi-ar rcs .pioenvs/zero/libFrameworkArduinoVariant.a .pioenvs/zero/FrameworkArduinoVariant/variant.o
arm-none-eabi-ranlib .pioenvs/zero/libFrameworkArduinoVariant.a
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduino/RingBuffer.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUSB_MANUFACTURER="PlatformIO" -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804d "-DUSB_PRODUCT=\"Arduino Zero\"" -DARDUINO=10606 -I.pioenvs/zero/FrameworkArduino -I.pioenvs/zero/FrameworkCMSISInc -I.pioenvs/zero/FrameworkLibSam -I.pioenvs/zero/FrameworkLibSam/include -I.pioenvs/zero/FrameworkDeviceInc -I.pioenvs/zero/FrameworkDeviceInc/d21g18a/include -I.pioenvs/zero/FrameworkArduinoVariant .pioenvs/zero/FrameworkArduino/RingBuffer.cpp
arm-none-eabi-g++ -o .pioenvs/zero/FrameworkArduino/SERCOM.o -c -fno-rtti -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -ffunction-sections -fdata-sections -Wall -mthumb -mcpu=cortex-m0plus -nostdlib -MMD --param max-inline-insns-single=500 -DF_CPU=48000000L -DUSBCON -DUS [... cut ...]
debug_level: 2
Info : only one transport option; autoselect 'swd'
adapter speed: 500 kHz
adapter_nsrst_delay: 100
cortex_m reset_config sysresetreq
4444
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : CMSIS-DAP: FW Version = 02.01.0157
Info : SWCLK/TCK = 1 SWDIO/TMS = 1 TDI = 1 TDO = 1 nTRST = 0 nRESET = 1
Info : CMSIS-DAP: Interface ready
Info : clock speed 500 kHz
Info : SWD IDCODE 0x0bc11477
Info : at91samd21g18.cpu: hardware has 4 breakpoints, 2 watchpoints
````

orome, 2015-12-23:
@valeros, @ivankravets : And fwiw, `2.6.1.dev2` is downgrading `click` (to 5.; latest should be 6.2).

ivankravets, 2015-12-23:
> And fwiw, 2.6.1.dev2 is downgrading click (to 5.; latest should be 6.2).

See related issue for that https://github.com/platformio/platformio/issues/349

orome, 2015-12-30:
@ivankravets - Ugh; any idea when 3.0.0 is expected?

ivankravets, 2015-12-30:
I don't plan to release PlatformIO 2.8.0. However, if @valeros resolves this issue before PlatformIO 3.0, I'll release it in 2.8.0. 

@valeros has vacation till Jan 11, 2016 :christmas_tree: If he has spare time till Jan 11, 2016, he will help you. 

Sorry for inconvenience.

orome, 2015-12-31:
@ivankravets — Will `zeroUSB` work with 2.7.0; or will neither `zeroUSB` nor `zero`?

ivankravets, 2015-12-31:
@orome please wait for @valeros. I've discussed with him today. He will prepare for your new branch with the support for `zero`. Of course, the branch will be based on the latest PlatformIO 2.7.0.

orome, 2015-12-31:
@ivankravets — Excellent, thanks!

closed 2016-01-05 by valeros

orome, 2016-01-05:
@valeros — Does this mean that the Zero is supported in 2.7.1 (with both `zeroUSB` and `zero`)?

ivankravets, 2016-01-05:
Yes! :) PlatformIO 2.7.1 will be released tomorrow.

orome, 2016-01-05:
:star:

chrise96, 2017-03-02:
`mzeropro` is working on my .org arduino m0 pro. (Bootloader_D21)
But the `zero` doesn't work for me with the .cc Arduino Zero. (Bootloader SAM-BA)

ivankravets, 2017-03-02:
@chrise96 please file an issue here https://github.com/platformio/platform-atmelsam/issues

commit 9dde962a (2016-01-05): Initial support for Arduino Zero board // Resolve #356
commit fcf50a80 (2016-01-05): Merge pull request #436 from valeros/develop

Initial support for Arduino Zero board // Resolve #356