platformio/platformio-core #357: Json (library 64) gets incorrect headers

lbuchy, 2015-12-06:
Tried installing the 'Json' library (ArduinoJson) and incorporating it into a project.  Basically:

```
platformio lib install 64
```

And in the code
```c++
#include <ArduinoJson.h>
```

I am getting this error:
```
bash# platformio run
[Sat Dec  5 17:42:57 2015] Processing esp12e (platform: espressif, board: esp12e, framework: arduino, build_flags: -D USE_US_TIMER -I $lib_dir)
--------------------------------------------------------------------------------
xtensa-lx106-elf-g++ -o .pioenvs/esp12e/src/tmp_ino_to.o -c -fno-rtti -fno-exceptions -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -MMD -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266 -DESP8266 -DUSE_US_TIMER -DARDUINO=200 -DPLATFORMIO=020401 -I/home/lbuchy/.platformio/packages/framework-arduinoespressif/tools/sdk/include -I.pioenvs/esp12e/FrameworkArduino -I.pioenvs/esp12e/FrameworkArduinoVariant -I.pioenvs/esp12e/ESP8266WiFi -I.pioenvs/esp12e/ESP8266WebServer -I.pioenvs/esp12e/ESP8266mDNS -I.pioenvs/esp12e/MicrosecondTicker -I.pioenvs/esp12e/Json_ID64 -I.pioenvs/esp12e/LaserMon src/tmp_ino_to.cpp
In file included from .pioenvs/esp12e/LaserMon/Config.hpp:6:0,
from src/tmp_ino_to.cpp:7:
.pioenvs/esp12e/Json_ID64/ArduinoJson.h:13:36: fatal error: ../include/ArduinoJson.h: No such file or directory
#include "../include/ArduinoJson.h"
^
compilation terminated.
scons: *** [.pioenvs/esp12e/src/tmp_ino_to.o] Error 1
```

Here is my platformio.ini
```
[env:esp12e]
platform = espressif
framework = arduino
board = esp12e
lib_install = 64
lib_use = Json
#targets = upload
build_flags=-D USE_US_TIMER
```

I'm not exactly well versed with how this works under the hood.  
It looks like Platformio is picking up the ArduinoJson.h in the library's src/ folder.  This file then includes '../include/ArduinoJson.h'.  This file was added to the library to 'play nice' with the Arduino IDE.  Why is this file copied into .pioenvs/esp12e/Json_ID64/?  Why does this happen?  Could we not just reference the library directly where it stands?

Would a solution be to add the following to the Json library's library.json?
```
"exclude": [
  "src/ArduinoJson.h"
],
"include": [
  "include"
]
```
How would I go about testing this?  I don't see a way to install 'local' libraries with platformio.

ivankravets, 2015-12-06:
> How would I go about testing this? I don't see a way to install 'local' libraries with platformio.

Please place the source of this code temporary to `lib` directory (which does mean local). See http://docs.platformio.org/en/latest/projectconf.html#lib-dir

lbuchy, 2015-12-06:
Does Platformio read the library.json file for  libraries in this directory?

ivankravets, 2015-12-07:
PlatformIO 2.0 doesn't handle `library.json` file. However, PlatformIO 3.0 will do it. See https://github.com/platformio/platformio/issues/289

P.S: I see the issue with this library. I will propose your a few solutions later. Sorry. I'm working on the PlatformIO 2.5 which should be released tomorrow.

itsjustvenky, 2015-12-11:
Just use 
```
#include "include/ArduinoJson.h"
```
and it works.

I used these parameters in platformio.ini
```
lib_install = 64
lib_use = Json
```

itsjustvenky, 2015-12-11:
here it is how it worked for me
1) Downloaded library from 
```
https://github.com/bblanchon/ArduinoJson
```
2) Installed into "lib" folder of my project as "ArduinoJson".

3) Added to my ino/cpp : #include "include/ArduinoJson.h"

4) built project and it worked fine.

ivankravets, 2015-12-11:
@itsjustvenky how about without direct downloading? Like
```
platformio lib install 64

#include "include/ArduinoJson.h"
```
?

itsjustvenky, 2015-12-11:
It didn't work for me.

ivankravets, 2015-12-16:
@itsjustvenky what is wrong with@bblanchon manifest file https://github.com/bblanchon/ArduinoJson/blob/master/library.json ?

Could you correct it according http://docs.platformio.org/en/latest/librarymanager/config.html ?

Thanks!

librae8226, 2015-12-22:
Hello @ivankravets, according to @itsjustvenky, `#include "include/ArduinoJson.h"` can get it work, however in fact it has nothing to do with direct downloading Json_ID64.
To get it pass the build, I just removed the `test/` directory in ArduinoJson.
The `gtest/gtest.h` thing is one of the unit test code of ArduinoJson itself I guess, so it may not fit platformio, sounds reasonable.
But there might still be a bit difference with Arduino IDE environment.
Since in the original <ArduinoJson.h> file, it says
```
// About this file
// ---------------
// This file is here to please the Arduino IDE. It must be present in the src/
// for the IDE to find it. Feel free to ignore this file if your working in
// another environment

#include "../include/ArduinoJson.h"
```

sticilface, 2015-12-22:
I'm getting various errors when trying to include this lib... either via install 64 or by downloading it.
anyone know how to get around the multiple builds error?
```

scons: warning: Two different environments were specified for target /Users/amelvin/platformio/melvana/.pioenvs/nodemcu/libArduinoJson.a,
but they appear to have the same action: $AR $ARFLAGS $TARGET ${_huge_sources_hook(SOURCES)}
$RANLIB $RANLIBFLAGS $TARGET
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 203, in BuildLibrary

scons: *** Multiple ways to build the same target were specified for: /Users/amelvin/platformio/melvana/.pioenvs/nodemcu/libArduinoJson.a  (from ['/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Arduino/Print.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/Comments.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/Encoding.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/IndentedPrint.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/JsonParser.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/List.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/Prettyfier.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/Internals/StaticStringBuilder.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/JsonArray.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/JsonBuffer.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/JsonObject.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/JsonVariant.o'] and from ['/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Arduino/Print.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/Comments.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/Encoding.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/IndentedPrint.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/JsonParser.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/List.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/Prettyfier.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/Internals/StaticStringBuilder.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/JsonArray.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/JsonBuffer.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/JsonObject.o', '/Users/amelvin/platformio/melvana/.pioenvs/nodemcu/ArduinoJson/src/JsonVariant.o'])
File "/Library/Python/2.7/site-packages/platformio/builder/tools/platformio.py", line 203, in BuildLibrary
```

ivankravets, 2015-12-22:
1 sec. I'm working on it.

sticilface, 2015-12-22:
ah very good thanks:)

ivankravets, 2015-12-22:
Guys, please `platformio lib update` and re-test that it works. I've just tested and it works good for me.

Need to use this `include`
```cpp
#include <include/ArduinoJson.h>
```

ivankravets, 2015-12-22:
Guys, you should say thanks to @bblanchon, he has just added support for PlatformIO.

Summary:

1. Please update library via `platformio lib update`
2. Use general include `#include <ArduinoJson.h>`

Related issue https://github.com/bblanchon/ArduinoJson/issues/181

closed 2015-12-22 by ivankravets